start-jsk/rtmros_tutorials #170: Failed to run hrpsys/ProjectGenerator in catkin build

garaemon, 2015-02-06:
At first time building, rostest fails to find hrpsys

```
... logging to /home/hrpuser/.ros/log/rostest-hrp2017v-28703.log
testprojectgenerator (rostest.runner.RosTest) ... [ROSTEST]setup[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch] run_id[51986010-add0-11e4-98f9-00031d0c7c5c] starting
started roslaunch server http://hrp2017v:43950/
ros_comm version 1.10.12

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion

NODES
  /
    modelloader (openhrp3/openhrp-model-loader)

[ROSTEST]setup[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch] run_id[51986010-add0-11e4-98f9-00031d0c7c5c] done
Package [hrpsys] for test node [hrpsys/ProjectGenerator] does not exist
FAIL
[ROSTEST]tearDown[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch]
shutting down processing monitor...
... shutting down processing monitor complete
[ROSTEST]rostest teardown /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch complete

======================================================================
FAIL: testprojectgenerator (rostest.runner.RosTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rostest/runner.py", line 94, in fn
    self.fail(message)
AssertionError: Package [hrpsys] for test node [hrpsys/ProjectGenerator] does not exist

----------------------------------------------------------------------
Ran 1 test in 0.142s

FAILED (failures=1)

WARNING: overall test result is not accurate when --text is enabled
rostest log file is in /home/hrpuser/.ros/log/rostest-hrp2017v-28703.log
make[2]: *** [/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.xml]  1
make[1]: *** [CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/all]  2
make: *** [all]  2
[hrpsys_ros_bridge_tutorials] <== '/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials/build_env.sh /usr/bin/make -j4 -l4' failed with return code '2'
                                                                                                                                                                                                                                                                
[build] There were errors:                                                                                                                                                                                                                                     

    Failed to build package 'hrpsys_ros_bridge_tutorials' because the following command:

        # Command run in directory: /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials
        /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials/build_env.sh /usr/bin/make -j4 -l4

    Exited with return code: 2 
[build] Runtime: 10 minutes and 41.6 seconds 

```

garaemon, 2015-02-06:
in order to compile it, now we need to
1. compile hrpsys
2. source devel/setup.bash
3. compile hrpsys_ros_bridge_tutorials

garaemon, 2015-02-06:
Even though `rosrun hrpsys ProjectGenerator` is possible, _gen_project.launch fails

```
$ ./build_env.sh make
[  0%] Generating /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.xml
Starting omniORB omniNames: hrp2017v:2971
omniORB: Failed to bind to address :: port 2971. Address in use?
omniORB: Error: Unable to create an endpoint of this description: giop:tcp::2971
Failed to initialise the ORB / POA: INITIALIZE_TransportError
Is omniNames already running?
omniNames was not properly started.
(一部のプロセスが識別されますが, 所有していないプロセスの情報は
表示されません。それら全てを見るにはルートになる必要があります.)
omniNames (pid: 7650) is running
Kill anyway and start omniNames again? (y/N)y
omniNames (pid: 7650) are killed
Starting omniORB omniNames: hrp2017v:2971

Fri Feb  6 17:56:52 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000e0000003139322e3136382e39362e3234009b0b0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000d481d45401002e5d
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
omniNames properly started
... logging to /home/hrpuser/.ros/log/rostest-hrp2017v-11881.log
testprojectgenerator (rostest.runner.RosTest) ... [ROSTEST]setup[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch] run_id[18e8de26-adde-11e4-bc02-00031d0c7c5c] starting
started roslaunch server http://192.168.96.24:33384/
ros_comm version 1.10.12

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion

NODES
  /
    modelloader (openhrp3/openhrp-model-loader)

[ROSTEST]setup[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch] run_id[18e8de26-adde-11e4-bc02-00031d0c7c5c] done
Package [hrpsys] for test node [hrpsys/ProjectGenerator] does not exist
FAIL
[ROSTEST]tearDown[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch]
shutting down processing monitor...
... shutting down processing monitor complete
[ROSTEST]rostest teardown /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch complete

======================================================================
FAIL: testprojectgenerator (rostest.runner.RosTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/opt/ros/hydro/lib/python2.7/dist-packages/rostest/runner.py", line 94, in fn
    self.fail(message)
AssertionError: Package [hrpsys] for test node [hrpsys/ProjectGenerator] does not exist

----------------------------------------------------------------------
Ran 1 test in 0.142s

FAILED (failures=1)

WARNING: overall test result is not accurate when --text is enabled
rostest log file is in /home/hrpuser/.ros/log/rostest-hrp2017v-11881.log
make[2]: *** [/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.xml] エラー 1
make[1]: *** [CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/all] エラー 2
```

garaemon, 2015-02-06:
VERBOSE=1
```
$ VERBOSE=1 ./build_env.sh make
/usr/bin/cmake -H/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials -B/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials/CMakeFiles /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: ディレクトリ `/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials' に入ります
make -f CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/build.make CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/depend
make[2]: ディレクトリ `/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials' に入ります
cd /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials/CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/DependInfo.cmake --color=
make[2]: ディレクトリ `/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials' から出ます
make -f CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/build.make CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/build
make[2]: ディレクトリ `/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials' に入ります
/usr/bin/cmake -E cmake_progress_report /home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials/CMakeFiles 
[  0%] Generating /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.xml
/home/hrpuser/ros/hydro/devel/lib/openrtm_aist/bin/rtm-naming 2971 || echo fail\ to\ run\ rtm_naming,\ but\ try\ to\ continue
Starting omniORB omniNames: hrp2017v:2971
omniORB: Failed to bind to address :: port 2971. Address in use?
omniORB: Error: Unable to create an endpoint of this description: giop:tcp::2971
Failed to initialise the ORB / POA: INITIALIZE_TransportError
Is omniNames already running?
omniNames was not properly started.
(一部のプロセスが識別されますが, 所有していないプロセスの情報は
表示されません。それら全てを見るにはルートになる必要があります.)
omniNames (pid: 11869) is running
Kill anyway and start omniNames again? (y/N)y
omniNames (pid: 11869) are killed
Starting omniORB omniNames: hrp2017v:2971

Fri Feb  6 17:59:39 2015:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000e0000003139322e3136382e39362e3234009b0b0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000007b82d4540100326c
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
omniNames properly started
sh -c CMAKE_PREFIX_PATH=:/home/hrpuser/ros/hydro/devel:/opt/ros/hydro\ ROS_PACKAGE_PATH=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools:/share/hrpsys:/home/hrpuser/ros/hydro/devel/share/openhrp3:/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_common/3rdparty/assimp_devel:/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/geneus:/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/rtmbuild:/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/openrtm_tools:/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_roseus/roseus:/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools:/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_ros_bridge:/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_model_tools/euscollada:/home/hrpuser/ros/hydro/src/jsk-ros-pkg/jsk_pr2eus/pr2eus:/opt/ros/hydro/share:/opt/ros/hydro/stacks\ rostest\ -t\ /home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch\ CORBA_PORT:=2971\ INPUT:=/home/hrpuser/ros/hydro/devel/share/OpenHRP-3.1/idl/../sample/model/PA10/pa10.main.wrl\ OUTPUT:=/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.xml\ \ OBJECT_MODELS:=/home/hrpuser/ros/hydro/devel/share/OpenHRP-3.1/sample/model/longfloor.wrl,0,0,0,0,0,1,0\ \ \ 
... logging to /home/hrpuser/.ros/log/rostest-hrp2017v-12922.log
testprojectgenerator (rostest.runner.RosTest) ... [ROSTEST]setup[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch] run_id[7c85631e-adde-11e4-a844-00031d0c7c5c] starting
started roslaunch server http://192.168.96.24:60153/
ros_comm version 1.10.12

SUMMARY
========

PARAMETERS
 * /rosdistro
 * /rosversion

NODES
  /
    modelloader (openhrp3/openhrp-model-loader)

[ROSTEST]setup[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch] run_id[7c85631e-adde-11e4-a844-00031d0c7c5c] done
Package [hrpsys] for test node [hrpsys/ProjectGenerator] does not exist
FAIL
[ROSTEST]tearDown[/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_common/hrpsys_tools/launch/_gen_project.launch]
shutting down processing monitor...
... shutting down processing monitor complete
[... cut ...]
WARNING: overall test result is not accurate when --text is enabled
rostest log file is in /home/hrpuser/.ros/log/rostest-hrp2017v-12922.log
make[2]: *** [/home/hrpuser/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/pa10.xml] エラー 1
make[2]: ディレクトリ `/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials' から出ます
make[1]: *** [CMakeFiles/pa10_hrpsys_ros_bridge_tutorials_compile.dir/all] エラー 2
make[1]: ディレクトリ `/home/hrpuser/ros/hydro/build/hrpsys_ros_bridge_tutorials' から出ます
make: *** [all] エラー 2

```

garaemon, 2015-02-06:
I think `find_package(hrpsys)` fails
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/cmake/compile_robot_model.cmake#L472

How can we know location of hrpsys?

garaemon, 2015-02-06:
```
[http://hrp2017v:10017][192.168.96.24] hrpuser@hrp2017v:~/ros/hydro$ find . -name hrpsys.pc
[http://hrp2017v:10017][192.168.96.24] hrpuser@hrp2017v:~/ros/hydro$ find . -name hrpsysConfig.cmake
[http://hrp2017v:10017][192.168.96.24] hrpuser@hrp2017v:~/ros/hydro$ find . -name hrpsys-config.cmake
```

k-okada, 2015-02-06:
I see,
```
    find_package(hrpsys QUIET)
    if(NOT hrpsys_QUIET))
      find_package(PkgConfig)
      pkg_check_modules(hrpsys hrpsys REQUIRED)
    endif()
    set(hrpsys_PACKAGE_PATH ${hrpsys_SOURCE_DIR})

```
should work,
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/cmake/compile_robot_model.cmake#L223

garaemon, 2015-02-06:
according to hrpsy_ros_bridge, I found that I needed to add following code to hrpsys_ros_bridge_tutorials
```
find_package(hrpsys QUIET) # on indigo, hrpsys is not ros package                                                                                                
if(NOT ${hrpsys_FOUND})
  find_package(PkgConfig)
  pkg_check_modules(hrpsys hrpsys-base REQUIRED)
endif()
```

k-okada, 2015-02-06:
On Fri, Feb 6, 2015 at 6:44 PM, Ryohei Ueda <notifications@github.com>
wrote:

> pkg_check_modules(hrpsys hrpsys-base REQUIRED)

yes, this is correct

--
◉ Kei Okada

garaemon, 2015-02-06:
fixed by https://github.com/start-jsk/rtmros_tutorials/pull/173

closed 2015-02-06 by garaemon