start-jsk/rtmros_tutorials #172: duplication of hironx_tutorials

garaemon, 2015-02-06:
```
$ catkin build jsk_hrp2_ros_bridge -v
------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/hydro
Workspace:                   /home/hrpuser/ros/hydro
Source Space:       [exists] /home/hrpuser/ros/hydro/src
Build Space:        [exists] /home/hrpuser/ros/hydro/build
Devel Space:        [exists] /home/hrpuser/ros/hydro/devel
Install Space:     [missing] /home/hrpuser/ros/hydro/install
DESTDIR:                     None
------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
------------------------------------------------------------
Workspace configuration appears valid.
------------------------------------------------------------ 
[build] Runtime: 0.4 seconds 
Traceback (most recent call last):
  File "/usr/bin/catkin", line 9, in <module>
    load_entry_point('catkin-tools==0.2.0', 'console_scripts', 'catkin')()
  File "/usr/lib/python2.7/dist-packages/catkin_tools/commands/catkin.py", line 198, in main
    sys.exit(args.main(args) or 0)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/cli.py", line 219, in main
    no_notify=opts.no_notify
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 416, in build_isolated_workspace
    packages_to_be_built, packages_to_be_built_deps, all_packages = determine_packages_to_be_built(packages, context)
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 151, in determine_packages_to_be_built
    workspace_packages = find_packages(context.source_space_abs, exclude_subspaces=True)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 92, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "hironx_tutorial":
- rtm-ros-robotics/rtmros_hironx/hironx_tutorial
- rtm-ros-robotics/rtmros_tutorials/hironx_tutorial

```

garaemon, 2015-02-06:
fixed via https://github.com/start-jsk/rtmros_hironx/pull/321

closed 2015-02-06 by garaemon

130s, 2015-02-06:
:sake: 

DEB of `rtmros_hironx` should become available in a few days in shadow-f if necessary.