start-jsk/rtmros_tutorials #183: hrpsys_ros_bridge_tutorialsのcatkin buildが終了しない

orikuma, 2015-02-11:
catkin build hrpsys_ros_bridge_tutorials  --make-args -j1
でhrpsys_ros_bridge_tutorialsでcatkin bulidするとcatkin buildが終了しません.
psを見るとhrp2jskのmin-max-tableの作成で止まっているように見えます.

```bash
iori     19928  0.0  0.0  12612  2268 pts/10   S+   01:42   0:00 /usr/bin/make -f CMakeFiles/hrp2jsk_hrpsys_ros_bridge_tutorials_compile2.dir/build.make CMakeFiles/hrp2jsk_hrpsys_ros_bridge_tutorials_compile2.dir/build
iori     19935  0.0  0.0   4464   868 pts/10   S+   01:42   0:00 /bin/sh -c /home/iori/ros/indigo/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsk.l ""\(write-min-max-table\ \(hrp2jsk\)\ \"/home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsk.l\"\ :margin\ 1.0\)"" "(exit)" && touch /home/iori/ros/indigo/build/hrpsys_ros_bridge_tutorials/hrp2jsk_joint_minmax_done
iori     19936  0.0  0.3 244140 26420 pts/10   S+   01:42   0:00 /home/iori/ros/indigo/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsk.l (write-min-max-table (hrp2jsk) "/home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsk.l" :margin 1.0) (exit)
iori     23104  1.5  0.3 374496 28440 pts/10   Sl+  01:41   4:22 /usr/bin/python /usr/bin/catkin build hrpsys_ros_bridge_tutorials --make-args -j1
iori     26499  0.0  0.0  16936  2164 pts/71   S+   06:21   0:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=.cvs --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn hrpsys
```

k-okada, 2015-02-12:
cd build/hrpsys_ros_bridge_tutorials; make VERBOSE=1
してみてください．

garaemon, 2015-02-12:
たぶんomniNamesが他に立ち上がっています

✉︎ Ryohei Ueda

k-okada, 2015-02-12:
closed due to @orikuma has more important tasks to focus on

garaemon, 2015-02-12:
> closed due to @orikuma has more important tasks to focus on

It's not a reasonable  reason to close issue

garaemon, 2015-02-12:
I found a lot of problems of hrpsys_ros_bridge_tutorials with `catkin build` but I don't have time now to play with and solve issues of building system. So I write downs several hacks to build this project in `catkin build`

My solution is run following command when you have error in compiling hrpsys_ros_bridge_tutorials.
```
# amazingly, we cannot use rocsd after successful compilation
cd ~/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials
# clean up
git clean -xfd
# compile with 1 process
catkin build -p1 hrpsys_ros_bridge_tutorials --force-cmake --start-with hrpsys_ros_bridge_tutorials --make-args -j1
```

1. kill omniNames before building hrpsys_ros_bridge_tutorials.

  If you have error or sent ctrl-c in building hrpsys_ros_bridge_tutorials, you may have zombie omniNames. Please kill it before re-building hrpsys_ros_bridge_tutorials.
2. invoking euslisp in parallel may cause strange error in catkin build.
 
  I don't know why and I have never seen this erorr in catkin_make. (DISPLAY environmetal variable may be a problem... https://github.com/tork-a/jskeus-release/pull/2)
  In order to avoid this problem, run catkin build with -p1 and -j1 like:
```
catkin build -p1 hrpsys_ros_bridge_tutorials --fore-cmake --make-args -j1
```
3. Invoking many export-to-collada processes may cause strange error.
  
  I don't know why and I have never seen this error in catkin_make.
  You can avoid this problem with the same approach in 2.

garaemon, 2015-02-12:
And alsot STARO model and Jaxson model conversion might have wrong dependency.
I though I have solved this problem [here](https://github.com/start-jsk/rtmros_tutorials/pull/166), but confirmation is not perfect.

orikuma, 2015-02-23:
この問題は jsk-ros-pkg/jsk_model_tools#98 で解決しますが, 
生成されたモデルの関節順序がおかしい問題がまだ残っています.
(jsk-ros-pkg/jsk_model_tools#99)

garaemon, 2015-03-03:
* https://github.com/start-jsk/rtmros_common/pull/675
* https://github.com/start-jsk/rtmros_tutorials/pull/193

These two should solve compilation problem on hydro