start-jsk/rtmros_tutorials #184: samplerobot-interface.lが立ち上がらない

orikuma, 2015-02-11:
Ubuntu14.04+indigoでsamplerobot-interface.lが立ち上がりません.

```bash
$ roseus samplerobot-interface.l
configuring by "/home/iori/ros/indigo/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb
;; extending gcstack 0x4943de0[16374] --> 0x4d9fc80[32748] top=378a
irtgl irtviewer
EusLisp 9.10(0f0f62e 1.0.2) for Linux64 created on iori-x1c(Wed Feb 11 23:04:30 JST 2015)
roseus ;; loading roseus("fba5d29") on euslisp((9.10 iori-x1c Wed Feb 11 23:04:30 JST 2015 0f0f62e 1.0.2))
eustf roseus_c_util [ERROR] [1423694012.028704114]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.028972603]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.029638946]: Could not find roseus messages for catkin under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh catkin
[ERROR] [1423694012.029676011]: Could not find roseus messages for catkin under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh catkin
[ERROR] [1423694012.030336441]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.030376944]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.060952702]: Could not find roseus messages for rosgraph under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh rosgraph
[ERROR] [1423694012.061049543]: Could not find roseus messages for rosgraph under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh rosgraph
[ERROR] [1423694012.061305161]: Could not find roseus messages for roslz4 under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh roslz4
[ERROR] [1423694012.061352127]: Could not find roseus messages for roslz4 under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh roslz4
[ERROR] [1423694012.061784496]: Could not find roseus messages for rospack under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh rospack
[ERROR] [1423694012.061825321]: Could not find roseus messages for rospack under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh rospack
[ERROR] [1423694012.100065703]: Could not find roseus messages for rosclean under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh rosclean
[ERROR] [1423694012.100141596]: Could not find roseus messages for rosclean under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh rosclean
[ERROR] [1423694012.101896966]: Could not find roseus messages for euslisp under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh euslisp
[ERROR] [1423694012.101941688]: Could not find roseus messages for euslisp under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh euslisp
[ERROR] [1423694012.187068406]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.187125469]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.187306138]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.187355681]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.188611945]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.188665273]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.188846187]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.188890152]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.189714370]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.189758494]: Could not find roseus messages for genmsg under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh genmsg
[ERROR] [1423694012.189972061]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.190006925]: Could not find roseus messages for cpp_common under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh cpp_common
[ERROR] [1423694012.191581549]: Could not find roseus messages for rosbag_migration_rule under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
[... 61 lines cut ...]
gen_eus.py: error: please specify args
ERROR in /opt/ros/indigo/share/rosclean
generating... 7/9
generating manifest... cpp_common
Usage: gen_eus.py file

gen_eus.py: error: please specify args
ERROR in /opt/ros/indigo/share/cpp_common
generating... 8/9
generating manifest... rosbag_migration_rule
Usage: gen_eus.py file

gen_eus.py: error: please specify args
ERROR in /opt/ros/indigo/share/rosbag_migration_rule
[ERROR] occurred while processing /home/iori/ros/indigo/src/jsk-ros-pkg/jsk_roseus/roseus/scripts/generate-all-msg-srv.sh
/opt/ros/indigo/share/genmsg
/opt/ros/indigo/share/catkin
/opt/ros/indigo/share/cpp_common
/opt/ros/indigo/share/rosgraph
/opt/ros/indigo/share/roslz4
/opt/ros/indigo/share/rospack
/opt/ros/indigo/share/rosclean
/opt/ros/indigo/share/cpp_common
/opt/ros/indigo/share/rosbag_migration_rule
```

orikuma, 2015-02-11:
generate-all-msg-srv.shはjsk_roseusのPR#224まで更新済みです.

k-okada, 2015-02-11:
https://github.com/jsk-ros-pkg/jsk_roseus/pull/226 で直しました

orikuma, 2015-02-12:
修正を適用しましたがpr2_controllers_msgsのmsgが見えていないようです.
roseus generate-all-msg-srv.sh pr2_controllers_msgsはエラーなくできています.

```bash
configuring by "/home/iori/ros/indigo/devel/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb
;; extending gcstack 0x4b42de0[16374] --> 0x4f9ec80[32748] top=378a
irtgl irtviewer
EusLisp 9.10(0f0f62e 1.0.2) for Linux64 created on iori-x1c(Wed Feb 11 23:04:30 JST 2015)
roseus ;; loading roseus("fba5d29") on euslisp((9.10 iori-x1c Wed Feb 11 23:04:30 JST 2015 0f0f62e 1.0.2))
eustf roseus_c_util [ERROR] [1423702354.845305123]: Could not find roseus messages for /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs
[ERROR] [1423702354.845414343]: Could not find roseus messages for /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs under (/home/iori/ros/indigo/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)
try rosrun roseus generate-all-msg-srv.sh /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs
/home/iori/ros/indigo/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 no such package "PR2_CONTROLLERS_MSGS""PR2_CONTROLLERS_MSGS" in (apply #'ros::load-org-for-ros ros::fullname args)E: (exit)
[ INFO] [1423702373.295701736]: cell* ROSEUS_EXIT(context*, int, cell**)
```

```bash
package:pr2_controllers_msgs -> /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs
generating... 1 files with ALL=No, COMPILE=No option
generating... 0/1
generating msg... /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs/msg/Pr2GripperCommand.msg
generating msg... /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs/msg/JointControllerState.msg
generating msg... /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs/msg/JointTrajectoryControllerState.msg
generating srv... /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs/srv/QueryCalibrationState.srv
generating srv... /home/iori/ros/indigo/src/rtm-ros-robotics/rtmros_common/pr2_controllers_msgs/srv/QueryTrajectoryState.srv
generating manifest... pr2_controllers_msgs
```

YoheiKakiuchi, 2015-02-12:
@orikuma さんには他のキューに入っている仕事に集中してもらいたい気がします．

k-okada, 2015-02-12:
Closed due to reporter have more important tasks

garaemon, 2015-02-12:
> Closed due to reporter have more important tasks

It's not a reasonable reason to close issue

garaemon, 2015-02-12:
But it does not mean @orikuma should resolve this issue now

k-okada, 2015-02-12:
@orikuma should not resolve issuen on 14.04 until @YoheiKakiuchi said it's ok.

snozawa, 2016-04-08:
ubuntu 14.04 + indigoでchoreonoidまで使えているので治ってると思います

closed 2016-04-08 by snozawa