start-jsk/rtmros_tutorials #233: JAXONの腕の関節名がおかしい

YoheiKakiuchi, 2015-04-09:
jaxon.yamlに定義されている腕の関節名がおかしい。
例えば、１軸目はshoulder-yaw だが collar-yaw としたほうがいいなど。
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/models/jaxon.yaml

mmurooka, 2015-04-10:
既に :shoulder-y で書いてしまっているプログラムもそれなりにありそうですが，
:shoulder-y も維持しつつ :collar-y でもアクセスできるようになるでしょうか．

YoheiKakiuchi, 2015-04-10:
>  :shoulder-y も維持しつつ :collar-y でもアクセスできるようになるでしょうか．

ちょっと難しいかと思います．

> 既に :shoulder-y で書いてしまっているプログラムもそれなりにありそうですが，

どれぐらいあるかな？
まだ変更可能な数だったりしないかな？

kindsenior, 2015-04-10:
>:shoulder-y も維持しつつ :collar-y でもアクセスできるようになるでしょうか．

すみません，あんまり分かっていないかもしれませんが，
間違い(今) 正解
:shoulder-y -> :collar-y
:elbow-y       -> :shoulder-y
だと思っていて,
その場合shoulder-yは変更後にelbow-yだった軸に割り当てられるので
「shoulder-yを維持しておく」というのは無理なんじゃないでしょうか?

間違ってたらごめんなさい

_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/
東京大学 大学院
学際情報学府 学際情報学専攻
〒113-8656
東京都文京区本郷7-3-1 工学部2号館 8階83B1
小島 邦生 (KOJIMA KUNIO)
k-kojima@jsk.imi.i.u-tokyo.ac.jp
Tel 03-5841-7416
Fax 03-5841-6285
携帯 090-4501-5285
_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/_/

mmurooka, 2015-04-10:
> どれぐらいあるかな？
まだ変更可能な数だったりしないかな？

https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/torque_talker/hrp-talk-torque.l#L29-L48
でロボットモデルに存在しうる関節の候補をリストアップしているところに，
collar-yを追加しないといけないとかでしょうか．
でも，思ったより全然なさそうだったので，変えるなら今な気がしてきました．

eisoku9618, 2015-04-10:
遅くなってすみません＋名前間違えていてすみません．
上で対応しました．

closed 2015-04-11 by YoheiKakiuchi