start-jsk/rtmros_tutorials #245: testbedをgazeboで表示する

furushchev, 2015-04-16:
irteusのモデルから
https://gist.github.com/bd5bd4a9366216cb86b1
のようなコードを使ってurdfとworldファイルを生成してみました。
これを使ってgazeboを起動すると、オブジェクトは表示されますが、位置などがおかしいようです。
正解はどのような感じなのでしょうか？（確認できるものはありますか？）

[image]

mmurooka, 2015-04-16:
irtモデルは以下のモデルから変換されているだろうから，
https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_2015_06_hrp_drc/drc_task_common/euslisp/drc-testbed-models.l#L838 で，pickviewに表示される配置が正解です．

https://gist.github.com/furushchev/bd5bd4a9366216cb86b1#file-drc-irtmodel-to-urdf-l-L89-L94
のposやrpyが正しく取れていなさそうです．
前にもあった気がするけれど，解決できたか，どうやって解決したかすぐに思い出せないです．

furushchev, 2015-04-20:
@mmurooka 遅れました。アドバイスありがとうございます。

@YoheiKakiuchi に手伝っていただき、irtviewerでもオブジェクトの位置を正しくだせるようになりました。
なのでworldファイルは正しく出力されていると思うのですが、gazeboでは依然としておかしな座標になっています。

mmurooka, 2015-04-20:
worldファイルは本当に正しかったかな．
前に同じような現象がおきたときは，

irtviewerでは正しい位置にでるけれど，
なぜか
https://gist.github.com/furushchev/bd5bd4a9366216cb86b1#file-drc-irtmodel-to-urdf-l-L89-L94
で正しい位置がとれず，worldファイルの位置がおかしかった記憶があります．

furushchev, 2015-04-20:
スクリーンショットを貼り忘れました。

- irtviewer:
[image]

- gazebo:
[image]

YoheiKakiuchi, 2015-04-20:
とりあえず、worldファイルはどうなってるか見てみたい。

furushchev, 2015-04-20:
@YoheiKakiuchi 以下に貼りました。
https://gist.github.com/911a5032b9a29235de2a

furushchev, 2015-04-20:
@YoheiKakiuchi にデバッグしてもらいつつgazeboで表示出来ました。
どこにpullreqを送るのが正解でしょうか。
https://bitbucket.org/osrf/gazebo_models
はworldファイルはコミットできない気がします。

mmurooka, 2015-04-20:
結局問題は何でしたか？

furushchev, 2015-04-21:
@mmurooka 結局irteusモデルからmeshファイルへの変換時にworldcoordsがそのまま反映されていたため、meshファイルの時点でオフセットがついてしまっていたところに、さらにworldファイルでオフセットを加えていたのが原因でした。（つまりworldファイルのすべてのposeを0 0 0 0 0 0にすると正常に表示される）
そのため
https://gist.github.com/furushchev/bd5bd4a9366216cb86b1#file-drc-irtmodel-to-urdf-l-L96-L99
のように一旦変換するすべてのeusオブジェクトのcoordsをresetして変換してから、そのcoordsを戻してworldファイルを生成するようにしました。これでオブジェクト単体の移動なども行いやすいはずです。

mmurooka, 2015-04-21:
なるほど。以下のところもそのように修正しておいてもらえると、今後使う人が同じところにハマらないでしょうか。
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_gazebo_tutorials/euslisp/hrpsys-gazebo-utils.l#L69-L73

@furushchev  のGistにあるdrc-irtmodel-to-urdf.lも、
今後Testbed以外でまた使いそうなところは、
このファイルかどこかにマージしてもらえるとよさそうです。

k-okada, 2015-04-21:
:100:  for @mmurooka 's comment

snozawa, 2016-04-08:
表記の問題はなおってるのでcloseします

closed 2016-04-08 by snozawa