start-jsk/rtmros_tutorials #355: JAXON_RED cannot generate urdf model in 32bit?

garaemon, 2015-07-20:
in 32bit, we cannot generate JAXON_RED_WH_SENSORS.urdf

```
[hrpsys_ros_bridge_tutorials] Collada Info: joint type free not supported                                                  
[hrpsys_ros_bridge_tutorials] [ 90%] Generating /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridg
e_tutorials/models/JAXON_WH.urdf 
[hrpsys_ros_bridge_tutorials] ;                                                                                            
[hrpsys_ros_bridge_tutorials]       0/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]     910/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    1820/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    2730/  16471[ 90%] Generating /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/h
rpsys_ros_bridge_tutorials/models/JAXON_WH_SENSORS.urdf 
[hrpsys_ros_bridge_tutorials] ;                                                                                            
[hrpsys_ros_bridge_tutorials]    3640/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    4550/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    5460/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    6370/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    7280/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]    8190/  16471[ 90%] Built target JAXON_WH_SENSORS.urdf_generate                            
[hrpsys_ros_bridge_tutorials] ;                                                                                            
[hrpsys_ros_bridge_tutorials]    9100/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]   10010/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]   10920/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]   11830/  16471terminate called after throwing an instance of 'std::bad_alloc'               
[hrpsys_ros_bridge_tutorials]   what():  std::bad_alloc                                                                    
[hrpsys_ros_bridge_tutorials] ;                                                                                            
[hrpsys_ros_bridge_tutorials]   12740/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]   13650/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]   14560/  16471;                                                                             [hrpsys_ros_bridge_tutorials]   15470/  16471;                                                                             
[hrpsys_ros_bridge_tutorials]   16380/  16471                                                                              
[hrpsys_ros_bridge_tutorials] [ 91%] Generating staro_joint_minmax_conf_done                                               
[hrpsys_ros_bridge_tutorials] make[2]: *** [/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.dae]  () 
[hrpsys_ros_bridge_tutorials] make[1]: *** [CMakeFiles/jaxon_red_hrpsys_ros_bridge_tutorials_compile_dae.dir/all]  2       [hrpsys_ros_bridge_tutorials] make[1]: *** ....                                                                            
[hrpsys_ros_bridge_tutorials] Xserver connection failed[ 91%] Built target staro_hrpsys_ros_bridge_tutorials_compile_joint_minmax 
```

garaemon, 2015-07-20:
not urdf model but collada model

garaemon, 2015-07-20:
gdb
```
gdb --args /home/leus/ros/hydro_parent/devel/bin/openhrp-export-collada -i /home/leus/ros/hydro/src/rtm-ros-roboti
cs/rtmros_hrp2/jsk_models/JAXON_RED/JAXON_REDmain.wrl -o /home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.dae
```

```
            Joint node RLEG_JOINT5
              Segment node RLEG_LINK5
                ForceSensorrfsensor
Collada Info: init COLLADA writer version: 1.5.0, namespace: http://www.collada.org/2008/03/COLLADASchema
Collada Info: joint type free not supported
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Program received signal SIGABRT, Aborted.
0xb7fdd424 in __kernel_vsyscall ()
(gdb) where
#0  0xb7fdd424 in __kernel_vsyscall ()
#1  0xb72c3e0f in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xb72c7455 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0xb753713d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#4  0xb7534ed3 in ?? () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#5  0xb7534f0f in std::terminate() () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#6  0xb753505e in __cxa_throw () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#7  0xb753567f in operator new(unsigned int) () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#8  0xb753574b in operator new[](unsigned int) () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#9  0x080ac2cb in ShapeSetInfo_impl::shapes() ()
#10 0x08087558 in ColladaWriter::_WriteLink(BodyInfo_impl*, int, daeSmartRef<daeElement>, daeSmartRef<ColladaDOM150::domNode>, std::string const&, std::map<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > >) ()
#11 0x0808832e in ColladaWriter::_WriteLink(BodyInfo_impl*, int, daeSmartRef<daeElement>, daeSmartRef<ColladaDOM150::domNode>, std::string const&, std::map<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > >) ()
#12 0x0808832e in ColladaWriter::_WriteLink(BodyInfo_impl*, int, daeSmartRef<daeElement>, daeSmartRef<ColladaDOM150::domNode>, std::string const&, std::map<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > >) ()
#13 0x0808832e in ColladaWriter::_WriteLink(BodyInfo_impl*, int, daeSmartRef<daeElement>, daeSmartRef<ColladaDOM150::domNode>, std::string const&, std::map<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > >) ()
#14 0x0808832e in ColladaWriter::_WriteLink(BodyInfo_impl*, int, daeSmartRef<daeElement>, daeSmartRef<ColladaDOM150::domNode>, std::string const&, std::map<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > >) ()
#15 0x0808832e in ColladaWriter::_WriteLink(BodyInfo_impl*, int, daeSmartRef<daeElement>, daeSmartRef<ColladaDOM150::domNode>, std::string const&, std::map<std::string, int, std::less<std::string>, std::allocator<std::pair<std::string const, int> > >) ()
#16 0x0808a2bd in ColladaWriter::WriteKinematics_model(BodyInfo_impl*) ()
#17 0x0808ffb1 in ColladaWriter::_WriteInstance_kinematics_model(BodyInfo_impl*, daeSmartRef<daeElement>, std::string const&) ()
#18 0x08091e37 in ColladaWriter::_WriteRobot(BodyInfo_impl*) ()
#19 0x08079f0b in ColladaWriter::Write(BodyInfo_impl*) ()
#20 0x0806aa2d in main ()

```

garaemon, 2015-07-20:
hmm, need to recompile openhrp3?

garaemon, 2015-07-20:
JAXON.dae is successfully converted

garaemon, 2015-07-20:
compiled with "-g"
```
(gdb) where
#0  0xb7fdd424 in __kernel_vsyscall ()
#1  0xb72c3e0f in raise () from /lib/i386-linux-gnu/libc.so.6
#2  0xb72c7455 in abort () from /lib/i386-linux-gnu/libc.so.6
#3  0xb753713d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#4  0xb7534ed3 in ?? () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#5  0xb7534f0f in std::terminate() () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#6  0xb753505e in __cxa_throw () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#7  0xb753567f in operator new(unsigned int) () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#8  0xb753574b in operator new[](unsigned int) () from /usr/lib/i386-linux-gnu/libstdc++.so.6
#9  0x080ac2cb in allocbuf (nelems=<optimized out>) at /usr/include/omniORB4/seqTemplatedecls.h:171
#10 copybuffer (newmax=<optimized out>, this=<optimized out>) at /usr/include/omniORB4/seqTemplatedecls.h:288
#11 length (len=8262, this=0xbfaae50c) at /usr/include/omniORB4/seqTemplatedecls.h:155
#12 operator= (this=0xbfaae50c, s=...) at /usr/include/omniORB4/seqTemplatedecls.h:260
#13 operator= (s=..., this=0xbfaae50c) at /usr/include/omniORB4/seqTemplatedecls.h:332
#14 operator= (s=..., this=0xbfaae50c) at /usr/include/omniORB4/seqTemplatedecls.h:576
#15 operator= (_s=..., this=0xbfaae50c) at /home/leus/ros/hydro_parent/build/openhrp3/hrplib/hrpCorba/OpenHRPCommon.hh:78
#16 operator= (this=0xbfaae4f4) at /home/leus/ros/hydro_parent/build/openhrp3/hrplib/hrpCorba/ModelLoader.hh:1057
#17 _CORBA_Sequence (s=..., this=0xbfaacc80) at /usr/include/omniORB4/seqTemplatedecls.h:236
#18 _CORBA_Unbounded_Sequence (s=..., this=0xbfaacc80) at /usr/include/omniORB4/seqTemplatedecls.h:325
#19 ShapeInfoSequence (_s=..., this=0xbfaacc80) at /home/leus/ros/hydro_parent/build/openhrp3/hrplib/hrpCorba/ModelLoader.hh:1090
#20 ShapeSetInfo_impl::shapes (this=0xbfffaa04) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ShapeSetInfo_impl.cpp:89
#21 0x08087558 in ColladaWriter::_WriteLink (this=0xbfffab84, bodyInfo=0xbfffa978, ilink=5, pkinparent=<error reading variable: Cannot access memory at address 0x4>, pnodeparent=..., strModelUri=..., mapjointnames=...)
    at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:1229
#22 0x0808832e in ColladaWriter::_WriteLink (this=0xbfffab84, bodyInfo=0xbfffa978, ilink=4, pkinparent=..., pnodeparent=..., strModelUri=..., mapjointnames=...) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:1277
#23 0x0808832e in ColladaWriter::_WriteLink (this=0xbfffab84, bodyInfo=0xbfffa978, ilink=3, pkinparent=..., pnodeparent=..., strModelUri=..., mapjointnames=...) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:1277
#24 0x0808832e in ColladaWriter::_WriteLink (this=0xbfffab84, bodyInfo=0xbfffa978, ilink=2, pkinparent=..., pnodeparent=..., strModelUri=..., mapjointnames=...) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:1277
#25 0x0808832e in ColladaWriter::_WriteLink (this=0xbfffab84, bodyInfo=0xbfffa978, ilink=1, pkinparent=..., pnodeparent=..., strModelUri=..., mapjointnames=...) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:1277
#26 0x0808832e in ColladaWriter::_WriteLink (this=0xbfffab84, bodyInfo=0xbfffa978, ilink=0, pkinparent=..., pnodeparent=..., strModelUri=..., mapjointnames=...) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:1277
#27 0x0808a2bd in ColladaWriter::WriteKinematics_model (this=0xbfffab84, bodyInfo=0xbfffa978) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:654
#28 0x0808ffb1 in ColladaWriter::_WriteInstance_kinematics_model (this=0xbfffab84, bodyInfo=0xbfffa978, parent=..., sidscope=...) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:467
#29 0x08091e37 in ColladaWriter::_WriteRobot (this=0xbfffab84, bodyInfo=0xbfffa978) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:280
#30 0x08079f0b in ColladaWriter::Write (this=0xbfffab84, bodyInfo=0xbfffa978) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/ColladaWriter.h:242
#31 0x0806aa2d in main (argc=5, argv=0xbfffaf54) at /home/leus/ros/hydro_parent/src/openhrp3/server/ModelLoader/exportCollada.cpp:130
```

garaemon, 2015-07-20:
It happens on jaxonred

garaemon, 2015-07-20:
I found that it crashed in reading/writing HEAD_LINK1.wrl.
If I replace JAXON_RED/HEAD_LINK1.wrl with JAXON/HEAD_LINK1.wrl, it would be no crash but eventually it crashes in LARM_LINK0.wrl.

snozawa, 2015-07-20:
How about reverting JAXON_RED vrml model?

YoheiKakiuchi, 2015-07-20:
How about using environment variable ORBgiopMaxMsgSize=2147483648

garaemon, 2015-07-20:
`export ORBgiopMaxMsgSize=2147483648` didn't work

YoheiKakiuchi, 2015-07-21:
Is this problem of CORBA ?

garaemon, 2015-07-21:
I dont know but it crashes in corba idl object

k-okada, 2015-07-21:
I don't think 
 /home/leus/ros/hydro_parent/devel/bin/openhrp-export-collada 
uses corba communication, maybe wrl is not correct.

garaemon, 2015-07-21:
Not communication but struct is defined via idl?

YoheiKakiuchi, 2015-07-22:
I have no idea for the actual cause.
Reducing corba communication solves this problem....
https://github.com/fkanehiro/openhrp3/pull/63

eisoku9618, 2015-10-20:
https://github.com/fkanehiro/openhrp3/pull/63 seems not to solve this problem.

In JAXON, ``catkin bt --start-with-this --force-cmake -v -p1 -j1`` in hrpsys_ros_bridge fails as follows.

```
Collada Info: init COLLADA writer version: 1.5.0, namespace: http://www.collada.org/2008/03/COLLADASchema
Collada Info: joint type free not supported
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
make[2]: *** [/home/leus/ros/hydro/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/JAXON_RED.dae]  ()
make[1]: *** [CMakeFiles/jaxon_red_hrpsys_ros_bridge_tutorials_compile_dae.dir/all]  2
make: *** [all]  2
```

After I replace JAXON_RED/LARM_LINK0.wrl and JAXON_RED/HEAD_LINK1.wrl with JAXON, the build command succeeded.

YoheiKakiuchi, 2015-10-22:
please check your openhrp3 version, confirm using correct binary.

iory, 2016-04-18:
@eisoku9618 @YoheiKakiuchi `結局どうなりましたか？

YoheiKakiuchi, 2016-04-18:
32ビットOSでは、起こる環境があるように思います。