start-jsk/rtmros_tutorials #356: TorqueFilter が ERROR: parameters are not set. を出力し続ける

eisoku9618, 2015-07-22:
hrpsys-simulatorだと起きない気がしますが，gazeboだとTorqueFilterが https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/Stabilizer/TwoDofController.cpp#L66 のprint文を出力してくることがあります．

snozawa, 2015-07-22:
> 出力してくることがあります

時々起こるのかな？

eisoku9618, 2015-07-22:
> 時々起こるのかな？

足裏の接地がいじできていると出ないみたいですが，かかと側のみの接地になると出るみたいです．

言葉では伝えづらいのですが，以下の動画（左側のターミナルがhrpsysのログ）のような感じです．

https://drive.google.com/open?id=0Bz4F1ykxZ2gjRVczU0t0aFQ5NE0

ですので，gazeboでフラフラしていたら（起動直後とかCtrl-Shift-r 直後とか）必ずエラーの表示がでるのかと思います．

どのタイミングで出るようになった分かっていないのですが，僕と笹渕さんは https://github.com/fkanehiro/hrpsys-base/pull/731 のタイミングで hrpsys_ros_bridgeをmakeしなおしてから出るようになりました．

snozawa, 2015-07-22:
何か所か`parameters are not set`がでるか所がありますが、
どのRTCのどこかは特定できたかな？

@orikuma さん
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceController.cpp#L606
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/ImpedanceController/ImpedanceOutputGenerator.h#L93
みたいに、TorqueControllerなど関係しそうなprint文で、どのRTCからでてるか追加しておいてもらえるでしょうか。

eisoku9618, 2015-07-23:
> 何か所かparameters are not setがでるか所がありますが、
どのRTCのどこかは特定できたかな？

- 実際に出力しているコードはStabilizer/TwoDofController.cpp
- Stabilizer/TwoDofController.cppを呼び出しているのはTorqueControllerしかない

という手順で，print文を出しているのはTorqueControllerだと特定しました．
※ issue のタイトルとかにはTorqueFilterと書いてしまっていましたが，タイポで，TorqueContollerでした．すみません．

orikuma, 2015-07-23:
https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/Stabilizer/Stabilizer.h#L262
でStabilizerもTwoDofControllerを使っていると思います.

eisoku9618, 2015-07-23:
> StabilizerもTwoDofControllerを使っていると思います.

すみません，書き方が悪かったです．

今回の``parameters are not set``を出力している関数は``TwoDofController::update``ですが，STにはその関数を呼び出す行が書かれているけども，使われてなさそう，という意味でした．

https://github.com/fkanehiro/hrpsys-base/blob/master/rtc/Stabilizer/TwoDofController.cpp#L56

orikuma, 2015-07-23:
確かにStabilzer内でcalcRUNST()は他で呼ばれていなさそうですね. 確認不足でした. 申し訳ございません.
tcを消したらエラーは消えるでしょうか? またトルクが過大になっていないでしょうか(具体的にはwrlのclimit*gearRatio*torqueConstを超えていないか)

eisoku9618, 2015-07-24:
> tcを消したらエラーは消えるでしょうか?

tcだけを消したらportの接続がどうこうで上手く起動しなかったので，実機と同様に tf / vs / tc / te / tl を消したら，エラーも消えて月面重力下で平面を歩くことができました．

一方，getRTCListをunstableに戻すと，hrpsysのログに``[tc]TwoDofController parameters are not set.`` が出るようになっていて，tcがなんかしているということがよく分かるようになりました．
ありががとうございます．

closed 2015-07-24 by eisoku9618

k-okada, 2015-07-24:
エラーなどのリミットにかかったりしていたということ？

orikuma, 2015-07-24:
本来tcはdisable時に何もしないべきですがモデルに記載されたclimitを超えるとトルク制限が有効になってしまうようなので直します.

eisoku9618, 2015-07-24:
> トルクが過大になっていないでしょうか

これはhrpsysの標準エラー出力に出るものでしょうか？であれば、``[tc]TwoDofController parameters are not set.``しか出ていなかったので過大になっていなさそうです。

snozawa, 2015-07-24:
ちなみに、今までどおりtcなどをいれた状態で、
https://github.com/fkanehiro/hrpsys-base/pull/737
の変更をくわえたら過大なトルクが治ったりしますでしょうか？

orikuma, 2015-07-24:
https://github.com/fkanehiro/hrpsys-base/pull/739 でエラーが出なくなることを期待しています.

eisoku9618, 2015-07-24:
> ちなみに、今までどおりtcなどをいれた状態で、
fkanehiro/hrpsys-base#737
の変更をくわえたら過大なトルクが治ったりしますでしょうか？

,

> fkanehiro/hrpsys-base#739 でエラーが出なくなることを期待しています.

fkanehiro/hrpsys-base#737 のPRがマージされていて，fkanehiro/hrpsys-base#739 のPRがマージされていない状態のhrpsysで同じ事をやったら，このissueで挙げさせてもらったプリント文は出なくなりました．

snozawa, 2015-07-24:
> fkanehiro/hrpsys-base#737 のPRがマージされていて，fkanehiro/hrpsys-base#739 のPRがマージされていない状態のhrpsysで同じ事をやったら，このissueで挙げさせてもらったプリント文は出なくなりました．

そうすると、
fkanehiro/hrpsys-base#737
が原因だったということかな。

eisoku9618, 2015-07-24:
はい．そうみたいです．

今 fkanehiro/hrpsys-base#737 有り・無しで純粋に比較してみたところ，
このPRがないとプリント文が出て，このPRがあるとプリント文が出なかった，となりましたので，まさにここが原因だったようです．

eisoku9618, 2015-07-24:
すみません，1つ上は嘘でした．

やはりgazeboでボインボインするとプリント文が出るみたいで， fkanehiro/hrpsys-base#737 があると出づらくなるかもしれないけども完全に出なくなるわけではない，というのが正しそうです．

gazeboでstを入れると足裏から体重の何十倍もの力を受けてジャンプする現象が起きますが，そのときにプリント文が出ている気がします．