start-jsk/rtmros_tutorials #412: hrpsys-simulatorでgo-posできない

mizmizo, 2015-10-13:
http://wiki.ros.org/rtmros_common/Tutorials/AddRobotsEnvironmentsHrpsysSimulator
の3.1.2に従ってHRP2JSKのトルクモードでのシミュレーションを試そうとしたところ、
start-auto-balancer,start-stの後go-posしようとするとエラーが出てシミュレートが止まってしまいます。どなたか原因、対処ご存知でしょうか

ログは以下のとおりです

hrpsys-simulator側
```
[abc] start auto balancer mode
[abc]   limb [rleg]
[abc]   limb [lleg]
[ INFO] [1444722614.202279850, 239.245000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1771 Hz, dropped 1001)
[ INFO] [1444722615.202812774, 240.245000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1775 Hz, dropped 1000)
[ INFO] [1444722615.323623849, 240.355000000]: AutoBalancerServiceROSBridge::startAutoBalancer() succeeded
[ INFO] [1444722616.218265199, 241.247000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1802 Hz, dropped 1001)
[INFO] [WallTime: 1444722616.913234] [241.950000] check 143 collision status, 28.469042 Hz
[ INFO] [1444722617.219984406, 242.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1776 Hz, dropped 1001)
[ INFO] [1444722618.220973322, 243.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1778 Hz, dropped 1000)
[ INFO] [1444722619.221775091, 244.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1770 Hz, dropped 1000)
[rtmlaunch] check connection/activation
[ INFO] [1444722620.222229182, 245.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 999[Hz] (exec 1733 Hz, dropped 999)
[ INFO] [1444722621.222976124, 246.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1775 Hz, dropped 1000)
[INFO] [WallTime: 1444722621.934558] [246.969000] check 143 collision status, 28.497409 Hz
[ INFO] [1444722622.223249960, 247.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1774 Hz, dropped 1001)
[ INFO] [1444722623.223695659, 248.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 999[Hz] (exec 1779 Hz, dropped 999)
[ INFO] [1444722624.224409839, 249.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 999[Hz] (exec 1770 Hz, dropped 999)
[ INFO] [1444722624.852055961, 249.879000000]: StabilizerServiceROSBridge::startStabilizer() called
[st] Start ST
[st] Sync IDLE => ST
[ INFO] [1444722625.224936964, 250.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1778 Hz, dropped 1000)
[ INFO] [1444722626.225303166, 251.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1783 Hz, dropped 1000)
[st] Start ST DONE
[ INFO] [1444722626.853275186, 251.879000000]: StabilizerServiceROSBridge::startStabilizer() succeeded
[INFO] [WallTime: 1444722626.958280] [251.986000] check 143 collision status, 28.497409 Hz
[ INFO] [1444722627.226042368, 252.249000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1775 Hz, dropped 1001)
[ INFO] [1444722628.226253511, 253.248000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 998[Hz] (exec 1785 Hz, dropped 998)
[ INFO] [1444722629.226459313, 254.244000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 996[Hz] (exec 1794 Hz, dropped 996)
[ INFO] [1444722630.227693475, 255.246000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1002[Hz] (exec 1773 Hz, dropped 1002)
[rtmlaunch] check connection/activation
[ INFO] [1444722631.229272328, 256.247000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1002[Hz] (exec 1758 Hz, dropped 1002)
[INFO] [WallTime: 1444722632.004795] [257.010000] check 143 collision status, 28.463376 Hz
[ INFO] [1444722632.229489918, 257.244000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 996[Hz] (exec 1769 Hz, dropped 996)
[ INFO] [1444722633.229931618, 258.245000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1775 Hz, dropped 1001)
[ INFO] [1444722634.230475765, 259.241000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 995[Hz] (exec 1759 Hz, dropped 995)
[ INFO] [1444722635.230689308, 260.241000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1785 Hz, dropped 1000)
[ INFO] [1444722636.231155906, 261.242000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1792 Hz, dropped 1001)
[INFO] [WallTime: 1444722637.025111] [262.026000] check 143 collision status, 28.508772 Hz
[ INFO] [1444722637.231917135, 262.242000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1000[Hz] (exec 1788 Hz, dropped 1000)
[ INFO] [1444722638.232729971, 263.243000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1791 Hz, dropped 1001)
[ INFO] [1444722639.233057857, 264.244000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1777 Hz, dropped 1001)
[ INFO] [1444722640.233559702, 265.246000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1002[Hz] (exec 1774 Hz, dropped 1002)
[ INFO] [1444722641.234303294, 266.244000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 996[Hz] (exec 1789 Hz, dropped 996)
[rtmlaunch] check connection/activation
[INFO] [WallTime: 1444722642.051686] [267.047000] check 143 collision status, 28.480382 Hz
[ INFO] [1444722642.234562077, 267.241000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 998[Hz] (exec 1758 Hz, dropped 998)
[ INFO] [1444722643.234875333, 268.243000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1740 Hz, dropped 1001)
[ INFO] [1444722644.235799791, 269.244000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 1001[Hz] (exec 1783 Hz, dropped 1001)
[ INFO] [1444722644.733304275, 269.733000000]: AutoBalancerServiceROSBridge::goPos() called
start ref coords
  pos =
    [1.88738e-17,  0,  7.63278e-17]
  rot =
    [           1,  1.31655e-36, -5.55112e-17]
    [-1.50463e-36,            1,  3.00927e-36]
    [ 5.55112e-17,  3.00927e-36,            1]
goal ref midcoords
  pos =
    [0.1,  -1.50463e-37,  8.18789e-17]
  rot =
    [           1,  1.41059e-36, -5.55112e-17]
[... 319 lines cut ...]
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LARM_LINK7" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LLEG_LINK0" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LLEG_LINK1" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LLEG_LINK2" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LLEG_LINK3" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LLEG_LINK4" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "LLEG_LINK5" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "RARM_LINK0" from authority "/hrpsys_state_publisher" because of a nan value\
 in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
         at line 240 in /tmp/buildd/ros-hydro-tf2-0.4.12-0precise-20150327-1914/src/buffer_core.cpp
```

garaemon, 2015-10-13:
joint_statesにnanがはいってないかな

snozawa, 2015-10-13:
`servo error limit`にかかっていて、サーボOFFを模擬するモードに移行してるのかもしれません。
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/util/BodyRTC.cpp#L507

```
(send *ri* :set-servo-error-limit "all" 1) ;; [rad]
```
くらいの極端におおきい値になると改善しますか？

mizmizo, 2015-10-13:
>>joint_statesにnanがはいってないかな

reset-pose,auto-balancerとstが入った状態でrostopic echo /joint_states
してみました。nanは入ってないみたいです。

```
---
header: 
  seq: 733337
  stamp: 
    secs: 753
    nsecs: 7999989
  frame_id: ''
name: ['RLEG_JOINT0', 'RLEG_JOINT1', 'RLEG_JOINT2', 'RLEG_JOINT3', 'RLEG_JOINT4', 'RLEG_JOINT5', 'LLEG_JOINT0', 'LLEG_JOINT1', 'LLEG_JOINT2', 'LLEG_JOINT3', 'LLEG_JOINT4', 'LLEG_JOINT5', 'CHEST_JOINT0', 'CHEST_JOINT1', 'HEAD_JOINT0', 'HEAD_JOINT1', 'RARM_JOINT0', 'RARM_JOINT1', 'RARM_JOINT2', 'RARM_JOINT3', 'RARM_JOINT4', 'RARM_JOINT5', 'RARM_JOINT6', 'RARM_JOINT7', 'LARM_JOINT0', 'LARM_JOINT1', 'LARM_JOINT2', 'LARM_JOINT3', 'LARM_JOINT4', 'LARM_JOINT5', 'LARM_JOINT6', 'LARM_JOINT7']
position: [2.8319112441270673e-06, 0.0028453726381484075, -0.4486649789343179, 0.8752589142742718, -0.42665004553926617, -0.0023288098027394438, 4.441439396596889e-06, 0.00411277852394104, -0.4477636527790263, 0.8734680397001778, -0.4256197909412924, -0.004244665340336738, -4.602346509336594e-09, -0.00014633150200511964, 3.290598317733234e-10, 4.0713843496444154e-06, 0.17451911777987136, -0.17440264857247037, 2.0756227069649788e-05, -0.4362619431932406, 5.550862971846371e-07, 8.624640959307225e-06, -0.17451696889798937, 0.2617993156131678, 0.17451912844611878, 0.1744027869449095, -2.0731524927866286e-05, -0.4362619371096362, -5.53536519427978e-07, -8.615728292571046e-06, -0.17451696792603413, -0.2617993153478459]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [-0.2272217511446583, -158.36342066259118, -7.192708868188902, -11.148902210916798, -5.457967088599025, 944.129323902846, 0.23330702438759632, 156.66071442744067, -4.810477280997039, -15.960761031931225, -4.5423903764146845, -939.4388903989606, -0.008388805548556715, 2.926172519184008, -4.343892130393678e-05, -0.0814500883291069, 0.27891325969475744, -2.6064617446319716, -0.41544337049739877, -1.4064350363729439, -0.011136382216226178, -0.17259872179041322, -0.31902225633678727, 0.0014403631154458196, 0.2733213893882369, 2.601690024984782, 0.4142812294970542, -1.4083412^C363419257, 0.011033978232159021, 0.17218457782693572, -0.3192027326578972, -0.0014521271357592624]
---
```

>>servo error limitにかかっていて、サーボOFFを模擬するモードに移行してるのかもしれません。
https://github.com/fkanehiro/hrpsys-base/blob/master/lib/util/BodyRTC.cpp#L507
(send *ri* :set-servo-error-limit "all" 1) ;; [rad]
くらいの極端におおきい値になると改善しますか？

```
2.irteusgl$ (send *ri* :set-servo-error-limit "all" 1)
[ERROR] [1444725485.333457868, 306.705999999]: attempted to call service  /RobotHardw\
areServiceROSBridge/setServoErrorLimit, but failed 
#<hrpsys_ros_bridge::openhrp_robothardwareservice_setservoerrorlimitresponse #Xb0de94\
8>
```
のようになってしまいました。どのプログラムを立ち上げればよいでしょうか

garaemon, 2015-10-13:
なるほど、後半のtfにNANが入っているというメッセージは、おそらくautobalancerの位置がnanを含むものになっていそう。
これに関しては`rostopic echo /odom`で確認できるはず

mizmizo, 2015-10-13:
rostopic echo /odomの結果です
```
---
header: 
  seq: 347879
  stamp: 
    secs: 1402
    nsecs: 678999973
  frame_id: odom
child_frame_id: BODY
pose: 
  pose: 
    position: 
      x: -0.00573303728768
      y: -0.00180673064218
      z: 0.658701851183
    orientation: 
      x: 1.60936259978e-20
      y: 2.77555756156e-17
      z: 2.82118644197e-37
      w: 1.0
  covariance: [5.566079999751641e-06, 8.440430350014493e-58, 0.0, 0.0, 0.0, 0.0, 8.440430350014493e-58, 5.566079999751641e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 5.566079999751641e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 5.566079999751641e-06, 8.440430350014493e-58, 0.0, 0.0, 0.0, 0.0, 8.440430350014493e-58, 5.566079999751641e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 5.566079999751641e-06]
twist: 
  twist: 
    linear: 
      x: 1.08420217249e-15
      y: 2.16840434497e-16
      z: 6.01783312644e-32
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1e-06]
---
```

garaemon, 2015-10-13:
あれ、出ていない。おかしいなぁ。

garaemon, 2015-10-13:
メッセージをよく見ると、imuのようでした。

mizmizo, 2015-10-13:
imuでした。

auto-balancer,stまで入れた状態で```rostopic echo /imu```すると
```
---
header: 
  seq: 52089
  stamp: 
    secs: 67
    nsecs: 410999999
  frame_id: gyrometer
orientation: 
  x: -7.00718102128e-05
  y: -0.000193802506089
  z: 1.47002581975e-05
  w: 0.999999978657
orientation_covariance: [2.89e-08, 0.0, 0.0, 0.0, 2.89e-08, 0.0, 0.0, 0.0, 2.89e-08]
angular_velocity: 
  x: 3.49306762933e-05
  y: 9.70119745503e-06
  z: -3.16058204864e-05
angular_velocity_covariance: [0.000144, 0.0, 0.0, 0.0, 0.000144, 0.0, 0.0, 0.0, 0.000144]
linear_acceleration: 
  x: 0.00414941654885
  y: -0.000886886860095
  z: 9.80670915605
linear_acceleration_covariance: [0.0096, 0.0, 0.0, 0.0, 0.0096, 0.0, 0.0, 0.0, 0.0096]
---
```
ですが、go-posを送った直後に
```
---
header: 
  seq: 228047
  stamp: 
    secs: 252
    nsecs: 223000000
  frame_id: gyrometer
orientation: 
  x: nan
  y: nan
  z: nan
  w: nan
orientation_covariance: [2.89e-08, 0.0, 0.0, 0.0, 2.89e-08, 0.0, 0.0, 0.0, 2.89e-08]
angular_velocity: 
  x: nan
  y: nan
  z: nan
angular_velocity_covariance: [0.000144, 0.0, 0.0, 0.0, 0.000144, 0.0, 0.0, 0.0, 0.000144]
linear_acceleration: 
  x: nan
  y: nan
  z: nan
linear_acceleration_covariance: [0.0096, 0.0, 0.0, 0.0, 0.0096, 0.0, 0.0, 0.0, 0.0096]
---
```
となってnanが入っています

mizmizo, 2015-10-13:
stなしでgo-posしてみても以下のようなエラーが出て失敗しました。
```
[ INFO] [1444728058.554300343, 56.858999999]: AutoBalancerServiceROSBridge::goPos() called
start ref coords
  pos =
    [0,  0,  0]
  rot =
    [1, 0, 0]
    [0, 1, 0]
    [0, 0, 1]
goal ref midcoords
  pos =
    [0.1,  0,  0]
  rot =
    [1, 0, 0]
    [0, 1, 0]
    [0, 0, 1]
[0] footstep
  name = [lleg]
  pos = [0,  0,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[1] footstep
  name = [rleg]
  pos = [0.1,  -0.21,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0.05[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[2] footstep
  name = [lleg]
  pos = [0.1,  0.05,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0.05[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[3] footstep
  name = [rleg]
  pos = [0.1,  -0.135,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0.05[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[4] footstep
  name = [lleg]
  pos = [0.1,  0.105,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0.05[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[5] footstep
  name = [rleg]
  pos = [0.1,  -0.105,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0.05[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[6] footstep
  name = [lleg]
  pos = [0.1,  0.105,  0]
  rot = [1, 0, 0] [0, 1, 0] [0, 0, 1]
  step_height = 0[m], step_time = 1.1[s], toe_angle = 0[deg], heel_angle = 0[deg]
[abc] start auto balancer mode
[abc]   limb [rleg]
[abc]   limb [lleg]
[ INFO] [1444728059.490968247, 57.799999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 999[Hz] (exec 1746 Hz, dropped 999)
[ INFO] [1444728060.491739083, 58.799999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 997[Hz] (exec 1735 Hz, dropped 996)
[ INFO] [1444728060.560290860, 58.863999999]: AutoBalancerServiceROSBridge::goPos() succeeded
[ INFO] [1444728060.561598457, 58.865999999]: AutoBalancerServiceROSBridge::waitFootSteps() called
[ INFO] [1444728061.492214830, 59.799999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 998[Hz] (exec 1778 Hz, dropped 998)
18:21:01.990062: servo error limit over: joint = RLEG_JOINT0, qRef = -0.226173[deg], q = -5684.07[deg]
error limit over RLEG_JOINT1(1), qRef=-0.154998, qCurrent=214.956 , Error=-215.111 > 0.18 (limit), servo_state = ON, q=-0.154998
error limit over RLEG_JOINT2(2), qRef=-0.477195, qCurrent=-237.54 , Error=237.063 > 0.18 (limit), servo_state = ON, q=-0.477195
error limit over RLEG_JOINT3(3), qRef=0.91385, qCurrent=-292.886 , Error=293.799 > 0.18 (limit), servo_state = ON, q=0.91385
error limit over RLEG_JOINT4(4), qRef=-0.436654, qCurrent=-348.306 , Error=347.869 > 0.18 (limit), servo_state = ON, q=-0.436654
error limit over RLEG_JOINT5(5), qRef=0.154997, qCurrent=337.706 , Error=-337.551 > 0.18 (limit), servo_state = ON, q=0.154997
error limit over LLEG_JOINT0(6), qRef=-7.44327e-09, qCurrent=40.8328 , Error=-40.8328 > 0.18 (limit), servo_state = ON, q=-7.44327e-09
error limit over LLEG_JOINT1(7), qRef=-0.135144, qCurrent=-119.893 , Error=119.758 > 0.18 (limit), servo_state = ON, q=-0.135144
error limit over LLEG_JOINT2(8), qRef=-0.439867, qCurrent=163.62 , Error=-164.06 > 0.18 (limit), servo_state = ON, q=-0.439867
error limit over LLEG_JOINT3(9), qRef=0.871352, qCurrent=169.527 , Error=-168.655 > 0.18 (limit), servo_state = ON, q=0.871352
[... 156 lines cut ...]
[ERROR] [1444728062.501877019, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK0" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.501960021, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK1" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.501974591, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK2" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.501990931, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK3" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.502011635, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK4" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.502030895, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK5" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.502051641, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK6" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.502070567, 60.322999999]: Ignoring transform for child_frame_id "RARM_LINK7" from authority "/hrpsys_state_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
[ERROR] [1444728062.502090611, 60.322999999]: Ignoring transform for child_frame_id "RLEG_LINK0" from authority "/hrpsys_state_publisher" because of a nan value in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
[... cut ...]
[ERROR] [1444728062.590078878, 60.326999999]: [HrpsysSeqStateROSBridge0] nan value detected in imu_floor! (input: r,p,y=nan,nan,nan)
[ERROR] [1444728062.906275384, 60.338999999]: [HrpsysSeqStateROSBridge0] nan value detected in imu_floor! (input: r,p,y=nan,nan,nan)
[ERROR] [1444728063.011383923, 60.342999999]: [HrpsysSeqStateROSBridge0] nan value detected in imu_floor! (input: r,p,y=nan,nan,nan)
```

また、
```
export BUSH_CUSTOMIZER_CONFIG_PATH=`rospack find hrpsys_ros_bridge_tutorials`/models/HRP2JSK_bush_customizer_param.conf
```
をせずにhrpsys-simulatorを立ち上げた場合は一応go-posを送ることができました

snozawa, 2016-04-08:
治ったようなのでcloseします。
何か問題あればopenしてください。

closed 2016-04-08 by snozawa