start-jsk/rtmros_tutorials #422: jaxonのhandのangle-vectorを送っているメソッドはどれでしょうか？

snozawa, 2015-10-27:
jaxonのhandのangle-vectorを送っているメソッドはどれでしょうか？
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/euslisp/jaxon-interface.l#L31
これ？

mmurooka, 2015-10-27:
はい，そこです．
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/euslisp/jaxon-interface.l#L14 で
https://github.com/jsk-ros-pkg/trans_system/blob/master/thk_hand_controller/euslisp/thkhand-simple-controller.l#L52 にいきます．

snozawa, 2015-10-27:
基本は片手につき1個のスカラーをsendするイメーじでいいかな？

mmurooka, 2015-10-27:
はい，1が開く，0が閉じるです．

https://github.com/jsk-ros-pkg/trans_system/blob/master/thk_hand_controller/euslisp/thkhand-simple-controller.l#L75

https://github.com/jsk-ros-pkg/trans_system/blob/master/thk_hand_controller/euslisp/thkhand-simple-controller.l#L79

snozawa, 2015-10-27:
関節角度的なものはあるんだっけ？

mmurooka, 2015-10-27:
> はい，1が開く，0が閉じるです．

0.5が半開きとなっていて，[0:1]の連続な数値が与えられます．

現在の関節角は，
https://github.com/jsk-ros-pkg/trans_system/blob/master/thk_hand_controller/euslisp/thkhand-simple-controller.l#L119
でとれます．
これも（確か），[0:1]の間の値が返って来ます．

snozawa, 2015-10-27:
了解です、ありがとうございます。

closed 2015-10-27 by snozawa

mmurooka, 2015-10-27:
https://github.com/jsk-ros-pkg/trans_system/pull/525 でREADMEを追加しました．