start-jsk/rtmros_tutorials #464: CMake Error: Could not find "hrpsys_gazebo_general"

KahnShi, 2016-03-17:
Dear all,

My ROS version is Indigo and Gazebo 2.0.
At first, I install rtmros-common, then I try to install rtmros_tutorial.
Everything goes well untill this procedure:
`catkin_make --only-pkg-with-deps hrpsys_gazebo_tutorials`
And the output about error is as follow:
```
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "hrpsys_gazebo_general" with any of the following names:

    hrpsys_gazebo_generalConfig.cmake
    hrpsys_gazebo_general-config.cmake

  Add the installation prefix of "hrpsys_gazebo_general" to CMAKE_PREFIX_PATH
  or set "hrpsys_gazebo_general_DIR" to a directory containing one of the
  above files.  If "hrpsys_gazebo_general" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/kahn/ros/ws_rtmros_tutorials/build/CMakeFiles/CMakeOutput.log".
See also "/home/kahn/ros/ws_rtmros_tutorials/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

Could anyone help me?
Thanks in advance!

And the whole output is:
```
kahn@kahn:~/ros/ws_rtmros_tutorials$ catkin_make --only-pkg-with-deps hrpsys_gazebo_tutorials
Base path: /home/kahn/ros/ws_rtmros_tutorials
Source space: /home/kahn/ros/ws_rtmros_tutorials/src
Build space: /home/kahn/ros/ws_rtmros_tutorials/build
Devel space: /home/kahn/ros/ws_rtmros_tutorials/devel
Install space: /home/kahn/ros/ws_rtmros_tutorials/install
Whitelisted packages: hrpsys_gazebo_tutorials, hrpsys_ros_bridge_tutorials
####
#### Running command: "cmake /home/kahn/ros/ws_rtmros_tutorials/src -DCATKIN_DEVEL_PREFIX=/home/kahn/ros/ws_rtmros_tutorials/devel -DCMAKE_INSTALL_PREFIX=/home/kahn/ros/ws_rtmros_tutorials/install -DCATKIN_WHITELIST_PACKAGES=hrpsys_gazebo_tutorials;hrpsys_ros_bridge_tutorials -G Unix Makefiles" in "/home/kahn/ros/ws_rtmros_tutorials/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/kahn/ros/ws_rtmros_tutorials/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/kahn/ros/ws_rtmros_tutorials/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: hrpsys_gazebo_tutorials;hrpsys_ros_bridge_tutorials
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - hrpsys_ros_bridge_tutorials
-- ~~  - hrpsys_gazebo_tutorials
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'hrpsys_ros_bridge_tutorials'
-- ==> add_subdirectory(rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials)
-- Using these message generators: gencpp;geneus;genlisp;genpy
-- set EUSDIR to /opt/ros/indigo/share/euslisp/jskeus/eus
-- set ARCHDIR to Linux64
-- [roseus.cmake] loading roseus.cmake from hrpsys_ros_bridge_tutorials
-- [roseus.cmake] roseus 1.4.1
-- [roseus.cmake] jskeus 
[... 95 lines cut ...]
hrpsys_ros_bridge_PACKAGE_PATH = /opt/ros/indigo/share/hrpsys_ros_bridge
-- Output /home/kahn/ros/ws_rtmros_tutorials/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/build/samplerobot-interface.l
-- +++ processing catkin package: 'hrpsys_gazebo_tutorials'
-- ==> add_subdirectory(rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials)
-- Using these message generators: gencpp;geneus;genlisp;genpy
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "hrpsys_gazebo_general" with any of the following names:

    hrpsys_gazebo_generalConfig.cmake
    hrpsys_gazebo_general-config.cmake

  Add the installation prefix of "hrpsys_gazebo_general" to CMAKE_PREFIX_PATH
  or set "hrpsys_gazebo_general_DIR" to a directory containing one of the
  above files.  If "hrpsys_gazebo_general" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/kahn/ros/ws_rtmros_tutorials/build/CMakeFiles/CMakeOutput.log".
See also "/home/kahn/ros/ws_rtmros_tutorials/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

```

mmurooka, 2016-03-17:
@k-okada ,
ros-indigo-hrpsys-gazebo-general is not found in apt package now.

hrpsys_gazebo_general is found in http://www.ros.org/debbuild/hydro.html, but not found in http://repositories.ros.org/status_page/ros_indigo_default.html.

k-okada, 2016-03-18:
at
https://github.com/start-jsk/rtmros_gazebo/issues/129#issuecomment-59895435,
you said "complete debiznize"....

also we can find them at https://github.com/ros/rosdistro/pull/8742 and
http://www.ros.org/debbuild/hydro.html?q=rtmros_gazebo

please check again...

mmurooka, 2016-03-18:
Yes, in "hydro", it is released and I can find it.
In "indigo", I can not find apt package.
Do we need some procedure to release in indigo?

k-okada, 2016-03-18:
yes, basically, we just run `bloom-release rtmros_gazebo --rosdistro indigo
--track indigo` and hit return key many times, released on ->
https://github.com/ros/rosdistro/pull/10758

It will take some time to get deb package, so indigo users have to compile
from sources.

k-okada, 2016-03-18:
@mmurooka please watch this ->
http://build.ros.org/job/Ibin_uT64__hrpsys_gazebo_general__ubuntu_trusty_amd64__binary/1/

If this continue failing , please look inside

2016年3月18日金曜日、Kei Okada<k-okada@jsk.t.u-tokyo.ac.jp>さんは書きました:

> yes, basically, we just run `bloom-release rtmros_gazebo --rosdistro
> indigo --track indigo` and hit return key many times, released on ->
> https://github.com/ros/rosdistro/pull/10758
>
> It will take some time to get deb package, so indigo users have to compile
> from sources.
>
> ◉ Kei Okada
>
>
>
>> Yes, in "hydro", it is released and I can find it.
>> In "indigo", I can not find apt package.
>> Do we need some procedure to release in indigo?