start-jsk/rtmros_tutorials #465: hrp2jsknts.launchが途中で落ちる

iory, 2016-04-11:
ubuntu14.04(indigo)環境で
```
rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts.launch
```
を実行すると，以下のようなエラーがでます．

```
 rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts.launch
[rtmlaunch] Start omniNames at port 15005 

Mon Apr 11 15:40:58 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003133332e31312e3231362e31343900009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000fa460b5701007f73
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iory/.ros/log/599109c4-ffb0-11e5-a082-a434d9a10e46/roslaunch-crab-32621.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.149:33234/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_description: <?xml version="1....
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    AutoBalancerServiceROSBridge (hrpsys_ros_bridge/AutoBalancerServiceROSBridgeComp)
    CollisionDetectorServiceROSBridge (hrpsys_ros_bridge/CollisionDetectorServiceROSBridgeComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    EmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
    HardEmergencyStopperServiceROSBridge (hrpsys_ros_bridge/EmergencyStopperServiceROSBridgeComp)
    HrpsysJointTrajectoryBridge (hrpsys_ros_bridge/HrpsysJointTrajectoryBridge)
    HrpsysSeqStateROSBridge (hrpsys_ros_bridge/HrpsysSeqStateROSBridge)
    ImpedanceControllerServiceROSBridge (hrpsys_ros_bridge/ImpedanceControllerServiceROSBridgeComp)
[... 711 lines cut ...]
[ INFO] [1460356898.945350214, 32.584000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1643 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   21 /30
[ INFO] [1460356899.947940194, 33.592000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 252[Hz] (exec 1627 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   22 /30
[ INFO] [1460356900.949795015, 34.596000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1655 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   23 /30
[ INFO] [1460356901.950534157, 35.600000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1628 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   24 /30
[ INFO] [1460356902.955514763, 36.608000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 252[Hz] (exec 1669 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   25 /30
[ INFO] [1460356903.958875702, 37.612000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1612 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   26 /30
[ INFO] [1460356904.962772511, 38.620000000]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 252[Hz] (exec 1635 Hz, dropped 0)
[ INFO] [1460356905.966841955, 39.623999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1656 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   27 /30
[ INFO] [1460356906.970799255, 40.631999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1674 Hz, dropped 1)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   28 /30
[ INFO] [1460356907.979377657, 41.639999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 252[Hz] (exec 1765 Hz, dropped 0)
[rtmlaunch] Wait for  /localhost:15005/sh.rtc:StateHolderService   29 /30
[ INFO] [1460356908.982438538, 42.647999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 252[Hz] (exec 1741 Hz, dropped 0)
[rtmlaunch] [ERROR] Could not Connect ( /localhost:15005/StateHolderServiceROSBridge.rtc:StateHolderService , /localhost:15005/sh.rtc:StateHolderService ):  No such object: /localhost:15005/sh.rtc:StateHolderService 
[ INFO] [1460356909.984268610, 43.655999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 252[Hz] (exec 1745 Hz, dropped 0)
[ INFO] [1460356910.987608613, 44.659999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1694 Hz, dropped 0)
[ INFO] [1460356911.988328466, 45.663999999]: [HrpsysSeqStateROSBridge0] @onExecutece 0 is working at 251[Hz] (exec 1628 Hz, dropped 0)
```

snozawa, 2016-04-11:
rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts_startup.launch
とするとhrpsys関係のみ起動します（rosbridgeなし）
これで起動して、ログをおくってもらえますか。

iory, 2016-04-11:
はい，以下になります．
```
 rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts_startup.launch
[rtmlaunch] Start omniNames at port 15005 

Mon Apr 11 16:10:21 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003133332e31312e3231362e31373400009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000dd4d0b5701007f0d
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iory/.ros/log/7450b30a-ffb4-11e5-983d-507b9d9efada/roslaunch-crab-32519.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.174:34576/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    hrpsys (hrpsys/hrpsys-simulator)
    hrpsys_py (hrpsys_ros_bridge_tutorials/hrp2jsknts_hrpsys_config.py)
    modelloader (openhrp3/openhrp-model-loader)

auto-starting new master
process[master]: started with pid [32534]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7450b30a-ffb4-11e5-983d-507b9d9efada
process[rosout-1]: started with pid [32550]
started core service [/rosout]
process[modelloader-2]: started with pid [32554]
ready
process[hrpsys-3]: started with pid [32580]
process[hrpsys_py-4]: started with pid [32581]
loading file:///home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2JSKNTS_for_OpenHRP3/HRP2JSKNTSmain.wrl
configuration ORB with localhost:15005
[hrpsys.py] waiting ModelLoader
[hrpsys.py] start hrpsys
[hrpsys.py] finding RTCManager and RobotHardware
Humanoid node
Joint node WAIST
  Segment node BODY
  Joint node RLEG_JOINT0
      Segment node RLEG_LINK0
      Joint node RLEG_JOINT1
          Segment node RLEG_LINK1
          Joint node RLEG_JOINT2
              Segment node RLEG_LINK2
              Joint node RLEG_JOINT3
                  Segment node RLEG_LINK3
                  Joint node RLEG_JOINT4
                      Segment node RLEG_LINK4
                      Joint node RLEG_JOINT5
                          Segment node RLEG_LINK5
                            ForceSensorrfsensor
                          Joint node RLEG_JOINT6
                            Segment node RLEG_LINK6
  Joint node LLEG_JOINT0
[... 263 lines cut ...]
    HrpsysConfigurator.init(self, robotname, url)
  File "/home/iory/catkin_ws/hrp2/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 2083, in init
    self.connectComps()
  File "/home/iory/catkin_ws/hrp2/devel/lib/python2.7/dist-packages/hrpsys/hrpsys_config.py", line 289, in connectComps
    connectPorts(self.sh.port("qOut"), tmp_controllers[0].port("qRef"))
  File "/home/iory/catkin_ws/hrp2/devel/lib/python2.7/dist-packages/hrpsys/rtm.py", line 541, in connectPorts
    if isConnected(outP, inP) == True and False:
  File "/home/iory/catkin_ws/hrp2/devel/lib/python2.7/dist-packages/hrpsys/rtm.py", line 480, in isConnected
    op = outP.get_port_profile()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/OpenRTM_aist/RTM_IDL/RTC_idl.py", line 1205, in get_port_profile
    return _omnipy.invoke(self, "get_port_profile", _0_RTC.PortService._d_get_port_profile, args)
omniORB.CORBA.COMM_FAILURE: CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE)
[hrpsys-3] process has died [pid 32580, exit code -6, cmd /home/iory/catkin_ws/hrp2/devel/lib/hrpsys/hrpsys-simulator /home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.xml -o manager.is_master:YES -o corba.nameservers:localhost:15005 -o naming.formats:%n.rtc -o logger.file_name:/tmp/rtc%p.log -endless -realtime -o exec_cxt.periodic.rate:1000000 -o manager.shutdown_onrtcs:NO -o manager.modules.load_path:/home/iory/catkin_ws/hrp2/devel/share/hrpsys/lib -o manager.modules.preload:HGcontroller.so -o manager.components.precreate:HGcontroller -o exec_cxt.periodic.type:SynchExtTriggerEC -o example.SequencePlayer.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.ForwardKinematics.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.ImpedanceController.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.AutoBalancer.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.StateHolder.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.TorqueFilter.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.TorqueController.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf -o example.ThermoEstimator.config_file:/home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.c [... cut ...]
cache found for file:///home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2JSKNTS_for_OpenHRP3/HRP2JSKNTSmain.wrl
Loading body customizer "/home/iory/catkin_ws/hrp2/devel/share/OpenHRP-3.1/customizer/libSampleCustomizer.so" for springJoint
[Bush customizer] Could not open [/opt/ros/indigo/share/openhrp3/../OpenHRP-3.1/customizer/sample1_bush_customizer_param.conf]
Loading body customizer "/home/iory/catkin_ws/hrp2/devel/share/OpenHRP-3.1/customizer/libSampleBushCustomizer.so" for 
cache found for file:///home/iory/catkin_ws/hrp4/devel/share/OpenHRP-3.1/sample/model/longfloor.wrl

```

snozawa, 2016-04-11:
```
[hrpsys.py]   eval : [self.co, self.co_svc, self.co_version] = self.createComp("CollisionDetector","co")
;;
;; Could not open /dev/console for writing.
;;
open: Permission denied
[co] onInitialize()
cache found for file:///home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2JSKNTS_for_OpenHRP3/HRP2JSKNTSmain.wrl
QH6214 qhull input error: not enough points(3) to construct initial simplex (need 4)

While executing:  | qhull Qt Tc
Options selected for Qhull 2012.1 2012/02/18:
  run-id 656808813  Qtriangulate  Tcheck-frequently  _pre-merge  _zero-centrum
QH6205 qhull error (qh_initqhull_start): qh_qh already defined.  Call qh_save_qhull() first
freeglut  ERROR:  Function <glutBitmapCharacter> called without first calling 'glutInit'.
terminate called after throwing an instance of 'omni_thread_fatal'
[hrpsys.py] Fail to createComps CORBA.COMM_FAILURE(omniORB.COMM_FAILURE_WaitingForReply, CORBA.COMPLETED_MAYBE)
```
で、CollisionDetectorでおちているように見えます。
```
rtmlaunch hrpsys samplerobot.launch
python -i `rospack find hrpsys`/samples/SampleRobot/samplerobot_data_logger.py
```
しても同じのがでるようなら
~/ros/indigo_parent/src/hrpsys/python/hrpsys_config.py
の
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L691
のCollisionDetectorの箇所をコメントアウトして、
hrpsys-baseをcatkin buildしなおしてみて再度samplerobot.launchやhrp2jsknts_startup.launchの実行結果をおくってください。

iory, 2016-04-11:
いかがhrp2jsknts_startup.launchの結果です．

```
rtmlaunch hrpsys_ros_bridge_tutorials hrp2jsknts_startup.launch
[rtmlaunch] Start omniNames at port 15005 

Mon Apr 11 16:25:35 2016:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000074000000010102000f0000003133332e31312e3231362e31373400009d3a00000b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c0000000100000001000100010000000100010509010100010000000901010003545441080000006f510b5701006f78
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iory/.ros/log/9586f08c-ffb6-11e5-8d53-507b9d9efada/roslaunch-crab-28530.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.174:43389/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /use_sim_time: True

NODES
  /
    hrpsys (hrpsys/hrpsys-simulator)
    hrpsys_py (hrpsys_ros_bridge_tutorials/hrp2jsknts_hrpsys_config.py)
    modelloader (openhrp3/openhrp-model-loader)

auto-starting new master
process[master]: started with pid [28545]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9586f08c-ffb6-11e5-8d53-507b9d9efada
process[rosout-1]: started with pid [28558]
started core service [/rosout]
process[modelloader-2]: started with pid [28561]
ready
process[hrpsys-3]: started with pid [28587]
process[hrpsys_py-4]: started with pid [28588]
loading file:///home/iory/catkin_ws/hrp2_debug/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2JSKNTS_for_OpenHRP3/HRP2JSKNTSmain.wrl
Humanoid node
Joint node WAIST
  Segment node BODY
  Joint node RLEG_JOINT0
      Segment node RLEG_LINK0
      Joint node RLEG_JOINT1
          Segment node RLEG_LINK1
          Joint node RLEG_JOINT2
              Segment node RLEG_LINK2
              Joint node RLEG_JOINT3
                  Segment node RLEG_LINK3
                  Joint node RLEG_JOINT4
                      Segment node RLEG_LINK4
                      Joint node RLEG_JOINT5
                          Segment node RLEG_LINK5
                            ForceSensorrfsensor
                          Joint node RLEG_JOINT6
                            Segment node RLEG_LINK6
  Joint node LLEG_JOINT0
      Segment node LLEG_LINK0
      Joint node LLEG_JOINT1
          Segment node LLEG_LINK1
[... 679 lines cut ...]
[abc] setGaitGeneratorParam
[abc]   toe_heel_phase_ratio is successfully set.
[abc]   stride_parameter = 0.15[m], 0.05[m], 10[deg], 0.05[m]
[abc]   leg_default_translate_pos =     [-0,  -0.105,  -0]    [0,  0.105,  0]    [0,  0,  0]    [0,  0,  0]
[abc]   default_step_time = 1.1[s]
[abc]   default_step_height = 0.05[m]
[abc]   default_double_support_ratio_before = 0.16, default_double_support_ratio_before = 0.16, default_double_support_static_ratio_before = 0, default_double_support_static_ratio_after = 0, default_double_support_static_ratio_swing_before = 0.16, default_double_support_static_ratio_swing_after = 0.16
[abc]   default_orbit_type = CYCLOIDDELAY
[abc]   swing_trajectory_delay_time_offset = 0.2[s], swing_trajectory_final_distance_weight = 3
[abc]   stair_trajectory_way_point_offset =     [0.03,  0,  0][m]
[abc]   cycloid_delay_kick_point_offset =     [-0.1,  0,  0][m]
[abc]   gravitational_acceleration = 9.80665[m/s^2]
[abc]   toe_pos_offset_x = 0.142869[mm], heel_pos_offset_x = -0.105784[mm]
[abc]   toe_zmp_offset_x = 0.079411[mm], heel_zmp_offset_x = -0.105784[mm]
[abc]   toe_angle = 0[deg]
[abc]   heel_angle = 0[deg]
[abc]   use_toe_joint = true, use_toe_heel_transition = false
[abc]   toe_heel_phase_ratio = [0.05 0.25 0.2 0 0.2 0.25 0.05 ]
[abc]   optional_go_pos_finalize_footstep_num = 0, overwritable_footstep_index_offset = 1
[es] getEmergencyStopperParam
[es] setEmergencyStopperParam
[es]   default_recover_time = 10[s], default_retrieve_time = 1[s]
[hrpsys_py-4] process has finished cleanly
log file: /home/iory/.ros/log/9586f08c-ffb6-11e5-8d53-507b9d9efada/hrpsys_py-4*.log
```

snozawa, 2016-04-11:
シミュレータは途中で落ちずに起動できたかな。
であればCollisionDetectorのみでおちていることになります。
```
[co] onInitialize()
cache found for file:///home/iory/catkin_ws/hrp2/src/rtm-ros-robotics/rtmros_hrp2/hrp2_models/HRP2JSKNTS_for_OpenHRP3/HRP2JSKNTSmain.wrl
QH6214 qhull input error: not enough points(3) to construct initial simplex (need 4)

While executing:  | qhull Qt Tc
Options selected for Qhull 2012.1 2012/02/18:
  run-id 656808813  Qtriangulate  Tcheck-frequently  _pre-merge  _zero-centrum
QH6205 qhull error (qh_initqhull_start): qh_qh already defined.  Call qh_save_qhull() first
freeglut  ERROR:  Function <glutBitmapCharacter> called without first calling 'glutInit'.
terminate called after throwing an instance of 'omni_thread_fatal'
```
だからglutまわりで落ちてるのかな。

iory, 2016-04-11:
すいませんsamplerobot
```
rtmlaunch hrpsys samplerobot.launch
python -i `rospack find hrpsys`/samples/SampleRobot/samplerobot_data_logger.py
```
は正常に起動しました．

mmurooka, 2016-04-11:
https://github.com/start-jsk/rtmros_hrp2/issues/261でした．

https://github.com/start-jsk/rtmros_hrp2/pull/262 でHRP2JSKNTだけ直しているので，
NTSも同様の修正が必要なようです．

iory, 2016-04-11:
@snozawa  さん@mmurooka さん
以下のようなPRで治すことができました．
ありがとうございます．
https://github.com/start-jsk/rtmros_hrp2/pull/385

closed 2016-04-11 by snozawa