start-jsk/rtmros_tutorials #469: gazebo_kinect_interactive.launchのturtlebot_descriptionへの依存について

ishiguroJSK, 2016-04-15:
あまり重要度高くないのですが，
`roslaunch hrpsys_gazebo_tutorials gazebo_kinect_interactive.launch`
などのチュートリアルで
```
while processing /home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_gazebo/hrpsys_gazebo_general/launch/gazebo_kinect.launch:
while processing /home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_gazebo/hrpsys_gazebo_general/launch/gazebo_sensor.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_gazebo/hrpsys_gazebo_general/robot_models/Kinect/Kinect.urdf.xacro'] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py '$(arg ROBOT_MODEL)'" name="robot_description"/>
The traceback for the exception was written to the log file
```
といった感じで失敗するのでturtlebotKinect.urdf.xacroの内部を調べると
```
<?xml version="1.0"?>
<robot name="Kinect" xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:include filename="$(find turtlebot_description)/urdf/turtlebot_library.urdf.xacro" />
  <link name="base_link" />
  <gazebo reference="base_link">
    <gravity>false</gravity>
  </gazebo>
(以下略)
```
のようにturtlebotのパッケージに依存しているので，apt版のros-indigo-turtlebot入れたら実行できました．
恐らく/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials/package.xmlにturtlebot_descriptionへの依存を追加すべきなんだと思います

ishiguroJSK, 2016-04-15:
修正ありがとうございます

closed 2016-04-15 by ishiguroJSK

mmurooka, 2016-04-17:
@ishiguroJSK, @iory 報告，対応ありがとうございました．