start-jsk/rtmros_tutorials #470: gazebo_hrp2jsknt_no_controllers.launchなどでURDFがないとエラーが出る

ishiguroJSK, 2016-04-15:
[https://github.com/start-jsk/rtmros_tutorials](url)の
```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknt_no_controllers.launch
rtmlaunch hrpsys_gazebo_tutorials hrp2jsknt_hrpsys_bringup.launch
```
などを試そうと思っているのですが，
```
ishiguro@ishiguroZ800:~$ roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknt_no_controllers.launch
WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/ishiguro/.ros/log/69cb01fe-0312-11e6-a85a-001f2905a411/roslaunch-ishiguroZ800-22595.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "VSProjects" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
Traceback (most recent call last):
  File "/opt/ros/indigo/share/xacro/xacro.py", line 62, in <module>
    xacro.main()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 687, in main
    process_includes(doc, os.path.dirname(args[0]))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 271, in process_includes
    raise XacroException("included file \"%s\" could not be opened: %s" % (filename, str(e)))
xacro.XacroException: included file "/home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNT_WH_SENSORS.urdf" could not be opened: [Errno 2] No such file or directory: '/home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNT_WH_SENSORS.urdf'
while processing /home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_gazebo/hrpsys_gazebo_general/launch/gazebo_robot_no_controllers.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/ishiguro/ros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials/robot_models/HRP2JSKNT/HRP2JSKNT.urdf.xacro'] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py '$(arg ROBOT_MODEL)'" name="robot_description"/>
The traceback for the exception was written to the log file
```
となってgazeboが起動しません．JAXON版の
`roslaunch hrpsys_gazebo_tutorials gazebo_jaxon_without_multisense_no_controllers.launch`
も同じようなエラーでした．
確かにros/indigo/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_gazebo_tutorials/robot_models/HRP2JSKNT/以下にはそれらしいURDFが生成されていません．
jsk-ros-pkg/tr○ns_systemはワークスペースに追加した後ビルドされているのですが，他に予め走らせておくべきコマンドなどがあるのでしょうか？

iory, 2016-04-15:
@ishiguroJSK さん

以下に書いてあるのですが，hrp2_modelというpackageがある状態でなければhrp2のwrlがないため生成されないです．（hrp2_modelの中にJAXONのモデルも存在しています）
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials%2FCMakeLists.txt#L8
生成しているところは以下のような場所です．
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials%2FCMakeLists.txt#L485

ですので，repositoryからrtmros_hrp2というpackageをworkspaceに持ってくることで解決するかと思います．(wstool set --git hogehoge)

ishiguroJSK, 2016-04-15:
ありがとうございます．
ビルド時にWARNING出ていたんですね．見逃してました．
早速試してみます

ishiguroJSK, 2016-04-16:
無事起動できました．ありがとうございます．

closed 2016-04-16 by ishiguroJSK