start-jsk/rtmros_tutorials #490: [hrpsys_ros_bridge_tutorials] HRP2W関連のビルドエラー

ishiguroJSK, 2016-10-02:
数週間前から確認されていますが，hrpsys_ros_bridge_tutorialsのビルド時に，
最近追加されたHRP2Wのモデル周り(?)でビルドが止まります．
CMakeLists.txt内の該当する部分をコメントアウトすれば通りますが，
一応報告上げておきます．

```
Starting  >>> hrpsys_ros_bridge_tutorials                                                                                                                                                                                                                                                                               
________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << hrpsys_ros_bridge_tutorials:make /home/ishiguro/catkin_ws/master/logs/hrpsys_ros_bridge_tutorials/build.make.002.log                                                                                                                                                                                      
Xserver connection failedcannot find limb: head
;; Segmentation Fault.
;; in (make-matrix 763 3)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=1046131888 x=3e5ab380 addr=0
signal=11 to thread 0, 
;; No, you cannot continue the previous evaluation.
make[2]: *** [hrp2w_joint_minmax_conf_done] エラー 11
make[1]: *** [CMakeFiles/hrp2w_hrpsys_ros_bridge_tutorials_compile_joint_minmax.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
make: *** [all] エラー 2
cd /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials; catkin build --get-env hrpsys_ros_bridge_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
........................................................................................................................................................................................................................................................................................................................
Failed     << hrpsys_ros_bridge_tutorials:make           [ Exited with code 2 ]                                                                                                                                                                                                                                         
Failed    <<< hrpsys_ros_bridge_tutorials                [ 2.3 seconds ]    
```

k-okada, 2016-10-02:
catkin bt --make-args VERVOSE=1
で失敗しているコマンドがわかるのでそれを実行してみて再現できるようにしてみて下さい

ishiguroJSK, 2016-10-02:
--make-args VERVOSE=1を付けても出力メッセージが変わらなかったのですが，
失敗しているコマンドというのは上記のログメッセージでいう
`cd /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials; catkin build --get-env hrpsys_ros_bridge_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
`
ということな気がするのですが，違うのでしょうか？
ちなみに上記のコマンドを実行すると以下のような出力を出したあたりで端末の制御が帰ってこなくなります
```
$ cd /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials; catkin build --get-env hrpsys_ros_bridge_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
[  1%] Built target jaxon_red_hrpsys_ros_bridge_tutorials_compile_dae
[  2%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_dae
[  3%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_dae
[  3%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_dae
[  4%] Built target hrp3hand_r_hrpsys_ros_bridge_tutorials_compile_dae
[  4%] Built target hrp2w_hrpsys_ros_bridge_tutorials_compile_dae
[  4%] Built target samplerobot_hrpsys_ros_bridge_tutorials_compile_dae
[  4%] Built target jaxon_hrpsys_ros_bridge_tutorials_compile_dae
[  5%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_dae
[  6%] Built target hrp3hand_l_hrpsys_ros_bridge_tutorials_compile_dae
[  7%] Built target jaxon_hrpsys_ros_bridge_tutorials_compile_urdf
[  8%] Built target jaxon_red_hrpsys_ros_bridge_tutorials_compile_urdf

[ 10%] Built target bond_generate_messages_eus
[ 10%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_eusif
[ 11%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_dae
[ 11%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_conf
[ 12%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_urdf
[ 12%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_lisp
[ 13%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_xml
[ 13%] Built target chidori_hrpsys_ros_bridge_tutorials_compile_top_launch
[ 13%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_eusif
[ 15%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_joint_minmax
[ 15%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_urdf
[ 16%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_lisp
[ 17%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_xml
[ 17%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_conf
[ 17%] Built target hrp2g_hrpsys_ros_bridge_tutorials_compile_top_launch
[ 17%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_eusif
[ 17%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_urdf
[ 19%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_joint_minmax
[ 20%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_conf
[ 22%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_xml
[ 23%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_lisp
[ 23%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_top_launch
[ 23%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_top_launch
[ 25%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_joint_minmax
[ 25%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_conf
[ 26%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_xml
[ 26%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_lisp
[ 26%] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_eusif
[ 26%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_eusif
[ 27%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_lisp
[ 28%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_xml
[ 28%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_conf
[ 30%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_joint_minmax
[ 30%] Built target hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_top_launch
[ 30%] Built target hrp2w_hrpsys_ros_bridge_tutorials_compile_top_launch
[ 31%] Built target hrp2w_hrpsys_ros_bridge_tutorials_compile_lisp
[ 32%] Built target hrp2w_hrpsys_ros_bridge_tutorials_compile_conf
[ 32%] Generating hrp2w_joint_minmax_conf_done
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
Xserver connection failed;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x58a0ad0[16374] --> 0x5cd5740[32748] top=3c7e
irtgl irtviewer 
EusLisp 9.19( 1.0.12) for Linux64 created on ip-172-31-15-39(Thu May 5 01:33:05 PDT 2016)

;; extending gcstack 0x5cd5740[32738] --> 0x62c6740[65476] top=4

;; extending gcstack 0x62c6740[65466] --> 0x6346580[130932] top=4

[... 2 lines cut ...]
[... cut ...]
;; extending gcstack 0xa232f00[8378378] --> 0xe21efc0[16756756] top=4

;; extending gcstack 0xe21efc0[16756746] --> 0x161f7080[33513492] top=4

;; extending gcstack 0x161f7080[33513482] --> 0x261a7140[67026964] top=4

;; extending gcstack 0x261a7140[67026954] --> 0x46107200[134053908] top=4

;; extending gcstack 0x46107200[134053898] --> 0x85fc72c0[268107796] top=48b2da1
;; Segmentation Fault.
;; in (make-matrix 763 3)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=-1989257680 x=896e5500 addr=0
```

k-okada, 2016-10-02:
VERBOSE=1

かな

ishiguroJSK, 2016-10-02:
すいません気づきませんでした…
ただVERBOSEでも出力メッセージは変わらないようです．

k-okada, 2016-10-02:
いや変わるはずなのでコマンド打ったところからログをください
だめなら
catkin -v -i --make-args VERBOSE=1
してみて下さい

ishiguroJSK, 2016-10-02:
一度catkin bt した後の2周目ですが
```
$ catkin bt --make-args VERBOSE=1
==> Expanding alias 'bt' from 'catkin bt --make-args VERBOSE=1' to 'catkin b --this --make-args VERBOSE=1'
==> Expanding alias 'b' from 'catkin b --this --make-args VERBOSE=1' to 'catkin build --this --make-args VERBOSE=1'
--------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/ishiguro/catkin_ws/master
--------------------------------------------------------------------
Source Space:       [exists] /home/ishiguro/catkin_ws/master/src
Log Space:          [exists] /home/ishiguro/catkin_ws/master/logs
Build Space:        [exists] /home/ishiguro/catkin_ws/master/build
Devel Space:        [exists] /home/ishiguro/catkin_ws/master/devel
Install Space:      [unused] /home/ishiguro/catkin_ws/master/install
DESTDIR:            [unused] None
--------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        VERBOSE=1
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------------
[build] Found '62' packages in 0.0 seconds.                                                                                                                                                                                           
[build] Updating package table.                                                                                                                                                                                                       
Starting  >>> hrp2_models                                                                                                                                                                                                             
Starting  >>> jaxon_description                                                                                                                                                                                                       
Starting  >>> jsk_models                                                                                                                                                                                                              
Starting  >>> openhrp3                                                                                                                                                                                                                
Starting  >>> openrtm_tools                                                                                                                                                                                                           
Starting  >>> rtmbuild                                                                                                                                                                                                                
Finished  <<< jaxon_description                          [ 0.9 seconds ]                                                                                                                                                              
Finished  <<< rtmbuild                                   [ 0.8 seconds ]                                                                                                                                                              
Finished  <<< jsk_models                                 [ 0.6 seconds ]                                                                                                                                                              
Finished  <<< hrp2_models                                [ 0.3 seconds ]                                                                                                                                                              
Finished  <<< openrtm_tools                              [ 0.2 seconds ]                                                                                                                                                              
______________________________________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << openhrp3:make /home/ishiguro/catkin_ws/master/logs/openhrp3/build.make.001.log                                                                                                                                          
Warning: Mass is zero. <Model>Sample4LegRobot <Link>RLEG_BUSH_ROLL
Warning: Mass is zero. <Model>Sample4LegRobot <Link>RLEG_BUSH_Z
Warning: Mass is zero. <Model>Sample4LegRobot <Link>LLEG_BUSH_ROLL
Warning: Mass is zero. <Model>Sample4LegRobot <Link>LLEG_BUSH_Z
Warning: Mass is zero. <Model>Sample4LegRobot <Link>RARM_BUSH_ROLL
Warning: Mass is zero. <Model>Sample4LegRobot <Link>RARM_BUSH_Z
[... cut ...]
[build]   Abandoned: None.                                                                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                                               
[build] Runtime: 36.2 seconds total.                                                                       
```

ishiguroJSK, 2016-10-03:
すいません忘れていましたが`catkin bt -v -i --make-args VERBOSE=1`と読み替えて実行してみると，やっとVERBOSEな感じにできました．
ただイマイチ「失敗しているコマンド」というのが見えてこないです．(これもログを全て遡るべきなのでしょうか)

```
(略)
[hrpsys_ros_bridge_tutorials:make] Built target hrp2jsknt_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                                      
[hrpsys_ros_bridge_tutorials:make] Built target staro_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                                          
[hrpsys_ros_bridge_tutorials:make] [ 98%] [ 98%] Built target testmdofarm_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                      
[hrpsys_ros_bridge_tutorials:make] Built target pa10_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                                           
[hrpsys_ros_bridge_tutorials:make] [ 98%] Built target hrp2jsk_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                                 
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' から出ます                                                                                                
[hrpsys_ros_bridge_tutorials:make] /usr/bin/make -f CMakeFiles/tqleg0_hrpsys_ros_bridge_tutorials_compile_all.dir/build.make CMakeFiles/tqleg0_hrpsys_ros_bridge_tutorials_compile_all.dir/build                                       
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' から出ます                                                                                                
[hrpsys_ros_bridge_tutorials:make] /usr/bin/make -f CMakeFiles/ystleg_hrpsys_ros_bridge_tutorials_compile_all.dir/build.make CMakeFiles/ystleg_hrpsys_ros_bridge_tutorials_compile_all.dir/build                                       
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' に入ります                                                                                                
[hrpsys_ros_bridge_tutorials:make] make[2]: `CMakeFiles/ystleg_hrpsys_ros_bridge_tutorials_compile_all.dir/build' に対して行うべき事はありません.                                                                                      
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' から出ます                                                                                                
[hrpsys_ros_bridge_tutorials:make] /usr/bin/cmake -E cmake_progress_report /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials/CMakeFiles                                                                                
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' に入ります                                                                                                
[hrpsys_ros_bridge_tutorials:make] make[2]: `CMakeFiles/tqleg0_hrpsys_ros_bridge_tutorials_compile_all.dir/build' に対して行うべき事はありません.                                                                                      
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' から出ます                                                                                                
[hrpsys_ros_bridge_tutorials:make] /usr/bin/cmake -E cmake_progress_report /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials/CMakeFiles                                                                                
[hrpsys_ros_bridge_tutorials:make] [ 98%] Built target ystleg_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                                  
[hrpsys_ros_bridge_tutorials:make] [ 98%] Built target tqleg0_hrpsys_ros_bridge_tutorials_compile_all                                                                                                                                  
[hrpsys_ros_bridge_tutorials:make] Xserver connection failed;; Segmentation Fault.                                                                                                                                                     
[hrpsys_ros_bridge_tutorials:make] ;; in (make-matrix 763 3)                                                                                                                                                                           
[hrpsys_ros_bridge_tutorials:make] ;; You are still in a signal handler.                                                                                                                                                               
[hrpsys_ros_bridge_tutorials:make] ;;Try reset or throw to upper level as soon as possible.                                                                                                                                            
[... cut ...]
[build]   Abandoned: None.                                                                                                                                                                                                            
[build]   Failed:    1 packages failed.                                                                                                                                                                                               
[build] Runtime: 1 minute and 38.2 seconds total.                                                   
```

snozawa, 2016-10-03:
@k-okadaさん

これと関連あるかアレですが、以前なぜかtravisでだけHRP2Wが同様にeusでセグフォしていた
のがあったと思うのですが、あれはどのようにしてなおしたのでしたでしょうか。

mmurooka, 2017-03-19:
これまだ起きますね．．

k-okada, 2017-03-21:
collada2eusやexport_colladaでどっかのリンクのdae/stlが、うまく扱えていないんだと思ったんだけど、
エラー見てみると、
 (make-matrix 763 3)
でおちているけど、みんなこれかな．

dae/stl の問題のときは、cadファイルを全部コメントアウトして、いっこずつもどしていくと、どこかでおちるから、そのファイルがおかしいとわかる．
あまり理由はわかっていないけど、sizeが０になっていたりするんじゃなかったっけか．

ishiguroJSK, 2017-03-22:
僕の環境でも落ちるところは(make-matrix 763 3)でしたが，
改めていじっているとCMakeLists.txtの
`gen_minmax_table_for_closed_robots(HRP2W HRP2W_for_OpenHRP3 HRP2W)`
で落ちている気がします(ここだけコメントアウトすると通るので)
```
[hrpsys_ros_bridge_tutorials:make] Xserver connection failed;; Segmentation Fault.                                                                                                                                                                                                                    
[hrpsys_ros_bridge_tutorials:make] ;; in (make-matrix 763 3)                                                                                                                                                                                                                                          
[hrpsys_ros_bridge_tutorials:make] ;; You are still in a signal handler.                                                                                                                                                                                                                              
[hrpsys_ros_bridge_tutorials:make] ;;Try reset or throw to upper level as soon as possible.                                                                                                                                                                                                           
[hrpsys_ros_bridge_tutorials:make] ;; code=108853488 x=67cf7c0 addr=0                                                                                                                                                                                                                                 
[hrpsys_ros_bridge_tutorials:make] signal=11 to thread 0,                                                                                                                                                                                                                                             
[hrpsys_ros_bridge_tutorials:make] ;; No, you cannot continue the previous evaluation.                                                                                                                                                                                                                
[hrpsys_ros_bridge_tutorials:make] make[2]: *** [hrp2w_joint_minmax_done] エラー 11                                                                                                                                                                                                                   
[hrpsys_ros_bridge_tutorials:make] make[1]: *** [CMakeFiles/hrp2w_hrpsys_ros_bridge_tutorials_compile_joint_minmax.dir/all] エラー 2                                                                                                                                                                  
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' から出ます                                                                                                                                                               
[hrpsys_ros_bridge_tutorials:make] make[1]: ディレクトリ `/home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials' から出ます                                                                                                                                                               
[hrpsys_ros_bridge_tutorials:make] make: *** [all] エラー 2                                                                                                                                                                                                                                           
Failed    <<< hrpsys_ros_bridge_tutorials                [ 1.6 seconds ]  
```

ishiguroJSK, 2017-03-22:
やっぱりmake-joint-min-max-table.l内でHRP2Wが下半身がないにも関わらず下半身のあるgen-min-max-table-hrp2を呼んでいるのが原因な気がしていまして，
```
  (let ((robot-name (send robot :name)))
    (cond ((substringp "hrp2" robot-name) (setq gen-min-max-table-func 'gen-min-max-table-hrp2))
          ((substringp "jaxon" robot-name) (setq gen-min-max-table-func 'gen-min-max-table-jaxon))
          ((substringp "staro" robot-name) (setq gen-min-max-table-func 'gen-min-max-table-staro))
          ))
```

こんな感じの修正でコンパイルは通っている(コリジョンしない保証は全く出来ない)のですがどうでしょうか？
https://github.com/start-jsk/rtmros_tutorials/compare/master...ishiguroJSK:fix-hrp2w-minmax?expand=1

k-okada, 2017-03-22:
> ...ishiguroJSK:fix-hrp2w-minmax?expand=1

お、いい感じに解答に近づいている気がします．
gen-min-max-table-hrp2 で wも通るようするのが良いと思うので、
https://github.com/start-jsk/rtmros_tutorials/compare/
master...ishiguroJSK:fix-hrp2w-minmax?expand=1#diff-
3b1258562a3d9d7b6a48e275e7833c4aR36
で返り値をしらべて、条件を変えてみるのがいいのでは．すくなくともsend robot :lleg :angle-vector は nil
なんだとおもうけど．

ishiguroJSK, 2017-03-24:
一応足の有無に関する条件式を付けましたが，「when( nil ~ )」の有無による怪奇現象が治らず，
該当部分をコメントアウトしましたがこのエラーが起こる環境に対する根本的な解決策じゃない気がします．
https://github.com/start-jsk/rtmros_tutorials/compare/master...ishiguroJSK:fix-hrp2w-minmax?expand=1

確かに存在しないリンク名などをsendすると
```
send *robot* :hoge
/home/ishiguro/catkin_ws/master/devel/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :hoge in (send *robot* :hoge)
```
とエラーになりますが，HRP2Wのrleg,llegに関してはnilが返ってくるので
```
send *robot* :rleg
nil
```
元々のmake-joint-min-max.lでも問題なく通るはずです(実際エラーが出ない人もいる)
min-maxに関するエラーが出るか否かは恐らくeuslispをソースから入れているかどうかで分岐しており，
僕のワークスペースでもeuslispを(jsk.rosbuild見ながら)ソースから入れればエラーが出なくなりました．
```
jsk_control
openhrp3
rtmros_common
rtmros_tutorials
trans_system
jsk_openni_kinect
rtmros_gazebo
hrpsys
rtmros_choreonoid 
rtmros_hrp2 
```
割と闇深い感あります・・・

k-okada, 2017-03-28:
幽霊を信じている人は、いつも霊が見えているのと一緒で、
闇がある、という風に前提で物を見ると闇しか見えないので、
一度、心を入れ替えて、綺麗な目で見てみて下さい．

とは言え、一回汚れてしまった目は中々戻らないだろうとおもって見てみると、
```
DISPLAY= /opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/k-okada/catkin_ws/ws_jsk_rtmros/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/k-okada/catkin_ws/ws_jsk_rtmros/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l "(hrp2w)"
```
でも落ちるので、いろいろコメントアウトすると、
```
;     (setq torso-end-coords (make-cascoords :coords (send CHEST_LINK1_lk :copy-worldcoords) :name :torso-end-coords))                                                           
;     (send torso-end-coords :translate (float-vector -3.2000000000000000e+01 0.0000000000000000e+00 -3.5069999999999999e+02))                                                   
;     (send CHEST_LINK1_lk :assoc torso-end-coords)                                                                                                                              
;     (setq head-end-coords (make-cascoords :coords (send HEAD_LINK1_lk :copy-worldcoords) :name :head-end-coords))                                                              
;     (send head-end-coords :translate (float-vector 7.9000000000000000e+01 0.0000000000000000e+00 1.0350000000000000e+02))                                                      
;     (send head-end-coords :rotate 1.7453275066650775e+00 (float-vector 0.0000000000000000e+00 1.0000000000000000e+00 0.0000000000000000e+00))                                  
;     (send HEAD_LINK1_lk :assoc head-end-coords)                                                                                                                                
(format *error-output* "hoge~%")
     (setq rarm-end-coords (make-cascoords :coords (send RARM_LINK6_lk :copy-worldcoords) :name :rarm-end-coords))
     (send rarm-end-coords :translate (float-vector 0.0000000000000000e+00 1.6899999999999999e+01 -1.7400000000000000e+02))
(format *error-output* "~A~%" rarm-end-coords)
;     (describe rarm-end-coords)                                                                                                                                                 
;     (send rarm-end-coords :rotate 1.5707963267948966e+00 (float-vector 0.0000000000000000e+00 1.0000000000000000e+00 0.0000000000000000e+00))                                  
;     (describe rarm-end-coords)                                                                                                                                                 
;     (send rarm-end-coords :rotate pi/2 #f(0 1 0));1.5707963267948966 (float-vector 0.0000000000000000 1.0000000000000000 0.0000000000000000))                                  
     (send RARM_LINK6_lk :assoc rarm-end-coords)
;; OK                                        
```
で、１行入れる入れないで、落ちると．
落ちるところをよく見ると、
(make-matrix 763 3
なので、そこから、
```
(defclass HRP2W_g1_11_geom0
  :super collada-body
  :slots ())
(defmethod HRP2W_g1_11_geom0
  (:init (&key (name))
;         (send self :def-gl-vertices)                                                                                                                                           
         (send-super :init :name name :replace-obj (send self :qhull-faceset))
;         (send self :assoc gl::aglvertices)                                                                                                                                     
         self)

```
と、
HRP2W_g1_19_geom0　
も同様にすると、とおりそう．
```
(list :vertices (let ((mat (make-matrix 763 3))) (
```
をhoge.l として、
wc hoge.l 
    1  2301 54665 hoge.l
となる．
```
(setq mat (make-matrix 763 3))
(setq fvector-replace (array-entity mat)
      #f(-3.0820001602172852e+01 -1.4864999771118164e+01 
```
というファイルにして`load hoge.l`でも落ちる
```
(setq aaa #f(....
(print (length aaa))                                                                                                                                                            
(print (array-dimensions mat))                                                                                                                                                  
(print (apply '* (array-dimensions mat)))                                                                                                                                       
(fvector-replace (array-entity mat) aaa)  
```
はOKっぽい．
#f か．
(float-vector にするとOKっぽい．

https://github.com/jsk-ros-pkg/jsk_model_tools/pull/203
で直ると思う．

ここまで、俗人度７５％の僕で、約１時間．修行、頑張って下さい．

ishiguroJSK, 2017-03-28:
ありがとうございます．
授かりし戒言を胸に改心に努めます．

k-okada, 2017-03-28:
動作確認はできたかな．まずはここから．

ishiguroJSK, 2017-03-28:
ちょうど今
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/203
を適用した状態でビルドが通ることを確認しました．
ただ以前の，hrpsys_ros_bridge_tutorials以下で`git clean -xfd && catkin bt`したら例のエラーが出る状態で
```
DISPLAY= /opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l "(hrp2w)"
```
でも
`(list :vertices (let ((mat (make-matrix 763 3))) (`〜から始まる長い1行をhoge.lにしてirteusglからloadしても僕の環境では特に落ちないのですが，何か状況が違ったのでしょうか・・・？
```
$ which irteusgl
/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/bin/irteusgl
$ which roseus
/opt/ros/indigo/bin/roseus
$ roscd euscollada/
/opt/ros/indigo/share/euscollada $ 
```

k-okada, 2017-03-28:
> jsk-ros-pkg/jsk_model_tools#203

hrp2w.l のコードの(mat (make-matrix ... の部分を確認してみて下さい．あるいは、hrp2w.l
の行頭にどのeuscolladaを使ったか表示されているので見てみて下さい．

ishiguroJSK, 2017-03-28:
```
;;
;; DO NOT EDIT THIS FILE
;;
;; this file is automatically generated from /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae on (Linux ishiguroZ640 3.13.0-112-generic x86_64) at Tue Mar 28 21:59:02 2017

;;
;; /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials $ /opt/ros/indigo/lib/euscollada/collada2eus /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.yaml /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l 
;;
```
となっており，例の部分は
```
(list :vertices (let ((mat (make-matrix 763 3))) (fvector-replace (array-entity mat) #f(-3.0820001602172852e+01 -1.4864999771118164e+01
```
です．

k-okada, 2017-03-28:
新しいcolladaeusではなくて古い/opt以下を見に行っていますね

--force-cmakeつけるとかあるとは思うけど、
~/catkin_ws/buildの汚れは心の汚れ、ここは一気にcatkin clean -y しましょう

ishiguroJSK, 2017-03-28:
新しいeuscolladaの動作確認するときや古いのに戻すときはそのあたりに気を使って動作確認しているので(たぶん)大丈夫だとは思うのですが，
 今疑問なのは古い(=apt)のeuscolladaの状態で生成したhrp2w.lをloadしたりしても僕の環境で落ちないことです．
@k-okadaさんの考えだと古いeuscolladaで生成したhrp2w.lに含まれる
`(list :vertices (let ((mat (make-matrix 763 3))) (fvector-replace (array-entity mat) #f(`
の行があるとエラーで落ちるということですよね？

k-okada, 2017-03-28:
というか上のhoge.l+optのeusで落とせます
opt/srcの違いは何があるかはまだわからないですが
コンパイルオプションが微妙に違う可能性はあります

ishiguroJSK, 2017-03-28:
```
$ less /tmp/hoge.l
(list :vertices (let ((mat (make-matrix 763 3))) (fvector-replace (array-entity mat) #f(-3.0820001602172852e+01 ~~~~~~~~~~ -1.4625000000000000e+02 )) mat))
$ wc /tmp/hoge.l
    1  2301 54665 /tmp/hoge.l
$ which irteusgl 
/opt/ros/indigo/share/euslisp/jskeus/eus//Linux64/bin/irteusgl
 $ irteusgl
1.irteusgl$ load "/tmp/hoge.l"
t
```
こういうことですよね・・・

k-okada, 2017-03-28:
https://github.com/start-jsk/rtmros_tutorials/issues/490#issuecomment-289762767 の
```
;; /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials $ /opt/ros/indigo/lib/euscollada/collada2eus 
```
はcollada2eusが新しくないのは大丈夫？
hoge.l はhttps://gist.github.com/k-okada/87bed1c1ee310fbe95c67970f610a591
に置きました．

```
k-okada@p40-yoga:/tmp$ wget https://gist.githubusercontent.com/k-okada/87bed1c1ee310fbe95c67970f610a591/raw/2758546e2420d9a602d442e6c62cde95d2c3122e/hoge.l
--2017-03-28 22:42:39--  https://gist.githubusercontent.com/k-okada/87bed1c1ee310fbe95c67970f610a591/raw/2758546e2420d9a602d442e6c62cde95d2c3122e/hoge.l
Resolving gist.githubusercontent.com (gist.githubusercontent.com)... 151.101.100.133
Connecting to gist.githubusercontent.com (gist.githubusercontent.com)|151.101.100.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 54832 (54K) [text/plain]
Saving to: ‘hoge.l’

100%[=======================================================================================================================================>] 54,832      --.-K/s   in 0.1s    

2017-03-28 22:42:40 (434 KB/s) - ‘hoge.l’ saved [54832/54832]

k-okada@p40-yoga:/tmp$ eusgl hoge.l
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble 
EusLisp 9.22() for Linux64 created on ip-172-31-3-45(Thu Dec 29 20:56:31 PST 2016)
Call Stack (max depth: 20):
  0: at (setq aaa (fvector-replace (array-entity mat) #1f(-30.82 -14.865 -147.25 -29.844
```

ishiguroJSK, 2017-03-28:
eusglだと落ちますね

ishiguroJSK, 2017-03-28:
> はcollada2eusが新しくないのは大丈夫？

https://github.com/jsk-ros-pkg/jsk_model_tools/pull/203
の動作確認時は
```
;; /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials $ /home/ishiguro/catkin_ws/master/devel/.private/euscollada/lib/euscollada/collada2eus /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.yaml /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l   
```
なので新しい方を参照しに行って無事コンパイルも通っています．

ishiguroJSK, 2017-03-28:
試しに現行のmaterのjsk_model_toolsのソースをワークスペースに含めると
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/203
の修正がない状態でもビルドが通ってしまいましたね・・・

k-okada, 2017-03-28:
その状態で `eusgl hoge.l`と`irteusgl hoge.l` をためすのはどうでしょう．
```
k-okada@p40-yoga:/tmp$ irteusgl hoge.l
configuring by "/opt/ros/indigo/share/euslisp/jskeus/eus//lib/eusrt.l"
;; readmacro ;; object ;; packsym ;; common ;; constants ;; stream ;; string ;; loader ;; pprint ;; process ;; hashtab ;; array ;; mathtran ;; eusdebug ;; eusforeign ;; coordinates ;; tty ;; history ;; toplevel ;; trans ;; comp ;; builtins ;; par ;; intersection ;; geoclasses ;; geopack ;; geobody ;; primt ;; compose ;; polygon ;; viewing ;; viewport ;; viewsurface ;; hid ;; shadow ;; bodyrel ;; dda ;; helpsub ;; eushelp ;; xforeign ;; Xdecl ;; Xgraphics ;; Xcolor ;; Xeus ;; Xevent ;; Xpanel ;; Xitem ;; Xtext ;; Xmenu ;; Xscroll ;; Xcanvas ;; Xtop ;; Xapplwin 
connected to Xserver DISPLAY=:0
X events are being asynchronously monitored.
;; pixword ;; RGBHLS ;; convolve ;; piximage ;; pbmfile ;; image_correlation ;; oglforeign ;; gldecl ;; glconst ;; glforeign ;; gluconst ;; gluforeign ;; glxconst ;; glxforeign ;; eglforeign ;; eglfunc ;; glutil ;; gltexture ;; glprim ;; gleus ;; glview ;; toiv-undefined ;; fstringdouble irtmath irtutil irtc irtgeoc irtgraph pgsql irtgeo euspqp pqp irtscene irtmodel irtdyna irtrobot irtsensor irtbvh irtcollada irtpointcloud irtx eusjpeg euspng png irtimage irtglrgb 
;; extending gcstack 0x5471de0[16374] --> 0x58ce120[32748] top=3d17
irtgl irtglc irtviewer 
EusLisp 9.22( 1.0.14) for Linux64 created on ip-172-31-3-45(Thu Dec 29 20:56:31 PST 2016)
Call Stack (max depth: 20):
  0: at (fvector-replace (array-entity mat) #1f(-30.82 -14.865 -147.25 -29.844 -14.616 -147.0 -30.679 -15.071 -147.0 -29.941 -14.386 -147.25 -28.955 -14.341 -147.0 -29.005 -14.096 -147.25 -28.905 -14.586 -146.75 -29.747 -14.847
```

ishiguroJSK, 2017-03-28:
その前に
```
wget https://gist.githubusercontent.com/k-okada/87bed1c1ee310fbe95c67970f610a591/raw/2758546e2420d9a602d442e6c62cde95d2c3122e/hoge.l
```
から落としたhoge.lだとirteusglでも落ちました．
単にhrp2w.lから`(list :vertices (let ((mat (make-matrix 763 3))) (fvector-replace (array-entity mat) #f(`の部分を抜き出したhoge.lではなかったですね勘違いしていました．

ishiguroJSK, 2017-03-28:
となると「euslispをソースから入れたら通る」「jsk_model_toolsをソースから入れたら通る」は通っているように見えているだけで，実際のエラーは解決していない状態ってことですかね？

k-okada, 2017-03-29:
もう一回落ち着いて確認してください．euslisp/jsk_model_toolsも入っていない状態で、catkin clean -a; git
clean -xfd した状態で、
ー コンパイルがおちる
jsk_model_tools の origin/master をソースで入れた状態で、
ー コンパイルがおちる （ここで通るのはかなり変、とおるならhoge.lがとおるか確認）
jsk_model_tools のPRをマージした状態で
ー コンパイルがとおる （通らない時は hrp2w.l のヘッダと、#fを確認、）
二
上と違うときは、ログ、実行結果を自分で編集せずに全て提供する．他にも
wstool info, env | grep ROS など、自分に勘違いな無いことを明らかにする情報は全て重要．

k-okada, 2017-03-29:
もう一回落ち着いて確認すると、

> というか上のhoge.l+optのeusで落とせます

は間違いでした．hoge.l 単体だとirtuegl では落ちない．でも、hrp2w.l の該当行を抜くとmin-max-tableの生成は成功する．どこかでリークしているとは思う．

ishiguroJSK, 2017-03-29:
> euslisp/jsk_model_toolsも入っていない状態で、catkin clean -a; git
> clean -xfd した状態で、
> ー コンパイルがおちる

→おちる
```
~/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials (master) $ catkin clean -y && git clean -xfd && catkin b
Xserver connection failed;; Segmentation Fault.
;; in (make-matrix 763 3)
;; You are still in a signal handler.
;;Try reset or throw to upper level as soon as possible.
;; code=-1833979344 x=92afb100 addr=0
signal=11 to thread 0, 
;; No, you cannot continue the previous evaluation.
make[2]: *** [hrp2w_joint_minmax_conf_done] エラー 11
make[1]: *** [CMakeFiles/hrp2w_hrpsys_ros_bridge_tutorials_compile_joint_minmax.dir/all] エラー 2
make[1]: *** 未完了のジョブを待っています....
```
> jsk_model_tools の origin/master をソースで入れた状態で、
> ー コンパイルがおちる （ここで通るのはかなり変、とおるならhoge.lがとおるか確認）

→通ってしまう・・・(irteusgl, eusglともにwgetしてきたhoge.lが落ちる)
```
~/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials (master) $ wstool info
workspace: /home/ishiguro/catkin_ws/master/src

 Localname         S SCM Version (Spec)   UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------         - --- --------------   -----------  ---------------------------
 jsk_model_tools     git master  (-)      7e54fba1669e github.com/jsk-ros-pkg/jsk_model_tools.git
 leptrino_reader     git master  (-)      8cc55c1595c2 github.com/ishiguroJSK/leptrino_reader.git
 openni_tracker      git hydro-devel  (-) ddf2684050fb github.com/ros-drivers/openni_tracker.git
 trans_system        git master  (-)      575460c13dfe github.com/jsk-ros-pkg/trans_system.git
 rtmros_tutorials    git master  (-)      cfd0b270e3d5 github.com/start-jsk/rtmros_tutorials.git
 rtmros_hrp2         git master  (-)      03af597cdfac github.com/start-jsk/rtmros_hrp2.git
 rtmros_gazebo       git master  (-)      7389a49529be github.com/start-jsk/rtmros_gazebo.git
 rtmros_common       git master  (-)      c5cd0c069d48 github.com/start-jsk/rtmros_common.git
 rtmros_choreonoid   git master  (-)      c4a77602670e github.com/start-jsk/rtmros_choreonoid.git
 openhrp3            git master  (-)      4858bfb65984 github.com/fkanehiro/openhrp3.git
 jsk_openni_kinect   git master  (-)      a40c37b4d090 github.com/jsk-ros-pkg/jsk_openni_kinect.git
 jsk_control         git master  (-)      b5698cf3cb93 github.com/jsk-ros-pkg/jsk_control.git
 hrpsys              git master  (-)      32d70e8eba40 github.com/fkanehiro/hrpsys-base.git

~/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials (master) $ catkin clean -y && git clean -xfd && catkin b
[build] Summary: All 62 packages succeeded!                                                                                                               
[build]   Ignored:   5 packages were skipped or are blacklisted.                                                                                          
[build]   Warnings:  22 packages succeeded with warnings.                                                                                                 
[build]   Abandoned: None.                                                                                                                                
[build]   Failed:    None.                       
```
```
 $ env | grep ROS
ROS_ROOT=/opt/ros/indigo/share/ros
ROS_PACKAGE_PATH=/home/ishiguro/catkin_ws/master/src/jsk_control/contact_states_observer:/home/ishiguro/catkin_ws/master/src/jsk_model_tools/eus_assimp:/home/ishiguro/catkin_ws/master/src/jsk_control/eus_nlopt:/home/ishiguro/catkin_ws/master/src/jsk_control/eus_qpoases:/home/ishiguro/catkin_ws/master/src/jsk_control/eus_qp:/home/ishiguro/catkin_ws/master/src/rtmros_gazebo/eusgazebo:/home/ishiguro/catkin_ws/master/src/jsk_model_tools/eusurdf:/home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hironx_tutorial:/home/ishiguro/catkin_ws/master/src/rtmros_hrp2/hrp2_models:/home/ishiguro/catkin_ws/master/src/rtmros_gazebo/hrpsys_moveit_configs/hrp2jsk_moveit_config:/home/ishiguro/catkin_ws/master/src/rtmros_gazebo/hrpsys_moveit_configs/hrp2jsknt_moveit_config:/home/ishiguro/catkin_ws/master/src/rtmros_gazebo/hrpsys_moveit_configs/hrp2jsknts_moveit_config:/home/ishiguro/catkin_ws/master/src/rtmros_gazebo/hrpsys_gazebo_msgs:/home/ishiguro/catkin_ws/master/src/trans_system/jaxon_description:/home/ishiguro/catkin_ws/master/src/trans_system/jaxon_gazebo_ros_plugins:/home/ishiguro/catkin_ws/master/src/jsk_control/joy_mouse:/home/ishiguro/catkin_ws/master/src/jsk_control/jsk_calibration:/home/ishiguro/catkin_ws/master/src/jsk_control/jsk_control:/home/ishiguro/catkin_ws/master/src/jsk_control/jsk_footstep_planner:/home/ishiguro/catkin_ws/master/src/jsk_control/jsk_footstep_controller:/home/ishiguro/catkin_ws/master/src/jsk_control/jsk_ik_server:/home/ishiguro/catkin_ws/master/src/jsk_model_tools/jsk_model_tools:/home/ishiguro/catkin_ws/master/src/rtmros_hrp2/jsk_models:/home/ishiguro/catkin_ws/master/src/jsk_openni_kinect/jsk_openni_kinect:/home/ishiguro/catkin_ws/master/src/jsk_control/jsk_teleop_joy:/home/ishiguro/catkin_ws/master/src/rtmros_choreonoid/jvrc_models:/home/ishiguro/catkin_ws/master/src/jsk_openni_kinect/kinect_near_mode_calibration:/home/ishiguro/catkin_ws/master/src/leptrino_reader:/home/ishiguro/catkin_ws/master/src/trans_system/multisense_sl_description:/home/ishiguro/catkin_ws/master/src/trans_system/multisense_sl_gazebo_ros_plugin:/home/ishiguro/catkin_ws/master/src/openhrp3:/home/ishiguro/catkin_ws/master/src/jsk_model_tools/euscollada:/home/ishiguro/catkin_ws/master/src/hrpsys:/home/ishiguro/catkin_ws/master/src [... cut ...] s_system/VSProjects:/home/ishiguro/catkin_ws/master/devel/:/opt/ros/indigo/share:/opt/ros/indigo/stacks
ROS_MASTER_URI=http://localhost:11311
ROS_MAVEN_DEPLOYMENT_REPOSITORY=/home/ishiguro/catkin_ws/master/devel/share/maven
ROS_MAVEN_PATH=/home/ishiguro/catkin_ws/master/devel/share/maven:/opt/ros/indigo/share/maven
ROS_HOSTNAME=133.11.216.127
ROS_MAVEN_REPOSITORY=https://github.com/rosjava/rosjava_mvn_repo/raw/master
ROS_DISTRO=indigo
ROS_IP=133.11.216.127
ROS_ETC_DIR=/opt/ros/indigo/etc/ros
```

ishiguroJSK, 2017-03-29:
jsk_model_tools の origin/master をソースで入れただけで何故か通ってしまう時のビルドログです
[hoge_catkin_b.txt](https://github.com/start-jsk/rtmros_tutorials/files/878063/hoge_catkin_b.txt)

k-okada, 2017-03-29:
このときのhrp2w.lの最初の数行と、
catkin b -vi hrpsys_ros_bridge_tutorials --make-args VERBOSE=1
はどうなっていますか．
jsk_model_tools  rtmros_hrp2  rtmros_tutorials  trans_system
だけでワークスペースを作るとどうなりますか．

ishiguroJSK, 2017-03-31:
このときのhrp2w.lの最初の数行は，
```
;; this file is automatically generated from /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae on (Linux ishiguroZ640 3.13.0-112-generic x86_64) at Wed Mar 29 15:32:03 2017

;;
;; /home/ishiguro/catkin_ws/master/build/hrpsys_ros_bridge_tutorials $ /home/ishiguro/catkin_ws/master/devel/.private/euscollada/lib/euscollada/collada2eus /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.yaml /home/ishiguro/catkin_ws/master/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l 
```
で， catkin clean hrpsys_ros_bridge_tutorials && catkin b -vi hrpsys_ros_bridge_tutorials --make-args VERBOSE=1 した時の端末出力が以下のファイルです．
[catkin_b_vi.txt](https://github.com/start-jsk/rtmros_tutorials/files/885135/catkin_b_vi.txt)

jsk_model_tools rtmros_hrp2 rtmros_tutorials trans_systemのみのワークスペースでもjsk_model_tools無しでは落ちる，有ると通る，となりました．

[catkin_b_simple_ws.txt](https://github.com/start-jsk/rtmros_tutorials/files/885505/catkin_b_simple_ws.txt)

k-okada, 2017-04-03:
> もjsk_model_tools無しでは落ちる，有ると通る

のときの、それぞれのときで出来たhrp2w.l のdiffをみてみましょう．

```
--------------------------------------------------------------------
Source Space:       [exists] /home/ishiguro/catkin_ws/master/src
Log Space:          [exists] /home/ishiguro/catkin_ws/master/logs
Build Space:        [exists] /home/ishiguro/catkin_ws/master/build
Devel Space:        [exists] /home/ishiguro/catkin_ws/master/devel
Install Space:      [unused] /home/ishiguro/catkin_ws/master/install
DESTDIR:            [unused] None
--------------------------------------------------------------------
```
みると、clean していないように見えるので、jsk_model_toolsの有り無しのそれぞれの、catkin bの前に、catkin clean
しましょう．git clean -xfd もしているかな．

ishiguroJSK, 2017-04-03:
> みると、clean していないように見えるので

すいませんこの時は`catkin clean`すると`--make-args VERBOSE=1`の出力があまりに長くなるので`catkin clean hrpsys_ros_bridge_tutorials`で代用していました

jsk_model_toolsのソース/aptで生成されたhrp2w.lのdiffは以下のようです
```
$ diff /tmp/hrp2w_with.l /tmp/hrp2w_without.l 
4c4
< ;; this file is automatically generated from /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae on (Linux ishiguroZ640 3.13.0-112-generic x86_64) at Mon Apr  3 13:54:56 2017
---
> ;; this file is automatically generated from /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae on (Linux ishiguroZ640 3.13.0-112-generic x86_64) at Mon Apr  3 13:58:02 2017
7c7
< ;; /home/ishiguro/catkin_ws/debug/build/hrpsys_ros_bridge_tutorials $ /home/ishiguro/catkin_ws/debug/devel/.private/euscollada/lib/euscollada/collada2eus /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.yaml /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l 
---
> ;; /home/ishiguro/catkin_ws/debug/build/hrpsys_ros_bridge_tutorials $ /opt/ros/indigo/lib/euscollada/collada2eus /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2W.dae /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.yaml /home/ishiguro/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2w.l 
23,26c23
<      (dolist (l (append links
<                         (mapcan #'(lambda (x) (if (and (derivedp (cdr x) bodyset-link)
<                                                        (not (memq (cdr x) links)))
<                                                   (list (cdr x)))) (send self :slots))))
---
>      (dolist (l links)
1973,1975c1970
<     (:reset-manip-pose
<       (&optional (limbs '(:torso :head :rarm :larm)))
<       """Predefined pose named reset-manip-pose."""
---
>     (:sensor-calib-pose (&optional (limbs '(:torso :head :rarm :larm)))
1980,1982c1975,1977
<           (:head (send self limb :angle-vector (float-vector 0.000000 40.000000)))
<           (:rarm (send self limb :angle-vector (float-vector 50.000000 -30.000000 -10.000000 -120.000000 -25.000000 -5.000000 -20.000000 60.000000)))
<           (:larm (send self limb :angle-vector (float-vector 50.000000 30.000000 10.000000 -120.000000 25.000000 5.000000 -20.000000 -60.000000)))
---
>           (:head (send self limb :angle-vector (float-vector 0.000000 0.000000)))
>           (:rarm (send self limb :angle-vector (float-vector 0.000000 0.000000 0.000000 -90.000000 0.000000 0.000000 0.000000 60.000000)))
>           (:larm (send self limb :angle-vector (float-vector 0.000000 0.000000 0.000000 -90.000000 0.000000 0.000000 0.000000 -60.000000)))
1985,1987c1980
<     (:reset-pose
<       (&optional (limbs '(:torso :head :rarm :larm)))
<       """Predefined pose named reset-pose."""
---
>     (:reset-manip-pose (&optional (limbs '(:torso :head :rarm :larm)))
1992,1994c1985,1987
<           (:head (send self limb :angle-vector (float-vector 0.000000 0.000000)))
<           (:rarm (send self limb :angle-vector (float-vector 10.000000 -10.000000 0.000000 -90.000000 0.000000 0.000000 -10.000000 15.000000)))
<           (:larm (send self limb :angle-vector (float-vector 10.000000 10.000000 0.000000 -90.000000 0.000000 0.000000 -10.000000 -15.000000)))
---
>           (:head (send self limb :angle-vector (float-vector 0.000000 40.000000)))
>           (:rarm (send self limb :angle-vector (float-vector 50.000000 -30.000000 -10.000000 -120.000000 -25.000000 -5.000000 -20.000000 60.000000)))
>           (:larm (send self limb :angle-vector (float-vector 50.000000 30.000000 10.000000 -120.000000 25.000000 5.000000 -20.000000 -60.000000)))
1997,1999c1990
<     (:sensor-calib-pose
<       (&optional (limbs '(:torso :head :rarm :larm)))
<       """Predefined pose named sensor-calib-pose."""
---
>     (:reset-pose (&optional (limbs '(:torso :head :rarm :larm)))
2005,2006c1996,1997
<           (:rarm (send self limb :angle-vector (float-vector 0.000000 0.000000 0.000000 -90.000000 0.000000 0.000000 0.000000 60.000000)))
<           (:larm (send self limb :angle-vector (float-vector 0.000000 0.000000 0.000000 -90.000000 0.000000 0.000000 0.000000 -60.000000)))
---
>           (:rarm (send self limb :angle-vector (float-vector 10.000000 -10.000000 0.000000 -90.000000 0.000000 0.000000 -10.000000 15.000000)))
>           (:larm (send self limb :angle-vector (float-vector 10.000000 10.000000 0.000000 -90.000000 0.000000 0.000000 -10.000000 -15.000000)))
```
~~pose周りは順番が入れ替わってるせいで，主に怪しいところはソース版が
```
(defmethod euscollada-robot
  (:init-ending
   (&optional (type :collada))
   (when (eq type :collada)
     ;; fix all links' mass properties ;; root link relative (collada) -> link local (euslisp)
[... 5 lines cut ...]
                                   (send (send (car links) :worldcoords) :transform-vector
                                         (l . acentroid))))
       (setq (l . inertia-tensor) (m* (transpose (send l :worldrot))
                                      (m* (send (car links) :worldrot)
                                          (l . inertia-tensor))))
       ))
```
に対してapt版が
```
(defmethod euscollada-robot
  (:init-ending
   (&optional (type :collada))
   (when (eq type :collada)
[... cut ...] p)
     (dolist (l links)
       (setq (l . acentroid) (send l :inverse-transform-vector
                                   (send (send (car links) :worldcoords) :transform-vector
                                         (l . acentroid))))
       (setq (l . inertia-tensor) (m* (transpose (send l :worldrot))
                                      (m* (send (car links) :worldrot)
                                          (l . inertia-tensor))))
       ))
```
なところでしょうか．
僕は修行が足りないので時間がかかりますが読み解いてみます．

k-okada, 2017-04-03:
https://github.com/jsk-ros-pkg/jsk_model_tools/commit/6e7657f2b85181d086a9cd35acb02618899aeac5
みると、aptとよんでいるのは0.2.xな気がして、（なんでそうなる？hydro?というのはおいておいて）

> > もjsk_model_tools無しでは落ちる，有ると通る

ソースが 0.2.x だと落ちて、0.3.xだと通る、という主張だろうか？

> すいませんこの時はcatkin cleanすると--make-args VERBOSE=1の出力があまりに長くなるのでcatkin clean
hrpsys_ros_bridge_tutorialsで代用していました

やるべきは、catkin clean -y && catkin ｂ hrpsys_ros_bridge_tutorials でしょうか
．source treeにhrpsys_ros_bridge_tutorialsとjsk_model_tools だとほとんど変わらないはずです．

また、問題になっているのは
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/CMakeLists.txt#L467-L472
ですが、この手のところがcatkin clean
hrpsys_ros_bridge_tutorialsでちゃんと値がかわるか、あまり自信がないので、catkin clean -ｙ
しましょう．

ishiguroJSK, 2017-04-03:
> jsk-ros-pkg/jsk_model_tools@6e7657f みると、aptとよんでいるのは0.2.xな気がして、（なんでそうなる？hydro?というのはおいておいて）

確かに僕の環境のapt版は0.2.4-0trusty-20160612-025110-0700でした
(ちなみにjsk-ros-pkg/jsk_model_tools@6e7657f のどこを見たら僕のバージョンが0.2.xだと分かるのでしょうか・・・？)

> ソースが 0.2.x だと落ちて、0.3.xだと通る、という主張だろうか？

僕のPCの主張はそうなのかもしれないです・・・
ros-indigo-euscolladaを0.3.5にアップグレードしてみましょうか？
(ただそれで通ったとしてhttps://github.com/jsk-ros-pkg/jsk_model_tools/pull/203 が無いと落ちる/有ると通る，という環境が再現できない気がしますが)

k-okada, 2017-04-03:
> (ちなみにjsk-ros-pkg/jsk_model_tools@6e7657f
> のどこを見たら僕のバージョンが0.2.xだと分かるのでしょうか・・・？)
>

左上に0.3.5 ... 0.3.0 と書いてあります

> ソースが 0.2.x だと落ちて、0.3.xだと通る、という主張だろうか？
>
> 僕のPCの主張はそうなのかもしれないです・・・
> ros-indigo-euscolladaを0.3.5にアップグレードしてみましょうか？
> (ただそれで通ったとしてjsk-ros-pkg/jsk_model_tools#203
> <https://github.com/jsk-ros-pkg/jsk_model_tools/pull/203>
> が無いと落ちる/有ると通る，という環境が再現できない気がしますが)
>

/var/cache かどこかにdeb が残っているのと
ソースでも0.2.4で落ちるかは意味ある情報かと思います

> --
--
◉ Kei Okada

ishiguroJSK, 2017-04-03:
apt版の
0.2.4(落ちる)→0.3.5(通る)
ソース版の
0.2.4(落ちる)←0.3.5(通る)
という結果でした

(/var/cache/apt/archivesにdebが残っていないのを知りつつも`sudo apt-get install ros-indigo-euscollada=0.2.4-0trusty-20160612-025110-0700`で戻せるだろうと思ってapt版をアップグレードしたら戻せなくなってしまいました・・・)

k-okada, 2017-04-03:
2017年4月3日 22:02 Yasuhiro Ishiguro <notifications@github.com>:

> ソース版の
> 0.2.4(落ちる)←0.3.5(通る)
>

0.2.4 から 0.3.5 のどこで落ちる／通るが変わるかは調べたかな．

--
◉ Kei Okada

ishiguroJSK, 2017-04-03:
0.2.4~0.2.5の間の
```
commit c6ef634771284a35bbed4d18d38ed9031fb79708
Merge: 4806401 755ab9f
Author: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Date:   Fri May 6 21:04:44 2016 +0900

    Merge pull request #174 from wkentaro/predefined-doc
    
    [euscollada] Add documentation for predefined poses

commit 755ab9f29adf54b6bc42044fabad66f540ae3774
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Fri Apr 15 20:20:52 2016 +0900

    Add documentation for predefined poses

commit 4806401d47487f695a360cb244f8ead90ac177cf
Merge: 745b7f0 e359e91
Author: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Date:   Mon May 2 08:30:14 2016 +0900

    Merge pull request #180 from k-okada/update-travis-16-05-01
    
    update .travis
```
あたりで通らない→通るようになります

k-okada, 2017-04-04:
もうちょっと突き詰めよう
https://github.com/jsk-ros-pkg/jsk_model_tools/blob/master/euscollada/CHANGELOG.rst
見ると可能性があるのは
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/174/files
と
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/186/files
だけどどっちだろう

mmurooka, 2017-04-04:
僕の手元のvirtual box Ubuntu14.04/indigoで@k-okadaと確認した結果は

euscolladaが
aptの0.2.5: エラー
aptの0.3.5: エラー
srcのorigin/master: エラー
srcのk-okada/fix-ishigro: 通る
となりました．

> srcのorigin/master: エラー
のときは，
```
DISPLAY= irteusgl euslisp/make-joint-min-max-table.l models/hrp2w.l "(hrp2w)"
```
で，エラーになりました． `DISPLAY=`を付けないと大丈夫です．

https://gist.github.com/mmurooka/0a5ce1ffecfa0754572108ed1a22f631 をダウンロードして，
```
DISPLAY= irteusgl hoge.l models/hrp2w.l "(hrp2w)"
```
しても同じエラーになるので，即値の変数みたいなのがたくさんあるとエラーになるメモリ関係の問題でしょうか．

k-okada, 2017-04-04:
https://github.com/euslisp/EusLisp/issues/2
https://github.com/euslisp/EusLisp/blob/EusLisp-9.23/lisp/c/reader.c#L656
みたいなバッファの制限関係でしょうか？確かに後者は`irteusgl hoge.l` したら呼び出されている．
前者はここの関数は呼び出されていないから関係ないのかな．
あとは、eus.h のMAXSTACK, SYMBOLHASH, あたりを増やすと通るのかなぁ．．．

ishiguroJSK, 2017-04-04:
僕のノートPCで試すとみなさんと同じ挙動になったのですが,
Z640やZ620だとやはりソース0.3.5のmasterでも落ちないので，
僕は無罪.

```
leus@chidori-HP-Z620-Workstation:~/catkin_ws/debug/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials (master) $ catkin clean -a && git clean -xfd && catkin b
[clean] Removing buildspace: /home/leus/catkin_ws/debug/build
[clean] Removing develspace: /home/leus/catkin_ws/debug/devel
Removing build/
Removing launch/chidori.launch
Removing launch/chidori_nosim.launch
Removing launch/chidori_ros_bridge.launch
Removing launch/chidori_startup.launch
Removing launch/hrp2g.launch
Removing launch/hrp2g_nosim.launch
Removing launch/hrp2g_ros_bridge.launch
Removing launch/hrp2g_startup.launch
Removing launch/hrp2jsk.launch
Removing launch/hrp2jsk_nosim.launch
Removing launch/hrp2jsk_ros_bridge.launch
Removing launch/hrp2jsk_startup.launch
Removing launch/hrp2jsknt.launch
Removing launch/hrp2jsknt_nosim.launch
Removing launch/hrp2jsknt_ros_bridge.launch
Removing launch/hrp2jsknt_startup.launch
Removing launch/hrp2jsknts.launch
Removing launch/hrp2jsknts_nosim.launch
Removing launch/hrp2jsknts_ros_bridge.launch
Removing launch/hrp2jsknts_startup.launch
Removing launch/hrp2w.launch
Removing launch/hrp2w_nosim.launch
Removing launch/hrp2w_ros_bridge.launch
Removing launch/hrp2w_startup.launch
Removing launch/hrp4r.launch
Removing launch/hrp4r_nosim.launch
Removing launch/hrp4r_ros_bridge.launch
Removing launch/hrp4r_startup.launch
Removing launch/jaxon.launch
Removing launch/jaxon_nosim.launch
Removing launch/jaxon_red.launch
Removing launch/jaxon_red_nosim.launch
Removing launch/jaxon_red_ros_bridge.launch
Removing launch/jaxon_red_startup.launch
Removing launch/jaxon_ros_bridge.launch
Removing launch/jaxon_startup.launch
Removing launch/samplerobot.launch
Removing launch/samplerobot_nosim.launch
Removing launch/samplerobot_ros_bridge.launch
Removing launch/samplerobot_startup.launch
Removing launch/staro.launch
Removing launch/staro_nosim.launch
Removing launch/staro_ros_bridge.launch
Removing launch/staro_startup.launch
Removing launch/testmdofarm.launch
Removing launch/testmdofarm_nosim.launch
Removing launch/testmdofarm_ros_bridge.launch
Removing launch/testmdofarm_startup.launch
Removing launch/tqleg0.launch
Removing launch/tqleg0_nosim.launch
Removing launch/tqleg0_ros_bridge.launch
Removing launch/tqleg0_startup.launch
Removing launch/urataleg.launch
Removing launch/urataleg_nosim.launch
Removing launch/urataleg_ros_bridge.launch
Removing launch/urataleg_startup.launch
Removing launch/ystleg.launch
Removing launch/ystleg_nosim.launch
Removing launch/ystleg_ros_bridge.launch
Removing launch/ystleg_startup.launch
Removing models/CHIDORI.RobotHardware.conf
[... 287 lines cut ...]
Finished <== thk_hand_controller             [ 4.7 seconds ]                                                                                    
Finished <== eus_assimp                      [ 5.6 seconds ]                                                                                    
Finished <== eusurdf                         [ 6.0 seconds ]                                                                                    
Finished <== staro_ros_bridge                [ 6.1 seconds ]                                                                                    
Finished <== trans_ros_bridge                [ 7.1 seconds ]                                                                                    
Starting ==> jaxon_ros_bridge                                                                                                                   
Starting ==> l0_ros_bridge                                                                                                                      
Finished <== jaxon_gazebo_ros_plugins        [ 7.3 seconds ]                                                                                    
Finished <== hrpsys_trans_bridge             [ 8.3 seconds ]                                                                                    
Finished <== multisense_sl_gazebo_ros_plugin [ 8.7 seconds ]                                                                                    
Finished <== jaxon_ros_bridge                [ 1.9 seconds ]                                                                                    
Finished <== l0_ros_bridge                   [ 3.5 seconds ]                                                                                    
Finished <== euscollada                      [ 11.2 seconds ]                                                                                   
Starting ==> hrpsys_ros_bridge_tutorials                                                                                                        
Starting ==> openhrp3_tutorials                                                                                                                 
Finished <== openhrp3_tutorials              [ 1.7 seconds ]                                                                                    
Finished <== hrpsys_ros_bridge_tutorials     [ 1 minute and 5.6 seconds ]                                                                       
Starting ==> hrpsys_gazebo_tutorials                                                                                                            
Starting ==> jsk_hrp2_ros_bridge                                                                                                                
Finished <== hrpsys_gazebo_tutorials         [ 3.0 seconds ]                                                                                    
Finished <== jsk_hrp2_ros_bridge             [ 13.3 seconds ]                                                                                   
[build] Finished.                                                                                                                               
[build] Runtime: 1 minute and 30.4 seconds 

```

k-okada, 2017-04-04:
> 僕のノートPCで試すとみなさんと同じ挙動になったのですが,
> Z640やZ620だとやはりソース0.3.5のmasterでも落ちないので，

のであれば、その２つの環境の違いを調べて下さい．euslisp/jskeusのバージョン、memoryのサイズ、、、、

--
◉ Kei Okada

k-okada, 2017-04-04:
まだ問題がつづいている、「僕のノート」と、「僕の手元のvirtual box
Ubuntu14.04/indigo」のeus/jskeus/openhpr3のバージョンを教えて下さい．
また、このときにmodels/以下を一回バックアップしてから、eus/jskeus/openhrp3をapt-get
updateで最新にして、もういっかい
DISPLAY=  irteusgl euslisp/make-joint-min-max-table.l models/hrp2w.l
"(hrp2w)"
が動くか確認してもらえると嬉しいです．

mmurooka, 2017-04-04:
「僕の手元のvirtual box Ubuntu14.04/indigo」
```bash
leus@leus-VirtualBox:~/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials$ rosversion openhrp3
3.1.9
leus@leus-VirtualBox:~/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials$ rosversion hrpsys
315.10.1

```
```bash
leus@leus-VirtualBox:~/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials$ aptitude show ros-indigo-openhrp3
パッケージ: ros-indigo-openhrp3          
状態: インストール済み
自動的にインストールされた: はい
バージョン: 3.1.9-0trusty-20170218-101918-0800
優先度: 特別
セクション: misc
メンテナ: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
アーキテクチャ: amd64
展開サイズ: 53.2 M
依存: collada-dom2.4-dp-base (>= 2.4.4.7-ubuntu1~trusty1), libblas3 | libblas.so.3,
        libboost-filesystem1.54.0, libboost-program-options1.54.0, libboost-regex1.54.0,
        libboost-system1.54.0, libc6 (>= 2.14), libgcc1 (>= 1:4.1.1), libjpeg8 (>= 8c), liblapack3 |
        liblapack.so.3, libomniorb4-1 (>= 4.1.6), libomnithread3c2 (>= 4.0.6), libpng12-0 (>= 1.2.13-4),
        libstdc++6 (>= 4.6), libxml2 (>= 2.7.4), ros-indigo-openrtm-aist, collada-dom-dev, f2c, jython,
        libatlas-base-dev, libboost-all-dev, libeigen3-dev, libf2c2, libf2c2-dev, libjpeg-dev,
        libpng12-dev, libxml2-dev, omniidl-python, python-omniorb, python-omniorb-omg
競合: ros-indigo-openhrp3
説明: This package does not only wrap OpenHRP3 but actually provides the built artifact from the code from its mainstream repository.
 Being ROS-agnostic by itself, you can also use this via ROS together with the packages in rtmros_common
 that bridge between two framework. OpenHRP3 (Open Architecture Human-centered Robotics Platform version
 3) is an integrated software platform for robot simulations and software developments. It allows the
 users to inspect an original robot model and control program by dynamics simulation. In addition,
 OpenHRP3 provides various software components and calculation libraries that can be used for robotics
 related software developments (excerpts from here). The package version number is synchronized to that of
 mainstream, based on this decision.
ホームページ: https://openrtp.jp/svn/hrg/openhrp/

leus@leus-VirtualBox:~/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials$ aptitude show ros-indigo-hrpsys
パッケージ: ros-indigo-hrpsys            
状態: インストール済み
自動的にインストールされた: はい
バージョン: 315.10.1-0trusty-20170218-103848-0800
優先度: 特別
セクション: misc
メンテナ: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
アーキテクチャ: amd64
展開サイズ: 84.8 M
依存: freeglut3, libboost-python1.54.0, libboost-system1.54.0, libc6 (>= 2.15), libgcc1 (>= 1:4.1.1), libgl1-mesa-glx | libgl1, libglew1.10 (>= 1.10.0), libglu1-mesa | libglu1, libirrlicht1.8, libomniorb4-1 (>=
        4.1.6), libomnithread3c2 (>= 4.0.6), libopencv-core2.4, libopencv-highgui2.4, libopencv-imgproc2.4, libpython2.7 (>= 2.7), libqhull6, libsdl1.2debian (>= 1.2.11), libstdc++6 (>= 4.6), libxml2 (>= 2.7.4),
        ros-indigo-openhrp3, ros-indigo-openrtm-aist, freeglut3-dev, libirrlicht-dev, libqhull-dev, libsdl1.2-dev, libxml2-dev, libxmu-dev, python-tk, ros-indigo-cv-bridge
競合: ros-indigo-hrpsys
説明: An OpenRTM-aist-based robot controller.
 This package is the most tailored for humanoid (dual-arm and/or biped) robots for historical reason. hrpsys package does not only wraps but build and installs the code from its mainstream repository
 (github.com/fkanehiro/hrpsys-base). The package version number is synchronized to that of mainstream, based on this decision. Its semantics: MAJOR: Synchronized with OpenHRP3 MINOR: Indicate IDL compatibility
 (packages with different version numbers in this section are NOT compatible) PATCH: Packages with different version numbers in this section ARE compatible. API document is auto-generated on github.
ホームページ: http://ros.org/wiki/hrpsys
```

mmurooka, 2017-04-04:
```
sudo aptitude update
sudo aptitude install ros-indigo-openhrp3 ros-indigo-hrpsys
source /opt/ros/indigo/setup.bash
rosversion openhrp3
rosversion hrpsys
```
でversionは変わらず，
```
catkin clean -y; git clean -dfx; source /opt/ros/indigo/setup.bash; catkin bt
```
で同じエラーが生じて
```
DISPLAY= irteusgl euslisp/make-joint-min-max-table.l models/hrp2w.l "(hrp2w)"
```
でも同じエラーが生じました．

k-okada, 2017-04-04:
euslisp/ jskeus もupdate して見ましょう

ishiguroJSK, 2017-04-04:
「僕のノート」が何してもビルドが通ってしまうようになって参考にならない気もしますが，
```
ros-indigo-euslisp                 9.21.0-1trusty-20160905-102150-0700
ros-indigo-jskeus                  1.0.13-1trusty-20160905-110204-0700 
ros-indigo-openhrp3             3.1.8-0trusty-20160322-012320-0700 
```
```
sudo apt-get update
sudo aptitude install ros-indigo-openhrp3 ros-indigo-euslisp ros-indigo-jskeus
```
```
ros-indigo-euslisp                      9.23.0-0trusty-20170313-063521-0700 
ros-indigo-jskeus                       1.1.0-0trusty-20170313-071400-0700
ros-indigo-openhrp3                  3.1.9-0trusty-20170313-050512-0700 
```
ビルドも`DISPLAY= irteusgl euslisp/make-joint-min-max-table.l models/hrp2w.l "(hrp2w)"もupdate前後共に通ってしまいます…
`

mmurooka, 2017-04-04:
はい、忘れてました。

```
ros-indigo-euslisp    9.22.0-0trusty-20161229-205027-0800
ros-indigo-jskeus    1.0.14-0trusty-20161230-025030-0800
```

```
sudo apt-get update
sudo aptitude install ros-indigo-euslisp ros-indigo-jskeus
```

```
ros-indigo-euslisp    9.23.0-0trusty-20170313-063521-0700
ros-indigo-jskeus    1.1.0-0trusty-20170313-071400-0700
```

のあとにbuildしたら通りました！
```
DISPLAY= irteusgl euslisp/make-joint-min-max-table.l models/hrp2w.l "(hrp2w)"
```
も出来ました。

tohirose, 2018-05-03:
https://github.com/start-jsk/rtmros_hrp2/blob/master/README.md
の手順に沿って、HRP2を動かす環境を構築しようとしているのですが、       
hrpsys_ros_bridge_tutorialsのパッケージで、

```
[hrpsys_ros_bridge_tutorials:make] /usr/bin/make -f CMakeFiles/hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_joint_minmax.dir/build.make CMakeFiles/hrp2jsknts_hrpsys_ros_bridge_tutorials_compile_joint_minmax.dir/build
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ '/home/hirose/catkin_ws/build/hrpsys_ros_bridge_tutorials' に入ります                                 
[hrpsys_ros_bridge_tutorials:make] [ 34%] Generating hrp2jsknts_joint_minmax_done                                                                              
[hrpsys_ros_bridge_tutorials:make] DISPLAY= /opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknts.l ""\(write-min-max-table-to-robot-model-file\ \(hrp2jsknts\)\ \"/home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknts.l\"\ :margin\ 1.0\)"" "(exit)" && touch /home/hirose/catkin_ws/build/hrpsys_ros_bridge_tutorials/hrp2jsknts_joint_minmax_done
[hrpsys_ros_bridge_tutorials:make] Xserver connection failedXserver connection failedXserver connection failedXserver connection failed;; HRP2JSK already has min-max table.
[hrpsys_ros_bridge_tutorials:make] [ 34%] Generating hrp2jsk_joint_minmax_conf_done                                                                            
[hrpsys_ros_bridge_tutorials:make] DISPLAY= /opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsk.l ""\(write-min-max-table-to-conf-file\ \(hrp2jsk\)\ \"/home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSK.conf\"\)"" "(exit)" && touch /home/hirose/catkin_ws/build/hrpsys_ros_bridge_tutorials/hrp2jsk_joint_minmax_conf_done
[hrpsys_ros_bridge_tutorials:make] ;; HRP2JSKNT already has min-max table.                                                                                     
[hrpsys_ros_bridge_tutorials:make] [ 35%] Generating hrp2jsknt_joint_minmax_conf_done                                                                          
[hrpsys_ros_bridge_tutorials:make] DISPLAY= /opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknt.l ""\(write-min-max-table-to-conf-file\ \(hrp2jsknt\)\ \"/home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNT.conf\"\)"" "(exit)" && touch /home/hirose/catkin_ws/build/hrpsys_ros_bridge_tutorials/hrp2jsknt_joint_minmax_conf_done
[hrpsys_ros_bridge_tutorials:make] Xserver connection failedXserver connection failed;; HRP2JSKNTS already has min-max table.                                  
[hrpsys_ros_bridge_tutorials:make] [ 35%] Generating hrp2jsknts_joint_minmax_conf_done                                                                         
[hrpsys_ros_bridge_tutorials:make] DISPLAY= /opt/ros/kinetic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/euslisp/make-joint-min-max-table.l /home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/hrp2jsknts.l ""\(write-min-max-table-to-conf-file\ \(hrp2jsknts\)\ \"/home/hirose/catkin_ws/src/rtm-ros-robotics/rtmros_tutorials/hrpsys_ros_bridge_tutorials/models/HRP2JSKNTS.conf\"\)"" "(exit)" && touch /home/hirose/catkin_ws/build/hrpsys_ros_bridge_tutorials/hrp2jsknts_joint_minmax_conf_done
[hrpsys_ros_bridge_tutorials:make] Xserver connection failed;; Segmentation Fault.                                                                             
[hrpsys_ros_bridge_tutorials:make] ;; in (make-matrix 763 3)                                                                                                   
[hrpsys_ros_bridge_tutorials:make] ;; You are still in a signal handler.                                                                                       
[hrpsys_ros_bridge_tutorials:make] ;;Try reset or throw to upper level as soon as possible.                                                                    
[hrpsys_ros_bridge_tutorials:make] ;; code=1519440240 x=5a90d040 addr=0                                                                                        
[hrpsys_ros_bridge_tutorials:make] signal=11 to thread 0,                                                                                                      
[hrpsys_ros_bridge_tutorials:make] ;; No, you cannot continue the previous evaluation.                                                                         
[hrpsys_ros_bridge_tutorials:make] make[2]: ディレクトリ '/home/hirose/catkin_ws/build/hrpsys_ros_bridge_tutorials' から出ます
```

とエラーが出てビルドできないのですが、どのように対処すれば良いかを教えていただけますと幸いです。
なお、エラーメッセージの全文は、下記の添付ファイルの通りです。
よろしくお願いいたします。
                                                           
[20180503_error_message.txt](https://github.com/start-jsk/rtmros_tutorials/files/1970897/20180503_error_message.txt)

ishiguroJSK, 2018-05-03:
最新のeuslispのソースには根本的解決がマージされているみたいですが，
現時点でのaptには反映されてないので，
ワークスペースにeuslispのソースを持ってくるか，
CMakeListsの`gen_minmax_table_for_closed_robots(HRP2W HRP2W_for_OpenHRP3 HRP2W)`だけコメントアウトすればとりあえず動くと思います．

tohirose, 2018-05-04:
ありがとうございます．状況が理解できました．
CMakeListsのgen_minmax_table_for_closed_robots(HRP2W HRP2W_for_OpenHRP3 HRP2W)をコメントアウトすることで，無事にビルドが通りました．

pazeshun, 2018-10-28:
@ishiguroJSK 
こちらの問題ですが、indigoでもkineticでも最新のeuslispリリースを使えば大丈夫になっている気がしますが、closeしても大丈夫でしょうか。

[1 more reply]

closed 2019-03-11 by ishiguroJSK