start-jsk/rtmros_tutorials #503: gazeboでhrp2が歩かない問題

otsubo, 2017-02-21:
先週の研究会で少し話にあった内容なのですが,最近gazeboでhrp2が歩かないようになってた
問題を少し調べていて,https://github.com/start-jsk/rtmros_gazebo/issues/217 このイシューの最後でご指摘していただいた通り,https://github.com/start-jsk/rtmros_tutorials/pull/419 このプルリクぐらいまで戻すと,問題なく歩きました.何が原因かはっきりわかっていないのですが,何かわかりましたら,またご報告いたします.

mmurooka, 2017-03-10:
ホントですね．最新版だと歩いてすぐ転ぶけれど，https://github.com/start-jsk/rtmros_tutorials/pull/419 まで戻すと問題なく歩きます．

mmurooka, 2017-03-10:
@k-okada, @YoheiKakiuchi , @snozawa 

hrpsys起動時に，hrp2_hrpsys_config.py でstのパラメータをセットしていますが，
セットすると転んで，セットしなければ（デフォルトのままにすれば）歩けることがわかりました．
（rtmros_tutorialsのバージョンを戻して歩けていた理由は，
前のバージョンではstのidlが合わなくてstのパラメータをセットできなかったからでした．）

rtmros_common, rtmros_tutorials, rtmros_gazeboは最新で，
例えばhrp2jsknt(hrp2017)の場合は
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/hrp2_hrpsys_config.py#L144
の一行ををコメントアウトすれば歩けます．

mmurooka, 2017-03-10:
stのパラメータをセットして転ぶ
（origin/masterそのまま）
https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/file/d/0B5qanGnXorOuY2t3S3NYT2ZmbG8/view?usp=sharing

stのパラメータをセットしないで歩ける
（origin/masterからhttps://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/hrp2_hrpsys_config.py#L144 をコメントアウト）
https://drive.google.com/a/jsk.imi.i.u-tokyo.ac.jp/file/d/0B5qanGnXorOuUGFhVVNCT1FvMEE/view?usp=sharing

snozawa, 2017-03-11:
@otsubo さん
ごめんなさい、こちら見落としてました。

@mmurooka さん
現象としては、
- #419まで戻しても、hrpsys_ros_bridge_tutorialsのpython設定が実機と同じ状態だと歩かない
- #419まで戻してかつpython設定を行わないと歩く

というのであってますか？

動いているときに
`(send *ri* :get-st-algorithm)`
を実行してみてもらえると助かります。

TPCCとかが帰ってくる場合、このSTの制御則であるいているということになります。
以前から、辛うじてgazeboであるけているのが確認されてるのはこのモードで、
今の実機とhrpsys-simulator系はTPCC以外の制御則であるいているので、
TPCC以外＋実機と同じパラメタで何とか歩かないか、というのがこれまでの経緯でした。
なので、TPCC以外＋実機と同じパラメタで歩けるとうれしい。

また、歩けない条件のときは、具体的にどのコマンドをうつとどうなりますか？
`(send *ri* :start-st)`
の時点でこける？発振する？
歩くまではなんとか持ちこたえてて歩くとダメ？

mmurooka, 2017-03-11:
> 現象としては、
    #419まで戻しても、hrpsys_ros_bridge_tutorialsのpython設定が実機と同じ状態だと歩かない
    #419まで戻してかつpython設定を行わないと歩く
というのであってますか？

はい，だいたいそうなります．
https://github.com/start-jsk/rtmros_tutorials/pull/419 まで戻すとhrp2_hrpsys_config.pyでセットするstのパラメータが最新のhrpsysと異なるため（stのパラメータの中でemergency stopのマージンがスカラーかベクトルか異なる），その状態ではstの設定を変えることはできないです．
- #419まで戻して，hrp2_hrpsys_config.pyのみ最新にして実機と同じ設定にすると歩かない
- #419まで戻してかつpython設定を行わないと歩く（#419にした時点でstのパラメータは異なるのでそのままでもエラーでセットされない）

ことを確認しました．

> 動いているときに
(send *ri* :get-st-algorithm)
を実行してみてもらえると助かります。

tpccでした．このときの歩き方は少し試した限りではあんまり危なげではなさそうでした．

転び方は，
https://drive.google.com/file/d/0B5qanGnXorOuY2t3S3NYT2ZmbG8/view
の45秒あたりのように，start-stをした時点で後ろ向きや前向きに足首が回って転びます．

snozawa, 2017-03-12:
> #419まで戻して，hrp2_hrpsys_config.pyのみ最新にして実機と同じ設定にすると歩かない

これは、適切にpythonセッティングをしてパラメタがセットされるのを確認しても
あるかないということだよね？

> #419 まで戻すとhrp2_hrpsys_config.pyでセットするstのパラメータが最新のhrpsysと異なるため（stのパラメータの中でemergency stopのマージンがスカラーかベクトルか異なる），その状態ではstの設定を変えることはできないです．

これがよくわからなかったんだけど、#419はrtmros_gazeboのPRであって、
hrpsysや、hrp2_hrpsys_config.pyといったものとリポジトリが違うので、
rtmros_gazeboを向かしにもどしてもhrpsysやhrp2_config_pyは変わらないので
こういう状況が起こらないと思うんだけど、違うのかな？

mmurooka, 2017-03-12:
> これは、適切にpythonセッティングをしてパラメタがセットされるのを確認しても
あるかないということだよね？

そうです．

> これがよくわからなかったんだけど、#419はrtmros_gazeboのPRであって、
hrpsysや、hrp2_hrpsys_config.pyといったものとリポジトリが違うので、
rtmros_gazeboを向かしにもどしてもhrpsysやhrp2_config_pyは変わらないので
こういう状況が起こらないと思うんだけど、違うのかな？

rtmros_tutorialsのPRです．（hrpsys_gazebo_tutorialsはrtmros_tutorialsにあります．）
という理解でもう一度読み返していただければお分かりいただけるかと思います．

snozawa, 2017-03-12:
> rtmros_tutorialsのPRです．（hrpsys_gazebo_tutorialsはrtmros_tutorialsにあります．）
> という理解でもう一度読み返していただければお分かりいただけるかと思います．

なるほど、わかりました。

iory, 2017-04-05:
具体的にgazeboを使って歩くことを試してみました．
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/hrp2_hrpsys_config.py#L82
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/hrp2_hrpsys_config.py#L181
https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_ros_bridge_tutorials/src/hrpsys_ros_bridge_tutorials/hrp2_hrpsys_config.py#L278
の
stp.st_algorithm=OpenHRP.StabilizerService.EEFMQPCOPを
stp.st_algorithm=OpenHRP.StabilizerService.TTPCに変更後，gazeboを立ち上げる

```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknts_no_controllers.launch
rtmlaunch hrpsys_gazebo_tutorials hrp2jsknts_hrpsys_bringup.launch
```

snozawa, 2017-04-05:
これ何目標でやってますか？
なんでも良いのでgazeboでgo-posくらいができる動画がとれればよいのと、普通にあるけるようにするのと、
普通に実機でやっているデモなどがうごいて透過的に使えるようにするのとで、色々違ってくるかなと思います。
それで、TPCCは実機では使われてなくて、また実機のEEFMほげほげのような制御モードとけっこう違いがあります。
後者２つもしくは一番最後の項目くらいを狙うなら、
「歩きました」＝「実機と同じ制御構成、（ほぼ）同じパラメタで歩きました」を調べたら良いと思います。
（後者２つが目標なら、実機いろいろ違うものを作って、動きます、というのは避けたいところ）

https://github.com/start-jsk/rtmros_tutorials/issues/503#issue-209063717
をはじめみたときは、kinematicsモードみたいなのを摘要する前に戻したら
いいかんじになるのかと思ったけど、後々きいてそうじゃないらしい、というのが分かりました。

> start-stの前にreset-poseにすると，何故かHRP2が回転してしまいますので，順番が大事です．

これはなんでだろうね。
前より状況が悪化してそうですね。
パラメタが何があって、何を調整してみてるかを共有すると良さそうですね。

snozawa, 2017-04-05:
あと、動画もあるとわかりやすそうですね。

iory, 2017-04-05:
> これ何目標でやってますか？

動作獲得の学習のために，simulator上で試行ができるといいと思っているので，
そこが目標です．プラス，他のデモなどもgazebo上で再現できるようになるといいです．
なので，後者２つが目標です．

なので，

> 「歩きました」＝「実機と同じ制御構成、（ほぼ）同じパラメタで歩きました」を調べたら良いと思います。

をやっていくのかと思います．

> これはなんでだろうね。
> 前より状況が悪化してそうですね。

reset-pose -> st 回転する
reset-pose -> st -> auto-balancer 回転しない
https://drive.google.com/drive/folders/0B_NiLAzvehC9WjJjdEVmUmluQk0
ということがわかりました．

ttpcで，st -> auto-balancer -> gopos
https://drive.google.com/open?id=0B_NiLAzvehC9bl9aOG12X1BZeGM

EEFMQPCOP -> auto-balancer -> gopos
https://drive.google.com/open?id=0B_NiLAzvehC9NS1zYWhFY19pZW8

snozawa, 2017-04-05:
```
  <world name="default">
    <physics type="ode">
      <gravity>0 0 -9.80665</gravity>
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>1000</real_time_update_rate>
      <ode>
        <solver>
          <type>world</type>
        </solver>
        <constraints>
          <cfm>1e-7</cfm>
          <erp>1e-2</erp>
        </constraints>
      </ode>
    </physics>
  </world>
```
ODEのquickをworldにするのを試したことがないので、
上記を適宜empty.worldかなにかにうつしてためせたりするかな？
http://answers.gazebosim.org/question/232/gazebo-physics-solver-type-quick-vs-world/

snozawa, 2017-04-05:
> 動作獲得の学習のために，simulator上で試行ができるといいと思っているので，そこが目標です．

これいいですね、ぜひやりましょう。

あと先ほど口頭で伝えましたが
start-auto-balancer->start-stの純で起動してみてください

iory, 2017-04-06:
> ODEのquickをworldにするのを試したことがないので、
> 上記を適宜empty.worldかなにかにうつしてためせたりするかな？
> start-auto-balancer->start-stの純で起動してみてください

quick -> worldに変更してみましたが，
EEFMQPCOPの挙動は変わらず，auto-balancerを入れても倒れず，
次にstを入れた時点で倒れました．

@YoheiKakiuchi さんから聞いたことをメモとして書きますと，
choreonoidだと足裏と地面の間にブッシュ要素があって，
それを入れないと歩かないということでした．
あとは，inertiaのリンク間の差が0.001 ~ 1000倍以下でないと，
加速度がかかった時に大きな力がかかり，その部分が吹っ飛び，それにつられて体全体も，
飛んでしまうということでした．
ですが，このinertiaのチェックは行っているとのことなので，
あとは他のパラメータの調整になるのかなと思います．

iory, 2017-04-06:
すいません．

> quick -> worldに変更してみましたが，
> EEFMQPCOPの挙動は変わらず，auto-balancerを入れても倒れず，
> 次にstを入れた時点で倒れました．

ちゃんと変更できていなくて，
worldにすると
collisiondetectorが見つからないというwarningが出るので，
stを入れられないので，
原因を検証します．

```
rtmlaunch hrpsys_gazebo_tutorials hrp2jsknts_hrpsys_bringup.launch
[rtmlaunch] Start omniNames at port 15005

Thu Apr  6 14:48:54 2017:

Starting omniNames for the first time.
Wrote initial log file.
Read log file successfully.
Root context is IOR:010000002b00000049444c3a6f6d672e6f72672f436f734e616d696e672f4e616d696e67436f6e746578744578743a312e300000010000000000000070000000010102000e0000003139322e3136382e39372e3732009d3a0b0000004e616d6553657276696365000300000000000000080000000100000000545441010000001c000000010000000100010001000000010001050901010001000000090101000354544108000000c6d6e5580100336b
Checkpointing Phase 1: Prepare.
Checkpointing Phase 2: Commit.
Checkpointing completed.
... logging to /home/iory/.ros/log/9d039dfe-1a88-11e7-8bf9-d050996634a1/roslaunch-valkyrie-13157.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/iory/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://133.11.216.78:39249/

SUMMARY
========

PARAMETERS
 * /HrpsysSeqStateROSBridge/publish_sensor_transforms: True
 * /controller_configuration: [{'group_name': '...
 * /diagnostic_aggregator/analyzers/computers/contains: ['HD Temp', 'CPU ...
 * /diagnostic_aggregator/analyzers/computers/path: Computers
 * /diagnostic_aggregator/analyzers/computers/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/hrpsys/contains: ['hrpEC', 'Emerge...
 * /diagnostic_aggregator/analyzers/hrpsys/path: Hrpsys
 * /diagnostic_aggregator/analyzers/hrpsys/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/mode/contains: ['Operating Mode'...
 * /diagnostic_aggregator/analyzers/mode/path: Mode
 * /diagnostic_aggregator/analyzers/mode/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/motor/contains: ['Motor']
 * /diagnostic_aggregator/analyzers/motor/path: Motor
 * /diagnostic_aggregator/analyzers/motor/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path:
 * /diagnostic_aggregator/pub_rate: 1.0
 * /robot_pose_ekf/debug: False
 * /robot_pose_ekf/freq: 30.0
 * /robot_pose_ekf/imu_used: False
 * /robot_pose_ekf/odom_used: True
 * /robot_pose_ekf/output_frame: odom
 * /robot_pose_ekf/self_diagnose: False
 * /robot_pose_ekf/sensor_timeout: 1.0
 * /robot_pose_ekf/vo_used: False
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    AutoBalancerServiceROSBridge (hrpsys_ros_bridge/AutoBalancerServiceROSBridgeComp)
    CollisionDetectorServiceROSBridge (hrpsys_ros_bridge/CollisionDetectorServiceROSBridgeComp)
    DataLoggerServiceROSBridge (hrpsys_ros_bridge/DataLoggerServiceROSBridgeComp)
    ForwardKinematicsServiceROSBridge (hrpsys_ros_bridge/ForwardKinematicsServiceROSBridgeComp)
[... 328 lines cut ...]
[HrpsysSeqStateROSBridge0]   localPos =     [-0.0042,  0.0392,  -0.1245][m]
[HrpsysSeqStateROSBridge0] End Effector [larm]LARM_JOINT6 CHEST_JOINT1
[HrpsysSeqStateROSBridge0]   target = LARM_LINK6, sensor_name = lhsensor
[HrpsysSeqStateROSBridge0]   localPos =     [-0.0042,  -0.0392,  -0.1245][m]
[ INFO] [1491457735.624211026]: [HrpsysSeqStateROSBridge] Loaded HRP2JSKNTS
[ INFO] [1491457735.624367652]: [HrpsysSeqStateROSBridge] @Initilize name : HrpsysSeqStateROSBridge0 done
[hrpsys.py] wait for RTCmanager : None
[hrpsys.py] wait for RobotHardware0 : None ( timeout 0 < 10)
[sensor_ros_bridge_connect.py] wait for RTCmanager : valkyrie
[sensor_ros_bridge_connect.py] wait for RobotHardware0 : None ( timeout 0 < 10)
[rtmlaunch] Wait for  /localhost:15005/RobotHardware0.rtc:q   1 /30
[WARN] [WallTime: 1491457736.320193] [0.000000] Could not found CollisionDetector(co), waiting...
[hrpsys.py] wait for RobotHardware0 : None ( timeout 1 < 10)
[sensor_ros_bridge_connect.py] wait for RobotHardware0 : None ( timeout 1 < 10)
[rtmlaunch] Wait for  /localhost:15005/RobotHardware0.rtc:q   2 /30
[WARN] [WallTime: 1491457737.321900] [0.000000] Could not found CollisionDetector(co), waiting...
[hrpsys.py] wait for RobotHardware0 : None ( timeout 2 < 10)
[sensor_ros_bridge_connect.py] wait for RobotHardware0 : None ( timeout 2 < 10)
[rtmlaunch] Wait for  /localhost:15005/RobotHardware0.rtc:q   3 /30
[WARN] [WallTime: 1491457738.324213] [0.000000] Could not found CollisionDetector(co), waiting...
[hrpsys.py] wait for RobotHardware0 : None ( timeout 3 < 10)
[sensor_ros_bridge_connect.py] wait for RobotHardware0 : None ( timeout 3 < 10)
[rtmlaunch] Wait for  /localhost:15005/RobotHardware0.rtc:q   4 /30
[WARN] [WallTime: 1491457739.326570] [0.000000] Could not found CollisionDetector(co), waiting...
```

iory, 2017-04-06:
これ何故できないのでしょう．
hrpsyspyでwait for RobotHardwareはどういう場合に出るでしょうか？

YoheiKakiuchi, 2017-04-06:
変えてる箇所は quick - world のところだけかな？

YoheiKakiuchi, 2017-04-06:
hrp2jsknt/joint_command や hrp2jsknt/robot_state をrostopic echoすると値が出ているかな？

あと、gazeboを上げる方のlaunchでのメッセージが 変更の前後で何か変わっていないかな？

ban-masa, 2017-04-06:
僕もquick->worldにすると同様の現象になりました。
ただ、そもそもgazebo_hrp2jsknts_no_controllers.launchのみを立ち上げた時点で、gazeboの時間が止まったまま進まなくなっているので、それも関係しているように思います。

iory, 2017-04-06:
> 変えてる箇所は quick - world のところだけかな？

はい，empty.worldのそこだけを変更しています．

```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknts_no_controllers.launch
```
を立ち上げた時に，
quickの時には
/HRP2JSKNTS/joint_command 出ない
/HRP2JSKNTS/robot_state 出る
worldの時には，
/HRP2JSKNTS/joint_command 出ない
/HRP2JSKNTS/robot_state 最初に数トピック出て以降，出ない
となっています．

iory, 2017-04-06:
> あと、gazeboを上げる方のlaunchでのメッセージが 変更の前後で何か変わっていないかな？

diffをとってみましたが，時間などの違い以外は
diffがないようです．

```
iory@valkyrie  ~/catkin_ws/hrp2_gazebo2/src/rtm-ros-robotics |        iory@valkyrie  ~/catkin_ws/hrp2_gazebo2/src/rtm-ros-robotic
... logging to /home/iory/.ros/log/e047dc82-1aac-11e7-ad18-d0 | ... logging to /home/iory/.ros/log/e047dc82-1aac-11e7-ad18-d0
Checking log directory for disk usage. This may take awhile.    Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt                                       Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/iory/.ros/log] is   WARNING: disk usage in log directory [/home/iory/.ros/log] is
It's recommended that you use the 'rosclean' command.           It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.97.72:41724/          | started roslaunch server http://192.168.97.72:46840/

SUMMARY                                                         SUMMARY
========                                                        ========

PARAMETERS                                                      PARAMETERS
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso    * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_senso
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/    * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/
[... cut ...]
[ WARN] [1491507175.480678278, 0.413000000]: IOBPlugin: could | [ WARN] [1491507251.690123008, 0.420000000]: IOBPlugin: could
[ WARN] [1491507175.481678304, 0.414000000]: IOBPlugin: could | [ WARN] [1491507251.690879403, 0.421000000]: IOBPlugin: could
[ WARN] [1491507175.482731722, 0.415000000]: IOBPlugin: could | [ WARN] [1491507251.691576373, 0.421000000]: IOBPlugin: could
[ WARN] [1491507175.483617310, 0.416000000]: IOBPlugin: could | [ WARN] [1491507251.692321446, 0.422000000]: IOBPlugin: could
```

YoheiKakiuchi, 2017-04-07:
@ban-masa の説明と、topicが止まっているとのことで、gazeboの時間が進んでいないのが原因かと思います。
gazeboがrostimeを出していて、gazeboのiobではそれを使ってるのでhrpsysが止まるのだと思います。
(gazeboが止まっているのでhrpsysが立ち上がってもあまり意味がないです）
なので、gazeboが止まる原因を追ってみましょう。

- empty.worldだけでgazeboの時間が進むか試してみる

hrpsys_gazebo_general/launch/gazebo_robot_no_controllers.launchで
SPAWN_MODELをfalseにするとロボットがでないと思う

- odeの他のパラメータも変更してみる

@snozawa の貼ってくれているパラメータと quick,world以外に違うところがある。
cfm や erp が違うのでこれを変更すると変わるかどうか。

- IOBPluginを抜いてみる

hrpsys_gazebo_tutorials/robot_models/HRP2JSKNT/HRP2JSKNT.urdf.xacro
のlibIOBPlugin.soをロードしている部分を抜いてみる
それでも動かないようなら、そのxacroの記述を最初の HRP2JSKNT_WH_SENSORS.urdf の
インクルードだけにしてみる。
いずれも、hrpsysは動かないし、ロボットは崩れ落ちると思うけれども、gazeboの時間が
進むかどうか

iory, 2017-04-07:
> empty.worldだけでgazeboの時間が進むか試してみる

quick -> worldにした状態で，立ち上げたところ，時間が進んでいるので，ここまでは大丈夫そうです．
empty.worldで時間が進むのを確認した上で，
SPAWN_MODELをtrueにすると，
時間が0.0234まで進んで止まっているので，
最初少しは時間が進むようです．一瞬joint_statesが出るのもわかりました．

> odeの他のパラメータも変更してみる

もともとcfmは0.0, erpは0.2でやっていましたが，
@snozawa さんの1e-7, 1e-2に変えたり，大きくしてみましたが変わりませんでした．

> IOBPluginを抜いてみる

IOBを抜くと，HRP2が以下のようになりますが，時間が進みます．
[image]

YoheiKakiuchi, 2017-04-07:
IOBPluginに何か問題があるように思う。

rtmros_tutorials/hrpsys_gazebo_tutorials/config/HRP2JSKNT.yaml の中の
  use_velocity_feedback: true
  use_joint_effort: true
をそれぞれ、falseにしてみてどうなるか試してみて欲しい。

これで上手く行かないようなら、IOBPluginにデバッグメッセージを仕込んでどこで止まっているか突き止めて、対応するしかないかな。

iory, 2017-04-07:
> rtmros_tutorials/hrpsys_gazebo_tutorials/config/HRP2JSKNT.yaml の中の
> use_velocity_feedback: true
> use_joint_effort: true
> をそれぞれ、falseにしてみてどうなるか試してみて欲しい。

これも時間が止まりますね．

IOBPluginのデバッグを進めてみます．

iory, 2017-04-07:
rosのclockが途中ででなくなることが問題なので，
iobpluginが直接悪くないような気がします．

iory, 2017-04-11:
HRP2JSKNTS.urdf.xacroを
```
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="HRP2JSKNTS" >
  <xacro:include filename="$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNTS_WH_SENSORS.urdf" />
  <!-- <xacro:include filename="$(find hrpsys_ros_bridge_tutorials)/models/SampleRobot.urdf" /> -->
  <!-- add IOB plugin -->
</robot>
```
というふうにしてやるとclockが進まないで，モデル自体も表示されないのですが，以下のようにsamplerobotに変更するとモデルが出て，clockも進みます．
```
<robot xmlns:xacro="http://www.ros.org/wiki/xacro" name="HRP2JSKNTS" >
  <!-- <xacro:include filename="$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNTS_WH_SENSORS.urdf" /> -->
  <xacro:include filename="$(find hrpsys_ros_bridge_tutorials)/models/SampleRobot.urdf" />
  <!-- add IOB plugin -->
</robot>
```

[image]

iory, 2017-04-11:
includeの部分で，
 <xacro:include filename="$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNTS_WH_SENSORS.urdf" /> の代わりに，WH_sensorsを抜いて
  <xacro:include filename="$(find hrpsys_ros_bridge_tutorials)/models/HRP2JSKNTS.urdf" />
でやると起動出来ます．
[image]

iory, 2017-04-11:
valkyrieがindigoでgazebo4を使っているようなので，
gazebo4をインストールしてみてworldでできるかを試してみました．
worldで実行できましたが，やはりSTをOnにすると倒れてしまいました．
https://drive.google.com/file/d/0B_NiLAzvehC9MzRLOGhRUi01eUE/view?usp=sharing

やはりbush_customizerなどを入れる必要があるのかと思います．

yoshimalucky, 2017-04-11:
valkyrieはindigo+gazebo7です(srcsim パッケージの話)
https://bitbucket.org/osrf/srcsim/wiki/system_setup

この環境を使って実験してますが、かなりちゃんと歩きますね。

また、余談ですが、apt-get install の drcsimにも複数個バージョンがあって、
drcsim5 や drcsim7 のように各gazeboのバージョンに対応したパッケージが存在しています。

ちなみに、ROS DISTROごとの対応付けは、
indigo -> gazebo2
jade -> gazebo5
kinetic -> gazebo7
になってます。
http://gazebosim.org/tutorials?tut=ros_wrapper_versions

こういう感じなので、gazebo7も視野に入れて問題の解決を考えた方がいい気もしますが、
gazebo7までいってしまうと、C++11でコンパイルする必要があり、
boostからSTLに取り込まれた機能に合わせて色々な箇所を書き換える必要が出てきて結構大変そうです。
(一度、gazebo7でrtmros_gazeboをコンパイルしようとしましたが、挫折しました）

iory, 2017-04-11:
@yoshimalucky さん
ありがとうございます．

いずれはkineticでgazebo7になるということで試してみます．

iory, 2017-04-12:
gazebo7でvalkyrieが動くまでを確認し，
rtmros_gazeboをc++11でコンパイルできるようにして
https://github.com/iory/rtmros_gazebo/commit/be012bdf0f3585bcceab61fb4461c1497e13a594
```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknts_no_controllers.launch
```
によってgazeboを立ち上げると，
初期状態から回転してしまいます．
何か解決方法はあるでしょうか．
https://drive.google.com/file/d/0B_NiLAzvehC9RFl2UlN5a1Y3Zzg/view?usp=sharing
立ち上げた時のlogは以下です．
```
 leus@leus-ThinkPad-W541  ~/catkin_ws/hrp2_gazebo/src/rtm-ros-robotics/rtmros_gazebo/hrpsys_gazebo_general   kinetic-devel ●  roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknts_no_controllers.launch
... logging to /home/leus/.ros/log/f8bf374a-1f68-11e7-aa6f-54ee75499bfb/roslaunch-leus-ThinkPad-W541-2726.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://133.11.216.86:37380/

SUMMARY
========

PARAMETERS
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors: ['rfsensor', 'lfs...
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lfsensor/frame_id: LLEG_LINK5
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lfsensor/joint_name: LLEG_JOINT5
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lfsensor/rotation: [1, 0, 0, 0]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lfsensor/translation: [0, 0, -0.105]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lhsensor/frame_id: LARM_LINK6
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lhsensor/joint_name: LARM_JOINT6
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lhsensor/rotation: [1, 0, 0, 0]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/lhsensor/translation: [0, 0, -0.0775]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rfsensor/frame_id: RLEG_LINK5
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rfsensor/joint_name: RLEG_JOINT5
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rfsensor/rotation: [1, 0, 0, 0]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rfsensor/translation: [0, 0, -0.105]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rhsensor/frame_id: RARM_LINK6
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rhsensor/joint_name: RARM_JOINT6
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rhsensor/rotation: [1, 0, 0, 0]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/force_torque_sensors_config/rhsensor/translation: [0, 0, -0.0775]
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0/d: 0.075
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0/i: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0/i_clamp: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0/p: 700.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT0/p_v: 250.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1/d: 0.075
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1/i: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1/i_clamp: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1/p: 700.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/CHEST_JOINT1/p_v: 250.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/d: 0.075
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/i: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/i_clamp: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/p: 700.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT0/p_v: 150.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/d: 0.075
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/i: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/i_clamp: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/p: 700.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/HEAD_JOINT1/p_v: 150.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT0/d: 0.075
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT0/i: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT0/i_clamp: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT0/p: 700.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT0/p_v: 150.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT1/d: 0.075
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT1/i: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT1/i_clamp: 0.0
 * /HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT1/p: 700.0
[... 434 lines cut ...]
[ WARN] [1491992141.729523892, 0.186000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/RARM_JOINT7/
[ WARN] [1491992141.731393239, 0.187000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT0/
[ WARN] [1491992141.732944628, 0.188000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT1/
[ WARN] [1491992141.734669555, 0.189000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT2/
[ WARN] [1491992141.736347257, 0.189000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/LARM_JOINT3/
[... cut ...]
[ WARN] [1491992141.762237981, 0.201000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/R_INDEXMP_P/
[ WARN] [1491992141.763823896, 0.202000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/R_INDEXPIP_R/
[ WARN] [1491992141.765619521, 0.202000000]: IOBPlugin: couldn't find a VP_V param for HRP2JSKNTS/hrpsys_gazebo_configuration/gains/R_MIDDLEPIP_R/

ODE Message 3: LCP internal error, s <= 0 (s=-0.0000e+00)
```

yoshimalucky, 2017-06-20:
@iory 遅くなりましたが、該当のコミットを自分の環境でsamplerobotを使って試してみましたが、やっぱり回転しますね。

まだ原因をしっかり突き止めてなくてあくまで予想なのですが、何か重力に関係する設定が上手く働いてないのかもしれません。このケースに関しては初期状態でモデルに働いている重力がゼロのような挙動に見えます。hrpsysを入れるとリアルタイムファクターが1になってものすごい勢いで吹っ飛んでいきますね・・・。

今、hrpsysとは関係ない話題でDRC-Huboをsrcsimの中で動くようにしましたが、重力をz=-9.8に設定してしまうと、床から受ける反力が大きくなりすぎるのか、モデルをデプロイした後に着地して重力を有効にした瞬間にものすごい勢いで吹っ飛ばされてモデルが粉々になってしまうので、火星の重力のz=-3.711で動かしているところです。

drc_huboのgazebo7対応は@ioryの修正がとても参考になりました。ありがとうございます。
https://github.com/JeongsooLim/drc_hubo/issues/9

Naoki-Hiraoka, 2018-09-01:
kinetic+gazebo7+EEFMQPCOPで，hrp2が歩きました．

動画
https://drive.google.com/open?id=1AQIyhrAJUNcECSHHaqseSL-6s4ImbHug

hrpsys-baseは
https://github.com/fkanehiro/hrpsys-base/commit/ddad4fe1e88f168362307c84c29b7be28f087eb7
rtmros_tutorialsは
https://github.com/Naoki-Hiraoka/rtmros_tutorials/tree/gazebo-hrp2-walk
rtmros_gazeboは
https://github.com/Naoki-Hiraoka/rtmros_gazebo/tree/gazebo-hrp2-walk
を使用し，他は現時点のmasterを使用しました．

```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknts_no_controllers.launch
```
```
rtmlaunch hrpsys_gazebo_tutorials hrp2jsknts_hrpsys_bringup.launch
```

```
roseus ./hrp2jsknts-interface.l
(hrp2jsknts-init)
(setq *robot* (instance hrp2jsknts-robot :init))
(send *ri* :start-auto-balancer);;倒れたらstop-auto-balancer,Ctrl-Shift-rしてやり直す
;;揺れが収まるまで待つ
(send *ri* :angle-vector (send *robot* :reset-pose) 3000)
;;揺れが収まるまで十分待つ
(send *ri* :start-st)
(send *ri* :go-pos 1 0 0)
```

**無重力のように見える問題**
  ->use_loose_synchronizedをtrueにしていると，hrpsysを立ち上げないとシミュレーションがゆっくりになる仕様でした．
**hrp2が吹っ飛ぶ問題**
  ->use_velocity_feedbackをfalseにしました．trueだと地面にぶつかる瞬間に速度を維持しようとして瞬間的に2000Nm近くのトルクが出て吹っ飛ぶようです．
**関節が振動する問題**
  ->inertiaのixx,iyy,izzの最小値をもとの100倍にしました．
  ->関節のfriction,dampingを大きくしました．
  ->gazebo側のIOBPluginの関節制御アルゴリズムを，単純なPD制御にしました．
**start-stすると転ぶ問題**
  ->hrp2の全身の振り子のような揺れが収まるまで十分待ってからstart-stすると転びませんでした．

mmurooka, 2018-09-01:
すばらしいです！
rtmros_tutorialsとrtmros_gazeboの変更を既存のプログラムと共存できるか確認しつつマージしていきましょう．

Naoki-Hiraoka, 2018-09-18:
kinetic+gazebo7で、hrp2017のシミュレーションを行えるようにしました。
先述の歩行を可能にするための変更に加え、新たに以下を可能にするための変更を行いました。
* 始めクレーンから吊り下がってサーボオフの状態から開始する
* 実機と(ほぼ)同じlaunchファイルやconfigファイルを使用し、実機と(ほぼ)同じ手続きでrtcdなどを立ち上げる
* HRP3Handが動かせる
* カメラのトピック名をcamera_remoteにする、取り付け位置をcamera_depth_frameにする

実行時の動画です
https://drive.google.com/open?id=1t3cuELUX_27rx4UJubFC9Z_Lk71o26PP

既存のプログラムとの干渉が心配なのは、
* IOBPluginの各関数名がgazebo7用になっていること(https://github.com/iory/rtmros_gazebo/tree/kinetic-devel を使用しました)
* hrpsys_gazebo_msgsのJointCommand.msg、RobotState.msgに要素が追加されたこと
* IOBPluginやiob.cppが、パラメータuse_servo_onをtrueにした場合の挙動が変更になったこと(hrpsys_gazebo_tutorials内のロボットはuse_servo_onをtrueにしているものは無いので問題ないと思われます)
* 従来のhrp2jskntsのシミュレーションで使用するhrp2jsknts_hrpsys_bringup.launchは、今回修正されたロボットモデルやconfigファイルを共通で使用するため明らかに影響を受けていること

です。
以下、使用方法になります。

hrpsys_gazebo_general
https://github.com/Naoki-Hiraoka/rtmros_gazebo/tree/gazebo-hrp2jsknts
hrpsys_gazebo_tutorials
https://github.com/Naoki-Hiraoka/rtmros_tutorials/tree/gazebo-hrp2jsknts
jsk_hrp2_ros_bridge
https://github.com/Naoki-Hiraoka/rtmros_hrp2/tree/gazebo-hrp2jsknts
を使用する．
hrpsysは
fkanehiro/hrpsys-base@eb5a3f7
を使用する．(これ以降はport番号を変更するコミットがmergeされており、rosbridgeが立ち上がらないため)
他は現時点のmasterを使用する(jsk-ros-pkg/jsk_3rdpartyが必要)

~/.bashrcで，
冒頭を以下のようにする
```
 # case $- in
 #      *i*) ;;
 #        *) return;;
 # esac
 [ -z "$PS1" ] && return 
```
~/.bashrcに以下を追加
```
source /opt/ros/kinetic/setup.bash
source {workspace}/devel/setup.bash
export ROBOT=HRP2JSKNTS
export HRP2NO=17
```
以下を行っておく
```
sudo apt-get install ros-kinetic-voice-text
sudo apt-get install ros-kinetic-jsk-calibration
sudo apt-get install ros-kinteic-pr2-teleop
sudo apt-get install ros-kinetic-pr2-computer-monitor
```
以下の2つのプログラムを順番に立ち上げるとシミュレーションできる
gazebo
```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknts_no_controllers.launch
```
dashboard
```
roslaunch hrpsys_gazebo_tutorials hrp2_dashboard_gazebo.launch
```