start-jsk/rtmros_tutorials #522: JAXONとHRP-2の足裏座標系と力センサ座標系について

mttamtam, 2017-07-09:
いま、足裏にかかる力と位置の関係を力センサ座標系で処理したいです。
```
(send *robot* :force-sensors) ;; get all of the coordinates of force sensors
(car (send *robot* :rleg :force-sensors)) ;; get the coordinates of right leg's force sensor
```
というところはわかるのですが、JAXONとHRP2とで力センサ座標系のz軸方向の正負が逆転しているように見えます。下記のコードで確認できます。

```
(load "package://hrpsys_ros_bridge_tutorials/euslisp/jaxon_red-interface.l")
(load "package://hrpsys_ros_bridge_tutorials/euslisp/hrp2jsknt-interface.l")
(jaxon_red)
(hrp2jsknt)
(send *jaxon_red* :fix-leg-to-coords (make-coords :pos #f(-500 0 0)))
(send *hrp2jsknt* :fix-leg-to-coords (make-coords))
(objects (list *jaxon_red* *hrp2jsknt*
               (send *jaxon_red* :lleg :end-coords :copy-worldcoords)
               (car (send *jaxon_red* :lleg :force-sensors :copy-worldcoords))
               (send *hrp2jsknt* :lleg :end-coords :copy-worldcoords)
               (car (send *hrp2jsknt* :lleg :force-sensors :copy-worldcoords))))
```
センサ座標系の向きはどういう規則で決まってるんでしょうか。
さらに、よく見るとHRP2に関してはセンサ座標系が明らかにセンサのゼロ点ではないところが原点になっている（地面すれすれ）ように見えますが、こういうものでしょうか。

あと、end-coordsと力センサの座標系の関係について、JAXONだと初号機でも二号機でも力センサとz軸の位置が異なるだけのように見えますが、HRP2だとそういう対応関係が無いように見えます。何か決める規則はあったりしますか？

YoheiKakiuchi, 2017-07-10:
eusのセンサ位置は途中で何もしていなければ、WRLに書かれた位置に出ます。
回転は、RobotHardwareのセンサポートから出てくるlfsensor/rfsensorが、
正しい力の方向を向くようにモデルのセンサのrotationを決めています。
（基本的に、センサの取り付け方向の通りにrotateする）

HRP2のセンサ位置は昔からこうだった気がするけど、理由あるんだっけ？
@snozawa

snozawa, 2017-07-10:
> HRP2のセンサ位置は昔からこうだった気がするけど、理由あるんだっけ？
> @snozawa

IOBの側のクローズドなもので、多分むかしからそうだった気がして、
足裏での圧力中心計測をするためなんだと思ってます。

mttamtam, 2017-07-12:
ありがとうございます。
もはやなぜそんなことを見つけたのか忘れましたが、
よく考えてみると当面はJAXONで実験する予定なのでまた困ったら質問させてください。

closed 2017-07-12 by mttamtam