start-jsk/rtmros_tutorials #535: msgsに関するエラー

itohdak, 2017-09-20:
```hrpsys_ros_bridge_tutorials```の中の```hrp2jsknts-interface.l```をロードするときに，
メッセージに関するエラーが大量に出てきます．
実機を動かす上では使用されない（？）メッセージなので今まで支障は出ていませんが，
報告しておきます．

```
eustf roseus_c_util 1.irteusgl$ load "package://hrpsys_ros_bridge_tutorials/euslisp/hrp2jsknts-interface.l"

;; extending gcstack 0x5cd5a70[32738] --> 0x66d0590[65476] top=7c0b
[ERROR] [1505885448.798418785]: Could not find roseus messages for dynamixel_controllers under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                             
try rosrun roseus generate-all-msg-srv.sh dynamixel_controllers
[ERROR] [1505885448.798495001]: Could not find roseus messages for dynamixel_controllers under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                             
try rosrun roseus generate-all-msg-srv.sh dynamixel_controllers
[ERROR] [1505885448.799981380]: Could not find roseus messages for dynamixel_msgs under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                    
try rosrun roseus generate-all-msg-srv.sh dynamixel_msgs
[ERROR] [1505885448.800009786]: Could not find roseus messages for dynamixel_msgs under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                    
try rosrun roseus generate-all-msg-srv.sh dynamixel_msgs
[ERROR] [1505885448.902690560]: Could not find roseus messages for jsk_network_tools under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                 
try rosrun roseus generate-all-msg-srv.sh jsk_network_tools
[ERROR] [1505885448.902738751]: Could not find roseus messages for jsk_network_tools under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                 
try rosrun roseus generate-all-msg-srv.sh jsk_network_tools
[ERROR] [1505885448.934098797]: Could not find roseus messages for stereo_msgs under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                       
try rosrun roseus generate-all-msg-srv.sh stereo_msgs
[ERROR] [1505885448.934159864]: Could not find roseus messages for stereo_msgs under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                       
try rosrun roseus generate-all-msg-srv.sh stereo_msgs
[ERROR] [1505885448.934821638]: Could not find roseus messages for axis_camera under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                       
try rosrun roseus generate-all-msg-srv.sh axis_camera
[ERROR] [1505885448.934861527]: Could not find roseus messages for axis_camera under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                       
try rosrun roseus generate-all-msg-srv.sh axis_camera
[ERROR] [1505885448.978973431]: Could not find roseus messages for laser_assembler under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                   
try rosrun roseus generate-all-msg-srv.sh laser_assembler
[ERROR] [1505885448.979021909]: Could not find roseus messages for laser_assembler under (/home/itohdak/catkin_ws/devel/share/roseus/ros /opt/ros/indigo/share/roseus/ros)                                   
try rosrun roseus generate-all-msg-srv.sh laser_assembler
t
2.irteusgl$
```

mmurooka, 2017-09-20:
https://github.com/jsk-ros-pkg/jsk_robot/pull/682 を参考にすると https://github.com/start-jsk/rtmros_hrp2/pull/489 でなおるはずでしょうか．
これをmergeして，
```
roscd jsk_hrp2_ros_bridge
catkin bt --force-cmake
```
してみてください．

itohdak, 2017-09-20:
助言通り，@mmurooka さんのbranchをmergeしてmakeするとうまく動きました．
ありがとうございました．

mmurooka, 2017-09-22:
https://github.com/start-jsk/rtmros_hrp2/pull/490 で解決したと思うのでcloseします．

closed 2017-09-22 by mmurooka