start-jsk/rtmros_tutorials #545: multisense_description のURDFモデル参照先

yoshimalucky, 2018-05-29:
こちらのモデルファイルではmultisense_descriptionのURDFへのリンクが以下となっており、
https://github.com/start-jsk/rtmros_tutorials/blob/1f1a9737d01fac75ead8f898b332968ee8751b2e/hrpsys_ros_bridge_tutorials/models/HRP2JSK.urdf.xacro#L3

現在では以下のように、multisense_description/urdf/multisenseSL/multisenseSL.urdf.xacro に変更になっています。
https://bitbucket.org/crl/multisense_ros/src/58b246feb1f0/multisense_description/urdf/multisenseSL/?at=default

これが原因で、hrpsys_ros_bridge_tutorialsがビルドできてないのですが、これで誰も困っている人はいないのでしょうか？

もしくは、これが放置されている何か特別な理由がありますか？（indigoのdebだとこのままでいいとか。こちらの環境はkineticです)

YutaKojio, 2018-05-29:
https://github.com/start-jsk/rtmros_tutorials/pull/504 で議論されています．
まだ明確な解決案は出ていないはずですが，
https://github.com/jsk-ros-pkg/jsk_common/pull/1507
のように現状.rosinstallにversionを指定して落としている人が多いかと思います．

yoshimalucky, 2018-05-29:
@YutaKojio 回答ありがとうございます。モデルの中身まで変更になって互換性が失われているということで、単純にパスを変えればいいという問題ではないんですね。

ひとまず、バージョンを3.x系に固定してソースからmultisenseを使おうと思います。

closed 2018-05-29 by yoshimalucky