start-jsk/rtmros_tutorials #557: urata_hrpsys_config.pyを分離する

kyawawa, 2018-12-22:
@YutaKojio さんと話したのですが，ロボット毎の設定が全てurata_hrpsys_config.pyに書かれており，非常に編集しづらいかつ，gitのdiffも見づらい状況になっています．
jaxon_hrpsys_config.pyやchidori_hrpsys_config.pyでURATAHrpsysConfiguratorを継承したクラスを作り，そこで設定を行おうと思うのですが問題のある方はいますでしょうか？ >
@kindsenior @YuyaNagamatsu @shintarokkk

kyawawa, 2018-12-22:
ロボット毎の継承にすると
https://github.com/start-jsk/rtmros_choreonoid/blob/master/hrpsys_choreonoid_tutorials/src/hrpsys_choreonoid_tutorials/choreonoid_hrpsys_config.py
など，URATAHrpsysConfiguratorを継承しているクラスの記述が大変になりそうだということに気付いて手が止まりました．

kyawawa, 2018-12-24:
> これは重要だと思います。この瞬間はバラバラになりましたね、でおわりますが、１年たつと、あのファイルは整理されていなくて使いずらい、新しいのを作るべきだ、となって、無限ループに入ります。

はい．この点に関しては気をつけながら作業を進めていきます．

>  URATAHrpsysConfiguratorを継承しているクラスの記述

に該当するファイルの中身をあまり理解できておらず，Choreonoidや実機特有の設定が書いていそうだということしか分かっていません．
経緯はわかりませんがabcとsensorをつなげている部分は現在はコメントアウトされており，rtcレベルで見てもabcがセンサ情報を使っている部分はないため，消して良い部分だと思われます．
その他で上流に渡せそうな箇所は渡していくつもりです．

> 例えば、簡単なところでは、どのRTCを組み込むかは人が書いてあるとしても、そうだった時の残りの作業は記述のひつようはないですね。

ここの理解が及ばなかったのですが，例えば今の場合どのような部分が必要のないところに当たるのでしょうか？
また，あえて，というのはどのような判断の下で行われるべきなのでしょうか(実行の複雑さ，速度，ソフトウェアの分離などでしょうか．)

YutaKojio, 2018-12-24:
> コメントアウトされていませんでした．(rtcレベルの話で)こちらはref force周りで必要なんでしたっけ？

ログを増やしているだけで制御には関係ないですが，おそらくchoreodnoiの導入当初のデバッグ用とかでしょうか？？

> Choreonoid以外でも必要なのであれば，上流に流していきます．
> その他で上流に渡せそうな箇所は渡していくつもりです．

ロボットの歩行制御等の確認用のログとしても必要なさそうなので，上流に流さず消すだけでいいと思います．

YoheiKakiuchi, 2018-12-24:
何回か、同じ議論をしている気がしていて、少し整理したいので一旦待ってもらえますか。

kyawawa, 2018-12-26:
> 何回か、同じ議論をしている気がしていて、少し整理したいので一旦待ってもらえますか。

整理していただいた後でも構わないですが，過去の議論のスレッドやログがあれば共有していただけると助かります．

> 本当にこれが達成されるかほどほどに疑問

if文を消してファイルとクラスを分けるだけで可読性は上がると考えていたのですがイマイチでしょうか．
ここの方針が間違っていると全てが無駄なので，何か意見ある方はお願いします．

> 過去に、hrpsysのidl (STやautobalancerなど)の変更で変更が必須になることがあって、

こちら，該当コミットなどはありますでしょうか．

YoheiKakiuchi, 2018-12-26:
> 整理していただいた後でも構わないですが，過去の議論のスレッドやログがあれば共有していただけると助かります．

https://github.com/jsk-ros-pkg/trans_system/pull/401
https://github.com/start-jsk/rtmros_tutorials/pull/333
ここらへんでしょうか。もっと継承の構成とかで議論があった気がするけど見つからない。

パラメータ設定部分が分かりにくいのはそうだろうと思うのですが、pythonまわりのわかりにくさの根本は、
どういう継承になっていて、どこにファイルがあって、メソッド、パラメータがどこで設定されているか分からない
ということなのだろうと思います。
なので、まずはこのpythonまわりのクラスの関係や、なぜそうなっているか、派生させる場合の例、気をつけポイント
などをまとめたドキュメントがあって欲しい。

>> 過去に、hrpsysのidl (STやautobalancerなど)の変更で変更が必須になることがあって、
> こちら，該当コミットなどはありますでしょうか．

これはこういう流れだね。
idlの変更
https://github.com/fkanehiro/hrpsys-base/commit/0f702bef14b7a82d0e58a61292c697ab08175329#diff-fdb21e47f442c727ac9cacb88bbb65a9

pythonの修正
https://github.com/start-jsk/rtmros_tutorials/commit/5ec611c75d7ba83f0163c1bff207ea0a61da8373#diff-7735c73bd331a515e088d009888ef158

pythonの修正はすべてのロボットに必要となるはず（設定していないパラメータがあれば問題は発覚しないが、隠れているだけ）

kyawawa, 2018-12-27:
> パラメータ設定部分が分かりにくいのはそうだろうと思うのですが、pythonまわりのわかりにくさの根本は、
> どういう継承になっていて、どこにファイルがあって、メソッド、パラメータがどこで設定されているか分からない
> ということなのだろうと思います。

これは僕もそう思います．
素晴らしく簡潔な継承の構成を見つければ解決する問題なのでしょうか．
それは難しそうなのでまずはPythonのクラスにdocstringを書いてみます．

> これはこういう流れだね。

これ僕のcommitですね．失礼しました．
この件は完全に忘れていましたが，パラメータの型や変数名が変わった時はすべての修正が必要だという認識はありました．
ただ，この問題への対応策は思いつきませんでした．．．

kyawawa, 2019-01-28:
https://github.com/start-jsk/rtmros_tutorials/pull/558/commits/dbbf0d58068995edd08e421e6cfbaa23de5e2751 にてdocstringを書いた上で，今回の変更を適用した場合の継承関係の図も書いてみました．

[image]

この構成である理由や注意点などは含められませんでしたが，クラスの関係性や各クラスでどのようなことをやっているのかの簡単な説明は理解できる図になったと思います．
今回の主な目的は，図真ん中のURATAHrpsysConfiguratorと下のRobotの設定を分離し，各ロボットを使う人が管理しやすくすることになります．

YutaKojio, 2019-03-15:
やっぱりこの分離が必要だと感じてきましたが，どうでしょう．

現在，urata_hrpsys_configに各ロボット固有のパラメータが書かれており，
それを継承した，
(1) choreonoidを使う場合： choreonoid_hrpsys_config
(2) 実機を使う場合: urata_real_hrpsys_config
を使うようになっています．

そのため，urata系ロボット以外(レポジトリ自体も違う)では上記の(1),(2)をそれぞれ継承してさらにロボット固有のパラメータをそれぞれ設定する必要があります．
固有パラメータが(1),(2)の場合の２つのファイルに同じ内容が書かれたものが存在してしまっていて，管理が非常にしづらいです．（某ロボットが実際今そうなっています）

そこで，https://github.com/start-jsk/rtmros_tutorials/issues/557#issuecomment-458049683 の図のような構成にすると簡潔に解決すると思っています．

ishiguroJSK, 2019-03-15:
継承シンプルにしたいのは僕も賛成です．

どう変更するかは置いておいて，こういう大規模な変更の時は，*_new.pyとか*_2.pyにしてしばらく併設して試験運用する，という形ならマージして貰いやすいのでしょうか？-> @k-okada

変更内容に関しては，

> if文を消してファイルとクラスを分けるだけで可読性は上がると考えていたのですがイマイチでしょうか．
> ここの方針が間違っていると全てが無駄なので，何か意見ある方はお願いします．

僕はResetPoseや，ST,ABCパラメータや，RTCリスト，ポート接続リスト，などの諸元はデータベースのように一つのcommon_robot_params.pyみたいなものにずらっと書かれていても悪くないと思っています．(これタブーだったりするんでしょうか)
他のロボットと比較調整することもありますし．

> 仕様変更があるたびに，それに気づいた人が自分のロボットのコンフィグだけ編集してロボットごとにフォーマットがバラバラになる状況が容易に想像できるので，

もありますし．

何ならReal/Simでの違いも同じ場所に書いたら，実機ではどうだったっけ/Choreonoidではどうだったっけ，と右往左往しなくて済みますし．
依然，クローズドなロボットは別ファイルになりますが，rtmros_tutorial側では引数で自由にclosed_robotA_param.pyを渡せるようにしておくとか．

k-okada, 2019-03-18:
現状の整理ですが

１） RTC（ノード／コネクタ／パラメータについて）
JSKでロボットにかかわらず共通に使っているもの／ロボット毎に共通に使っているもの／個人に依存して使っているもの

２）実機とSim（Choreonoidのこと？）で違う所

はどうなっていますでしょうか．

ishiguroJSK, 2019-03-18:
思いつく範囲でしか列挙できないので補足あればお願いしたいところですが，

> １） RTC（ノード／コネクタ／パラメータについて） 
JSKでロボットにかかわらず共通に使っているもの・・・stable的なものからst,abc等々
ロボット毎に共通に使っているもの・・・某ロボットの直動アクチュエータ変換周りのRTC
個人に依存して使っているもの ・・・個人でRTC追加やパラメータ変更はあるが，共有しないと困るものはないはず

２）実機とSim（Choreonoidのこと？）で違う所
- RTCの有無(Robot_Hardware,VirtualForceSensor, TorqueFilter)
- 一部ダンパ係数，エラーリミットなど
- JAXONは関節角数も(multisense,THK_hand)

kyawawa, 2019-03-18:
> １） RTC（ノード／コネクタ／パラメータについて）

ほとんどのRTCは全ロボット共通だと思います．
直動アクチュエータに関してはたしかIOBで吸収しているので，hrpsysのレイヤには出てきていないと思います．

> ２）実機とSim（Choreonoidのこと？）で違う所

見た感じJAXONとJAXON_REDのエラーリミット以外で異なるパラメータは無さそうですが．
multisenseやハンドの関節に関しては，wrlに追加されてはいますがhrpsysで使用していない(考慮しなくても問題ない)のではないのでしたっけ？
それと，実機の方にはサーボオンなどの関数が追加されていて，Choreonoid用にはデバッグ用のポートの追加などがありますね．

k-okada, 2019-03-18:
> JSKでロボットにかかわらず共通に使っているもの・・・stable的なものからst,abc等々
> ロボット毎に共通に使っているもの・・・某ロボットの直動アクチュエータ変換周りのRTC

これだったら，直動アクチューエタロボットだけconfig.pyが必要で後は
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L734
（に足りないものがあればそれを追加する）でよいのでは？

> RTCの有無(Robot_Hardware,VirtualForceSensor, TorqueFilter)

これは，
https://github.com/fkanehiro/hrpsys-base/blob/master/python/hrpsys_config.py#L975
でsimulationのときだけ入れる／入れないをしたらよいのでは？

> 一部ダンパ係数，エラーリミットなど

ダンパ係数は何の係数でしょうか？制御時のなんかのダンパ係数？ハードウェアに起因するものならモデルファイルに書いていないでしょうか？
エラーリミットはモデルファイルには書いていないでしょうか？

> JAXONは関節角数も(multisense,THK_hand)

関節数が違うと何が困るでしょうか？

--
◉ Kei Okada

ishiguroJSK, 2019-03-18:
はい，今でもロボットの差異，実機/simで微妙に違う部分を，○○_hrpsys_config.pyのように継承を重ねて，RTCListやST,ABCの係数などを上書きしながら使い分けているのですが，共通部分のまとめ方を改良しないと，JSK以外のロボットから継承して使いづらいという話です．

> 関節数が違うと何が困るでしょうか？

現状，同じようにsetResetPose()を呼んでも，実機用/sim用のどちらのpyを継承したかで関節数の違う実機/simにそれぞれ用に正しい関節角リストが渡るよう工夫されているので，それが崩れてしまうと思います．

k-okada, 2019-03-18:
> 現状，同じようにsetResetPose()を呼んでも，実機用/sim用のどちらのpyを継承したかで関節数の違う実機/simにそれぞれ用に正しい関節角リストが渡るよう工夫されているので，それが崩れてしまうと思います．

setResetPose () の中にangle-vector
が直書きになっているなら，実機用/sim用のどちらのpyというものはなくして，同じpyで関数の中でsimulation/実機で分岐させたら良いと思うのと，多分だけど，実機はmultisenseがなく，シミュレータはmultisenseがあるのかな？であればsimulatorを実機に併せて，multisenseはrobot
stateに入れないようにするんじゃないだろうか？実機もそうなっているんだよね？

ishiguroJSK, 2019-03-18:
> 直動アクチュエータに関してはたしかIOBで吸収しているので，hrpsysのレイヤには出てきていないと思います．

直動/回転の変換はそれ用のRTCが外部のレポジトリに存在して，.pyで起動，接続されているはずです．

> multisenseやハンドの関節に関しては，wrlに追加されてはいますがhrpsysで使用していない(考慮しなくても問題ない)のではないのでしたっけ？

具体的には把握してないけど，一手間増えたり減ったりしてたと思います
https://github.com/start-jsk/rtmros_choreonoid/blob/3084d4ab766312c07182aab63023f7676e6dc6a5/hrpsys_choreonoid_tutorials/scripts/jaxon_red_rh_setup.py#L43-L57

ishiguroJSK, 2019-03-18:
> 同じpyで関数の中でsimulation/実機で分岐させたら良いと思うのと，多分だけど，実機はmultisenseがなく，シミュレータはmultisenseがあるのかな？であればsimulatorを実機に併せて，multisenseはrobot
> stateに入れないようにするんじゃないだろうか？実機もそうなっているんだよね？

はい実機はmultisense,hand以外です．しかし，本当にmultisense，THK_handの扱いを統一できるかは自信ないです．

> 実機用/sim用のどちらのpyというものはなくして，同じpyで関数の中でsimulation/実機で分岐させたら良いと思うのと

僕も割とこちら派です．比較しやすいので．
しかし，継承でファイルを分化していくのではなく，一つのファイルで統一的に記述する方針をとると，urata_hrpsys_config, urata_real_hrpsys_config, choreonoid_hrpsys_config,...などなどを全部統合するような大きな話になりますね・・・

k-okada, 2019-03-18:
統合というのは concatenate ではなくて，一般化していく，ということだと考えて，

ー 実機とシミュレータは同じにして，choreonoide xxx というのは無くせないか
ー 必要なRTCは全てrtmros_common にいれて，個別のロボット向けのXXというのはなくせないか？
ー そのパラメータはモデルファイルには書いていないのか？

というのを確認するのがいいんじゃないでしょうか？
話を聞くと，他の人のはよくわからないけど自分のはこうで，，，，とい感じに聞こえて，その自分の部分でファイルなのかクラス構造が肥大化しているように見えるから，まずは他のロボットがどうなっているか知るところからでしょうか．

kyawawa, 2019-03-18:
> 直動/回転の変換はそれ用のRTCが外部のレポジトリに存在して，.pyで起動，接続されているはずです．

これはそうですね．失礼しました．

> 具体的には把握してないけど，一手間増えたり減ったりしてたと思います

これに関しては，実はいらないと思っています．
気のせいかもしれませんが，jointIdがふられていない関節数が増えてもhrpsysには影響しませんでした．
気のせいであれば，まあ上書きするのだと思います．

一つ一つ返信するのは難しいので僕の意見を書きます．
- simlation / 実機，各ロボットなどをif文で分岐させない．

ファイルが膨大になり編集しづらくなる・diffが見づらくなるというこのissueを立てた本来の問題が解決しないだけでなく，
> そのため，urata系ロボット以外(レポジトリ自体も違う)では上記の(1),(2)をそれぞれ継承してさらにロボット固有のパラメータをそれぞれ設定する必要があります．

こういった問題が頻発します．
一部のロボットだけ別のconfigファイルを用意する…ということをすると，そのロボットを使う人が混乱したり，upstreamの変更に取り残されると思います．

また，1つのシミュレーションのみを使っている場合は良いですが，他のシミュレーションを使いたくなって，仮にそれ固有の設定をしなければいけなくなった時に，またif文を増やすか，結局configを継承するかということになると思います．

> 統合というのは concatenate ではなくて，一般化していく，ということだと考えて，

これは完全におっしゃるとおりなのですが，
- シミュレーションでしか取れない情報をログに残したい要求がある (衝突力など)
- 各ロボットのパラメータを完全に同じにすることは不可能
- STやABCのパラメータをモデルファイルに書くことはできない

という点で，結局僕が以前書いた図のようにするしかないのではと信じています(URATAHrpsysConfiguratorを消したり，2つあるChoreonoidHrpsysConfiguratorを1つにするあたりは残っていますが)

ishiguroJSK, 2019-03-18:
> ファイルが膨大になり編集しづらくなる・diffが見づらくなるというこのissueを立てた本来の問題が解決しない

見づらいのもよくわかるけど，ファイルが増えて管理が行き届かなくなるのも中々の問題では？
泥臭い気がするけど，パラメータだけ別ファイルに分けてget関数なりdictなりでとってきてもいいわけだし

> そのため，urata系ロボット以外(レポジトリ自体も違う)では上記の(1),(2)をそれぞれ継承してさらにロボット固有のパラメータをそれぞれ設定する必要があります．
> 
> こういった問題が頻発します．
> 一部のロボットだけ別のconfigファイルを用意する…ということをすると，そのロボットを使う人が混乱したり，upstreamの変更に取り残されると思います．

外部のロボットがひと手間増えるのはしょうがない気がするけど，うまくいけば，rtmros_commonかrtmros_tutorialsの.pyを一回継承して実機/sim用のパラメータを上書きするような1つのpyだけで済むのでは？

> また，1つのシミュレーションのみを使っている場合は良いですが，他のシミュレーションを使いたくなって，仮にそれ固有の設定をしなければいけなくなった時に，またif文を増やすか，結局configを継承するかということになると思います．

ちょっと先を視過ぎな気がするけど，一時的に使うならchidori_simX_config.pyを作ればいいと思うけど，正式採用する際にまた○○_simX_config.pyがロボットの数だけ発生するのはいかがなものか

> シミュレーションでしか取れない情報をログに残したい要求がある (衝突力など)

これは継承後にconnect関数を上書きすればよいのでは？(正式採用するなら継承元のsim場合のconnectのとこに書く)

> 各ロボットのパラメータを完全に同じにすることは不可能
> STやABCのパラメータをモデルファイルに書くことはできない

できないのは賛成だけど，情報を一元化して管理を容易にすべき，という意味でファイルを分化させるのは大変では？(Pythonライクかどうかという話になると自信ないのだけど)

kyawawa, 2019-03-20:
ファイルが増えるのを嫌うのは分かりますが，同じファイルに複数の異なるロボットの設定があるのは同じくらい嫌ですし，現状起こっているようにどこかで歪が生じるはずです．

> また○○_simX_config.pyがロボットの数だけ発生するのはいかがなものか

これは確かにイマイチな点でした．
ただ，継承したファイルを作るだけならそれこそ自動生成できそうです．と今思いつきました．
もっとうまい方法がありそうなのは否めませんが…

結局，「例外的にこうしておいて，後で他と同じようにする」という運用を許してしまうと例外が何ヶ月にもわたって残ることは目に見えるので，始めから統一的な分け方をしておきたい，ということが大きいです．
ロボットは増えますし，シミュレータも，現に最近Choreonoidが主流になってきましたし，シミュレータを開発する方もいるので，(今ある)全てを一緒くたにするやり方はどうなのでしょう．

YoheiKakiuchi, 2019-03-20:
1) 個々のロボットの設定がひとつのファイルに書かれていて見にくい
2) Urataでないロボット(privateなロボット)は2箇所に設定が書いてあり使いにくい

論点が 1),2)の２つになってきているように思います。

1)はメリット/デメリットが拮抗しており、変更のメリットが薄そう。
ファイルを分けるといった方法よりも、パラメータ設定を書く場所でもっと良いところが無いか（パラメータがロボット単位になれば、同じファイルに複数ロボットの設定が無いは達成される）や、
idlの変更に対応してパラメータが健全であるかをチェックしてもらえるような機能があってほしいところ。

2)は、そもそも特定のロボットのpyが使いにくいだけではないかという考え方はどうだろうか？
これは、 @ishiguroJSK の意見に近いかな。
choreonoidや他のロボットの方は、動くことを優先してそれほど深く考えずにその時その時でなんとかした結果なので、そちらを見直せるならそれで解決する場合もある。

Naoki-Hiraoka, 2021-05-19:
SampleRobotやHRP2といった非URATA系のロボットでも使えるように，`ChoreonoidHrpsysConfiguratorOrg`や`RealHrpsysConfigurator`は`URATAHrpsysConfigurator`を継承するのではなく，`HrpsysConfigurator`を継承して欲しいです．