start-jsk/rtmros_tutorials #576: Questions about hrpsys_gazebo_tutorials: Cannot locate rosdep definition for [hrpsys_gazebo_general] and hrp models.

zhangOSK, 2020-05-06:
Dear Sir,
  When I install rtmros_tutorials, I met two questions.
  The first one is where can I download the `jsk_models` and `hrp2_models`?
  The second one is an Error occured when I run `rosdep install -r --from-paths src --ignore-src -y
`.  The error is:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
hrpsys_gazebo_tutorials: Cannot locate rosdep definition for [hrpsys_gazebo_general]
hrpsys_ros_bridge_tutorials: Cannot locate rosdep definition for [jsk_models]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully`

  My system is ubuntu16.04 and ROS kinetic. 
  I have also tried  `rosdep update` but it didn't solve my problem. Could you please tell me how to solve these two problems?
  Thank you!

ishiguroJSK, 2020-05-06:
First, sorry, `jsk_models` and `hrp2_models` are extra and private repositories due to its license.
Second, about the rosdep error,

- `jsk_models` is not essential and this can be ignored.
- `hrpsys_gazebo_general` is not released in kinetic yet (I think), but you can place the source package of `rtmros_gazebo`(https://github.com/start-jsk/rtmros_gazebo) in your workspace, and the dependency will be resolved.

However, we are now working on several issues about hrpsys_gazebo related tutorials, so I'm not sure if the sample tutorial works well.

zhangOSK, 2020-05-07:
Thank you for the reply!
The problem of dependency is solved and I can compile successfully!
Here is a short feedback about the sample tutorial. I tried the Sample Robot tutorial and I can see the robot after running `roslaunch hrpsys_gazebo_tutorials gazebo_samplerobot_no_controllers.launch` .
After running 
`rtmlaunch hrpsys_gazebo_tutorials samplerobot_hrpsys_bringup.launch`
and 
`roscd hrpsys_ros_bridge_tutorials/euslisp/
roseus samplerobot-interface.l
(samplerobot-init)
(send *ri* :angle-vector (send *sr* :reset-pose) 1000)
(send *ri* :start-auto-balancer)
(send *ri* :start-st)
(send *ri* :go-pos 0 0 0)`
Even though there is no error reported in terminal, those commands did not control the robot. The  robot still loses control and moving randomly.

Naoki-Hiraoka, 2020-05-07:
Currently, you can try sample with
```
roslaunch hrpsys_gazebo_general gazebo_samplerobot_no_controllers.launch
rtmlaunch hrpsys_gazebo_general samplerobot_hrpsys_bringup.launch
```
instead of
```
roslaunch hrpsys_gazebo_tutorials gazebo_samplerobot_no_controllers.launch
rtmlaunch hrpsys_gazebo_tutorials samplerobot_hrpsys_bringup.launch
```

We are now working on this issue in https://github.com/start-jsk/rtmros_tutorials/pull/568 .

zhangOSK, 2020-05-07:
Thank you for telling me the sample! The new one worked well!
Since the dependency and sample problems are solved, I will close the issue.

closed 2020-05-07 by zhangOSK