start-jsk/rtmros_tutorials #601: Unable to resolve paths for JAXON_RED.urdf.xacro

sktometometo, 2021-11-17:
[JAXON_RED.urdf.xacro]() includes JAXON_RED_body.urdf.xacro. But it does not exist. Where is it?

Naoki-Hiraoka, 2021-11-17:
It is auto-generated file.

https://github.com/start-jsk/rtmros_tutorials/blob/2edfaf5ef6c35ffdd5bbfb6799dd5c2090f31c8c/hrpsys_ros_bridge_tutorials/CMakeLists.txt#L538
https://github.com/start-jsk/rtmros_tutorials/blob/2edfaf5ef6c35ffdd5bbfb6799dd5c2090f31c8c/hrpsys_ros_bridge_tutorials/CMakeLists.txt#L735

To generate ` JAXON_RED_body.urdf`, you have to build `hrpsys_ros_bridge_tutorials` with `jsk_models`(private).

sktometometo, 2021-11-19:
Thanks, I understand.

closed 2021-11-19 by sktometometo

knorth55, 2021-11-19:
FYU: https://github.com/jsk-enshu/robot-programming/blob/master/cart_humanoid/urdf/JAXON_RED.urdf.xacro