start-jsk/rtmros_tutorials #624: NoeticのPCにHiroの手元PC環境を作成したい

Kanazawanaoaki, 2024-10-27:
最小構成や過不足ない情報では無いと思いますが，  
https://github.com/jsk-ros-pkg/trans_system/issues/817  などを見ながらNoeticでのHiroの手元PC環境構築を進めたので，その記録です．

## Setup
Download `ros-noetic-openrtm-aist-python_1.1.0-5focal_amd64.deb` from
https://drive.google.com/drive/u/0/folders/1bhD0r4uFIyW0e-fPLuPdTIC7NiIySMMt .
and install ros-noetic-openrtm-aist-python from deb file.
```bash
sudo apt install ./ros-noetic-openrtm-aist-python_1.1.0-5focal_amd64.deb
```
Setup ROS workspace.
```
source /opt/ros/noetic/setup.bash
mkdir -p ~/hiro_ws/src
cd ~/hiro_ws/src/
wstool init
wstool merge https://gist.githubusercontent.com/Kanazawanaoaki/07f5e5182d73381818afc878c3811919/raw/bcc0e29991b4177e2fd1d619b0256e3499fbc62f/hironxjsk_noetic.rosinstall 
wstool update
cd ..
rosdep update --include-eol-distros
rosdep install -y -r --from-paths src
catkin build hironx_tutorial
source ~/hiro_ws/devel/setup.bash
```

### 環境構築ができているかチェック
https://gist.github.com/Kanazawanaoaki/07f5e5182d73381818afc878c3811919#file-hiro-env-test-l を実行できるか試す
```bash
wget https://gist.githubusercontent.com/Kanazawanaoaki/07f5e5182d73381818afc878c3811919/raw/4abdf1d5a706f14993df00e8e74086c1a3a4a082/hiro-env-test.l
source ~/hiro_ws/devel/setup.bash
roseus hiro-env-test.l
```

### 備考
- https://mojel.blog.fc2.com/blog-entry-7.html のようなpyyamlのバージョンのエラー`pip install pyyaml==5.4.1`が必要
- https://github.com/jsk-ros-pkg/jsk_robot/issues/1892
- https://qiita.com/task-k/items/6cf9b17e65771fbf9205のように`fatal: unable to connect to github.com:`で`git config --global url."https://".insteadOf git://`が必要

などの対処が必要な可能性あり

pazeshun, 2024-12-11:
@k-okada openrtm_aist_pythonについて、
https://github.com/ros/rosdistro/pull/40441
でreleaseされてからずいぶん経つのに、まだapt installできないのでおかしいなと思って確認したところ、ビルドが失敗しているようです。
http://repositories.ros.org/status_page/noetic_default.html?q=openrtm-aist-python
https://build.ros.org/job/Nbin_uF64__openrtm_aist_python__ubuntu_focal_amd64__binary/lastBuild/console
ちょっと知識不足でPRレベルまで到達するのに時間かかりそうなので、一旦報告しました。

pazeshun, 2025-05-02:
ros-noetic-openrtm-aist-pythonがaptからインストールできるようになった
https://github.com/ros/rosdistro/pull/45247
ため、debファイルをダウンロードしてくる必要はなくなりました。

pazeshun, 2025-05-02:
現時点では、`python-is-python3`をインストールしておかないと`catkin build`時にエラーが出てしまう状況なので、修正予定です。

k-okada, 2025-05-09:
> 現時点では、python-is-python3をインストールしておかないとcatkin build時にエラーが出てしまう状況なので、修正予定です。

これってどこで出てきますか？

pazeshun, 2025-05-09:
@k-okada 

> > 現時点では、python-is-python3をインストールしておかないとcatkin build時にエラーが出てしまう状況なので、修正予定です。
> 
> これってどこで出てきますか？

僕の手元のdocker環境では、`catkin build hrpsys_ros_bridge`と`catkin build hrpsys_ros_bridge_tutorials`でエラーが出て、それぞれ
https://github.com/k-okada/rtmros_common/pull/15
https://github.com/start-jsk/rtmros_tutorials/pull/625
で修正したつもりです。
手元の環境ではエラーが出なくなって、 https://gist.github.com/Kanazawanaoaki/07f5e5182d73381818afc878c3811919#file-hiro-env-test-l がエラーなく実行できることを確認しました。